zephyrproject-rtos/zephyr #45894: up_squared：the test shows pass in the twister log but failed in daily report

shaoanxli, 2022-05-23:
**Describe the bug**
The test shows pass in the twister log but failed in daily report.  According to observation the twister log is not complete , this may caused the test case failed in daily report, once the revision(commit id: d5e7732e0d0f66bc9a3684f9597d9fd6a2d351b2 ) is withdrawn, the daily test report will return to normal.

Take tests/posix/common as an example.

**To Reproduce**
Steps to reproduce the behavior:
1. source zephyr-env.sh
2. twister -p up_squared --device-testing --device-serial $serial_name -T tests/posix/common/ --west-flash=$HOME/$USER/up_squared_efi.sh -vv
4. See error

**Expected behavior**
The test pass both in twister log and in daily report.

**Impact**
The test shows pass in the twister log but failed in daily report.

**Logs and console output**
```
Version 2.18.1263. Copyright (C) 2018 American Megatrends, Inc.
Press <DEL> or <ESC> to enter setup.

DEBUG   - DEVICE: >>Checking Media Presence......
DEBUG   - DEVICE: >>No Media Present......

DEBUG   - DEVICE: >>Checking Media Presence......
DEBUG   - DEVICE: >>Media Present......
DEBUG   - DEVICE: >>Start PXE over IPv4.
DEBUG   - DEVICE: Station IP address is 192.165.1.56
DEBUG   - DEVICE:
DEBUG   - DEVICE: Server IP address is 192.165.1.1
DEBUG   - DEVICE: NBP filename is up_squared/zephyr.efi
DEBUG   - DEVICE: NBP filesize is 2246389 Bytes

DEBUG   - DEVICE: >>Checking Media Presence......
DEBUG   - DEVICE: >>Media Present......
DEBUG   - DEVICE: Downloading NBP file...
DEBUG   - DEVICE:
DEBUG   - DEVICE: Succeed to download NBP file.
DEBUG   - DEVICE: *** Zephyr EFI Loader ***
DEBUG   - DEVICE: RSDP found at 0x79c34000
DEBUG   - DEVICE: Zeroing 5075584 bytes of memory at 0x117000
DEBUG   - DEVICE: Copying 32768 data bytes to 0x1000 from image offset
DEBUG   - DEVICE: Copying 94208 data bytes to 0x100000 from image offset 32768
DEBUG   - DEVICE: Copying 2112896 data bytes to 0x5ee280 from image offset 126976
DEBUG   - DEVICE: Jumping to Entry Point: 0x1137 (48 31 c0 48 31 d2 48)
DEBUG   - DEVICE: Running TESTSUITE posix_apis
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_posix_pthread_execution
DEBUG   - DEVICE: Thread 0 starting with scheduling policy 0 & priority 15
DEBUG   - DEVICE: Thread 1 starting with scheduling policy 0 & priority 15
DEBUG   - DEVICE: Thread 2 starting with scheduling policy 0 & priority 15
DEBUG   - DEVICE: Bounce test OK
DEBUG   - DEVICE: Barrier test OK
DEBUG   - DEVICE: PASS - test_posix_pthread_execution in 0.521 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_posix_pthread_error_condition
DEBUG   - DEVICE: PASS - test_posix_pthread_error_condition in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_posix_pthread_termination
DEBUG   - DEVICE: Thread 0 starting with a priority of 4
DEBUG   - DEVICE: Cancelling thread 0
DEBUG   - DEVICE: Thread 1 starting with a priority of 3
DEBUG   - DEVICE: Cancelling thread 1
DEBUG   - DEVICE: Thread 1 could not be cancelled
DEBUG   - DEVICE: Thread 2 starting with a priority of 2
DEBUG   - DEVICE: Cancelling thread 2
DEBUG   - DEVICE: Thread 3 starting with a priority of 1
DEBUG   - DEVICE: Cancelling thread 3
DEBUG   - DEVICE: Thread 3 could not be cancelled
DEBUG   - DEVICE: PASS - test_posix_pthread_termination in 1.16 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_posix_multiple_threads_single_key
DEBUG   - DEVICE:
DEBUG   - DEVICE: Different threads set different values to same key:
DEBUG   - DEVICE: thread 0: set value = 6214384 and retrieved value = 6214384
DEBUG   - DEVICE: thread 1: set value = 6214832 and retrieved value = 6214832
DEBUG   - DEVICE:
DEBUG   - DEVICE: PASS - test_posix_multiple_threads_single_key in 0.18 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_posix_single_thread_multiple_keys
DEBUG   - DEVICE:
DEBUG   - DEVICE: Single thread associates its value with different keys:
DEBUG   - DEVICE: key 0: set value = 6215232 and retrieved value = 6215232
DEBUG   - DEVICE: key 1: set value = 6215232 and retrieved value = 6215232
DEBUG   - DEVICE:
DEBUG   - DEVICE: PASS - test_posix_single_thread_multiple_keys in 0.18 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_posix_thread_attr_stacksize
DEBUG   - DEVICE: PASS - test_posix_thread_attr_stacksize in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_posix_clock
DEBUG   - DEVICE: POSIX clock APIs
DEBUG   - DEVICE: POSIX clock APIs test done
DEBUG   - DEVICE: PASS - test_posix_clock in 2.5 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_posix_semaphore
DEBUG   - DEVICE: PASS - test_posix_semaphore in 2.305 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_posix_normal_mutex
DEBUG   - DEVICE: mutex lock is taken
DEBUG   - DEVICE: PASS - test_posix_normal_mutex in 0.203 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_posix_recursive_mutex
DEBUG   - DEVICE: recursive mutex lock is taken
DEBUG   - DEVICE: PASS - test_posix_recursive_mutex in 0.4 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_posix_mqueue
DEBUG   - DEVICE: PASS - test_posix_mqueue in 0.11 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_posix_realtime
DEBUG   - DEVICE: PASS - test_posix_realtime in 1.802 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_posix_timer
DEBUG   - DEVICE: POSIX timer test
DEBUG   - DEVICE: Timer fires every 0 secs and  100000000 nsecs
DEBUG   - DEVICE: Time remaining to fire 0 secs and  899000000 nsecs
DEBUG   - DEVICE: Handler Signal value :20 for 1 times
DEBUG   - DEVICE: Handler Signal value :20 for 2 times
DEBUG   - DEVICE: Handler Signal value :20 for 3 times
DEBUG   - DEVICE: Handler Signal value :20 for 4 times
DEBUG   - DEVICE: Handler Signal value :20 for 5 times
DEBUG   - DEVICE: Handler Signal value :20 for 6 times
DEBUG   - DEVICE: Handler Signal value :20 for 7 times
DEBUG   - DEVICE: Handler Signal value :20 for 8 times
DEBUG   - DEVICE: Handler Signal value :20 for 9 times
DEBUG   - DEVICE: Handler Signal value :20 for 10 times
DEBUG   - DEVICE: Handler Signal value :20 for 11 times
DEBUG   - DEVICE: Handler Signal value :20 for 12 times
DEBUG   - DEVICE: PASS - test_posix_timer in 2.112 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_posix_rw_lock
DEBUG   - DEVICE:
DEBUG   - DEVICE: main acquire WR lock and 3 threads acquire RD lock
DEBUG   - Timed out while monitoring serial output on up_squared
DEBUG   - run status: up_squared/tests/posix/common/portability.posix.common.newlib passed
INFO    - 7/7 up_squared                tests/posix/common/portability.posix.common.newlib PASSED (device 122.595s)
```

**Environment:**
OS: Linux,
Toolchain Zephyr SDK 0.14.1
Commit SHA or Version: 4cc5c56d8c4a4727bcfb6a49cdc2d6ffba39295f

mbolivar-nordic, 2022-05-24:
@enjiamai is this possibly a duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/45942? Could you please close in favor of that issue if so, since it has a PR?

enjiamai, 2022-05-24:
Hi @mbolivar-nordic  @MaureenHelm,  yes, they are the same issue which is a twister issue. It duplicates with #45942.  So I am going to close it. 

By the way, the  #45929 is refer to an issue of x86, different with this one. Though it happens on the same case.

closed 2022-05-24 by enjiamai