zephyrproject-rtos/zephyr #45914: test: tests/kernel/usage/thread_runtime_stats/ test fail

hunterhu3000, 2022-05-23:
**Describe the bug**
The testcase "tests/kernel/usage/thread_runtime_stats/" gets assertion fail on both MEC15 and MEC17

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? ARM platform mec15xxevb_assy6853 and mec172xevb_assy6906
 - What have you tried to diagnose or workaround this issue?
 - do git bisect and found the test case is newly add by the commit of e83dde1628918e4751290d3fad8fa2519f5f1402

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b mec172xevb_assy6906 tests/kernel/usage/thread_runtime_stats/
2. check the serial log

**Expected behavior**
The test can pass

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
failed in daily tests

**Logs and console output**
```
Booting Zephyr OS build v3.1.0-rc2-15-ga8c7f825104a  ***
Running TESTSUITE usage_api
===================================================================
START - test_all_stats_usage
 PASS - test_all_stats_usage in 0.1 seconds
===================================================================
START - test_thread_stats_enable_disable
 PASS - test_thread_stats_enable_disable in 0.1 seconds
===================================================================
START - test_sys_stats_enable_disable
 PASS - test_sys_stats_enable_disable in 0.1 seconds
===================================================================
START - test_thread_stats_usage

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/usage/thread_runtime_stats/src/test_thread_runtime_stats.c:529: test_thread_stats_usage: stats5.average_cycles > stats3.average_cycles is false

 FAIL - test_thread_stats_usage in 0.20 seconds
===================================================================
TESTSUITE usage_api failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: ubuntu 20.04
 - Toolchain (e.g Zephyr SDK, ...): 0.14.1
 - Commit SHA or Version used: a8c7f825104aaead6cff624fb5e188b192664255

**Additional context**
None

hunterhu3000, 2022-05-31:
@peter-mitsis: Is this issue related to the change of [RTOS-6829][RTOS-6830][RTOS-6833]?

peter-mitsis, 2022-05-31:
@hunterhu3000: The change sets to which you refer is where this (failing) test case was first introduced. My expectation is that the cause of the bug is in one of three areas ...
  1) The gathering of the thread runtime statistics
  2A) A flaw in the test case
  2B) The platform is behaving in a way for which the test was not designed

Upon my own initial investigation into the bug, the failure at line 529 (as per the details in the reported issue), the expected ordering of average times (An) for stats samplings 3, 4 and 5 not as expected. Instead of A4 > A5 > A3, it is being reported as A3 >= A5.

The expected sequence of events is ...
1. Helper thread has three windows of execution of 10 ticks each. A3 should then be about 30/3 (or 10) ticks long.
2. Helper thread then has 1 window of execution of 20 ticks. A4 should then be about 50/4 (or 12.5) ticks long.
3. Helper thread then has 1 window of execution of 1 tick. A5 should then be about 51/5 (or 10.2) ticks long.

The expected values for A3 and A5 have the potential to be relatively close together in terms of ticks. If there is something off about the timing, that could conceivably give the observed report of A3 >= A5.

peter-mitsis, 2022-05-31:
I do not have a board with which to test the following idea. If it is indeed timing related, being off by a single tick with the numbers above has the potential to give the reported failure. To ensure that there is adequate buffering, the failing test could be modified such that the first three execution windows are of 5 ticks instead 10, and the fourth execution window could be set to 40 ticks instead of 20. This would result in expected average times for {A3, A4, A5} = {5, 13.75, 11.2}.

As I mentioned previously, I do not have the hardware with which to test whether this would resolve the issue or not.

hunterhu3000, 2022-06-02:
@peter-mitsis 
I have add some log before the assert(line 529)
```
*** Booting Zephyr OS build v3.1.0-rc3-32-g801fe1fa06bd  ***
Running TESTSUITE usage_api
===================================================================
START - test_all_stats_usage
 PASS - test_all_stats_usage in 0.1 seconds
===================================================================
START - test_thread_stats_enable_disable
 PASS - test_thread_stats_enable_disable in 0.1 seconds
===================================================================
START - test_sys_stats_enable_disable
 PASS - test_sys_stats_enable_disable in 0.1 seconds
===================================================================
START - test_thread_stats_usage
 stats1: execution_cycles, total_cycles, current_cycles, peak_cycles, average_cycles, idle_cycles: 0 0 0 0 0 0
 stats2: execution_cycles, total_cycles, current_cycles, peak_cycles, average_cycles, idle_cycles: 18 18 9 9 9 0
 stats3: execution_cycles, total_cycles, current_cycles, peak_cycles, average_cycles, idle_cycles: 27 27 9 9 9 0
 stats4: execution_cycles, total_cycles, current_cycles, peak_cycles, average_cycles, idle_cycles: 48 48 21 21 12 0
 stats5: execution_cycles, total_cycles, current_cycles, peak_cycles, average_cycles, idle_cycles: 49 49 1 21 9 0

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/usage/thread_runtime_stats/src/test_thread_runtime_stats.c:550: test_thread_stats_usage: stats5.average_cycles > stats3.average_cycles is false

 FAIL - test_thread_stats_usage in 0.69 seconds
===================================================================
TESTSUITE usage_api failed.
===================================================================
```

Theologically, it is true that "stats5.average_cycles > stats3.average_cycles". But actually the two values are very close. In the code, average_cycles is a integer value, so they can be the same value. So it seems we should change the assert to "stats5.average_cycles >= stats3.average_cycles". Or we can use a float value to record the average_cycles.

peter-mitsis, 2022-06-07:
@hunterhu3000 : Apologies for the late reply ... I was out of town for 5 days.

Thanks for gathering the numbers/data. As I look at the data you provided, it looks to me that rate at which the cycle count is increasing matches the system clock tick rate.  Assuming that interpretation is correct, then it looks like the base of the problem is a timer resolution issue.

I can think of two ways to resolve this. One way is (as you suggested) to change the assert to "stats5.average_cycles >= stats3.average_cycles". The second is (as I previously suggested) to change the length of each thread's execution window.

I think your proposal is both sufficient and simpler--and thus the better option. Can you make that change (as you seem to have access to the hardware to test it), and I will be happy to review the change.

rljordan, 2022-06-07:
@peter-mitsis  who should be owning this bug, @hunterhu3000 ?

hunterhu3000, 2022-06-07:
@rljordan I'll push a PR today

hunterhu3000, 2022-06-17:
Close fixed