zephyrproject-rtos/zephyr #45929: up_squared：failed to run test case tests/posix/common

shaoanxli, 2022-05-24:
**Describe the bug**
The test block on test_posix_rw_lock test case.

**To Reproduce**
Steps to reproduce the behavior:
1. source zephyr-env.sh
2. twister -p up_squared --device-testing --device-serial $serial_name -T tests/posix/common/ --west-flash=$HOME/$USER/up_squared_efi.sh -vv
3. See error
  or
  2. west build -p always -b up_squared tests/posix/common/
  3. screen /dev/ttyUP2 115200
  4. west flash $HOME/$USER/up_squared_efi.sh
  5. See error

**Expected behavior**
Test case should pass as expected.

**Impact**
The test case tests/posix/common seems to block.

**Logs and console output**
```
Running TESTSUITE posix_apis
===================================================================
START - test_posix_pthread_execution
Thread 0 starting with scheduling policy 0 & priority 15
Thread 1 starting with scheduling policy 0 & priority 15
Thread 2 starting with scheduling policy 0 & priority 15
Bounce test OK
Barrier test OK
 PASS - test_posix_pthread_execution in 0.521 seconds
===================================================================
START - test_posix_pthread_error_condition
 PASS - test_posix_pthread_error_condition in 0.1 seconds
===================================================================
START - test_posix_pthread_termination
Thread 0 starting with a priority of 4
Cancelling thread 0
Thread 1 starting with a priority of 3
Cancelling thread 1
Thread 1 could not be cancelled
Thread 2 starting with a priority of 2
Cancelling thread 2
Thread 3 starting with a priority of 1
Cancelling thread 3
Thread 3 could not be cancelled
 PASS - test_posix_pthread_termination in 1.16 seconds
===================================================================
START - test_posix_multiple_threads_single_key

Different threads set different values to same key:
thread 0: set value = 6210288 and retrieved value = 6210288
thread 1: set value = 6210736 and retrieved value = 6210736

 PASS - test_posix_multiple_threads_single_key in 0.18 seconds
===================================================================
START - test_posix_single_thread_multiple_keys

Single thread associates its value with different keys:
key 0: set value = 6211136 and retrieved value = 6211136
key 1: set value = 6211136 and retrieved value = 6211136

 PASS - test_posix_single_thread_multiple_keys in 0.18 seconds
===================================================================
START - test_posix_thread_attr_stacksize
 PASS - test_posix_thread_attr_stacksize in 0.1 seconds
===================================================================
START - test_posix_clock
POSIX clock APIs
POSIX clock APIs test done
 PASS - test_posix_clock in 2.5 seconds
===================================================================
START - test_posix_semaphore
 PASS - test_posix_semaphore in 2.305 seconds
===================================================================
START - test_posix_normal_mutex
mutex lock is taken
 PASS - test_posix_normal_mutex in 0.203 seconds
===================================================================
START - test_posix_recursive_mutex
recursive mutex lock is taken
 PASS - test_posix_recursive_mutex in 0.4 seconds
===================================================================
START - test_posix_mqueue
 PASS - test_posix_mqueue in 0.11 seconds
===================================================================
START - test_posix_realtime
 PASS - test_posix_realtime in 1.802 seconds
===================================================================
START - test_posix_timer
POSIX timer test
Timer fires every 0 secs and  100000000 nsecs
Time remaining to fire 0 secs and  899000000 nsecs
Handler Signal value :20 for 1 times
Handler Signal value :20 for 2 times
Handler Signal value :20 for 3 times
Handler Signal value :20 for 4 times
Handler Signal value :20 for 5 times
Handler Signal value :20 for 6 times
Handler Signal value :20 for 7 times
Handler Signal value :20 for 8 times
Handler Signal value :20 for 9 times
Handler Signal value :20 for 10 times
Handler Signal value :20 for 11 times
Handler Signal value :20 for 12 times
 PASS - test_posix_timer in 2.112 seconds
===================================================================
START - test_posix_rw_lock

main acquire WR lock and 3 threads acquire RD lock
Thread 2 sc
```

**Environment :**
OS: Linux,
Toolchain Zephyr SDK 0.14.1
Commit SHA or Version: 4cc5c56d8c4a4727bcfb6a49cdc2d6ffba39295f

enjiamai, 2022-05-25:
According to description, on up_squared, when CONFIG_UART_CONSOLE is unset (use the EFI console only), in this test_posix_rw_lock() test case, it will stuck in the scheduler during the usleep(1000ns) call. If we change all the to usleep < 999us (That means k_busy_wait 999us)  or just simply use k_yield() instead of all usleep(). It won't stuck.

So far I have no idea why unset the CONFIG_UART_CONSOLE will lead usleep to hang up in scheduler.  Hi @andyross , @dcpleung , could you please give some suggestion of what direction I can dive in? Thanks you!

enjiamai, 2022-05-30:
Hi @andyross , could you please help to look at this issue when you got a chance?  Thank you!

We found not only this testcase failed when removing CONFIG_UART_CONSOLE=y on up_squared, but also Following testcases also failed due to this: 
tests/kernel/interrupt/arch.interrupt.x86_64
tests/kernel/semaphore/sys_sem/kernel.semaphore.usage
tests/subsys/logging/log_core_additional/logging.add.log1_user
tests/benchmarks/sys_kernel/benchmark.kernel.core

It's kind of a problem that an IRQ lock / unlock do not work in pair some where, but I am not sure.

andyross, 2022-05-30:
In fact there's a known glitch with interrupts with the EFI console code on up_squared, see comments at: https://github.com/zephyrproject-rtos/zephyr/blob/main/arch/x86/core/efi.c#L76 .  The short version is that the firmware enables interrupts while in the console output code somewhere, which means that if you try to log something while holding an IRQ lock, the firmware will break your lock.

This seems very unlikely to be a problem with general code (the window is very small), but I wouldn't be surprised if we had code in the test suite that exercised this reliably.  If so, the only really robust workaround is to disable the EFI console for those use cases.  The EFI layer is really intended just for system bringup and debugging (because it works everywhere and doesn't require figuring out a UART driver configuration).  It's never going to be as good as a proper hardware driver for logging output.

Some other things might be worth trying though:

* Disable core interrupts at the IO-APIC level (won't work for MSI)
* Detect the case where we're trying to make an EFI call inside an ISR that interrupted an existing EFI thunk.  The only way to recover this would be to abort the log, but it might be good enough to make tests pass.
* Rework the tests a tiny bit so that they don't hit interrupt conditions inside of EFI calls (most likely, what's happening is that the test has latched an interrupt with a lock held, so when it gets released the interrupt happens synchronously).

enjiamai, 2022-05-30:
Thanks very much!  @andyross , I am trying your suggested solution.....

andyross, 2022-06-10:
I'm going to remove myself as an assignee here, since I just turned in the hardware I'd need to reproduce this.  Please continue to ping me with questions, though!

enjiamai, 2022-06-13:
Thanks @andyross , So far, there are 2 kinds of errors I can see:
(1) Like @andyross mentioned, the EFl simple text output call return with interrupt enable. we can workaround it by avoiding invoking printk() in the interrupt context.
(2) In some scenario, the k_sem_take() will be hang up. Not sure are these having the same root cause with the previous one.

enjiamai, 2022-06-17:
Summary the current issues when CONFIG_UART_CONSOLE is unset:

1. **use printk in ISR context**
tests/kernel/interrupt/     => print in kernel_isr

2. **stuck at k_sleep or k_sem_take ( actually in z_pend_curr())** 
tests/posix/common/	     => u_sleep hang up
tests/kernel/semaphore/sys_sem/		=> k_msleep hang up after ztest error hook.
samples/subsys/portability/cmsis_rtos_v1/philosophers/ =>  stuck at  k_sem_take()    
samples/kernel/metairq_dispatch/ => stuck at k_msgq_get() (in z_pend_curr)

3. **Pass when CONFIG_SMP**
tests/kernel/mutex/sys_mutex/  => stuck at z_smp_init() due to it cannot get the start_flag = 1
samples/subsys/tracing/   => stuck at z_mem_manage_boot_finish when having userspace  (samples/subsys/tracing/tracing.user) 

4. **Need to implement logging backend for EFI console at least**        
tests/subsys/logging/log_core_additional/    (logging.add.log1_user)
samples/subsys/logging/logger/ (samples/subsys/logging/logger/sample.logger.basic)
samples/subsys/logging/syst/

5.  **printf() has no output , fixed by specifying efi_console_putchar as __stdout_hook**
tests/benchmarks/sys_kernel/  =>   (benchmark.kernel.core)
samples/application_development/external_lib/  
samples/posix/eventfd/    => printf has no output
samples/userspace/hello_world_user/  => printf has no output 

So, I would like to apply a solution:
For 1,2,3 we still keep the CONFIG_UART_CONSOLE=y
For 4 also keep CONFIG_UART_CONSOLE=y until a logging backend for EFI console has implemented. 
For 5 we can solve it now.

Excuse me, @andyross,  For 2 and 3,  I have tried several way but I cannot have a good fix on it. Such as check a interrupt status or store the RFLAG after the efi_call, etc. It seems like your "cli" solution is the best so far, but still has issue on 1, 2, 3.   Any further suggestion on this issue?

andyross, 2022-06-17:
Do you have specifics on which tests fail?  It's likely to be specific tests that (1) do something while in a masked context that latches an interrupt and (2) do a printk(), which releases the interrupt lock and allows the interrupt to happen early.  Then later on something else happens that gets confused by the resulting corrupt state.  In those situations, I think the only reasonable workaround is just "don't use the EFI console" (though it might be possible to remove the offending printk).

It's a very useful feature, but that interrupt masking problem is going to make EFI logging unreliable.  There will be use cases where we just can't have it.

PR #46918 (2022-07-05): tests: fix these tests when using EFI console output function as default
This PR try to fix some testcases when applying the UEFI console as default output （with UART_CONSOLE disable)

Category 1: Implemented the simplest driver to support EFI console:
- Add a minimal EFI console driver to support printf, fix following tests: 
tests/benchmarks/sys_kernel/ 
samples/application_development/external_lib/  
samples/posix/eventfd/
samples/userspace/hello_world_user/

- Implement a logging backend EFI console to fix following tests:  
samples/subsys/logging/syst
tests/subsys/logging/log_core_additio [... cut]