zephyrproject-rtos/zephyr #45932: tests: subsys/logging/log_syst : failed to build on rpi_pico

LixinGuoX, 2022-05-24:
**Describe the bug**
Failed to build tests/subsys/logging/log_syst/ on rpi_pico

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p rpi_pico -T tests/subsys/logging/log_syst/
2. see error

**Logs and console output**
```
Loading Zephyr default modules (Zephyr base).
-- Application: /home/guo/zephyrproject/zephyr/tests/subsys/logging/log_syst
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Cache files will be written to: /home/guo/.cache/zephyr
-- Zephyr version: 3.1.0-rc2 (/home/guo/zephyrproject/zephyr)
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: rpi_pico
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.14.2 (/home/guo/zephyr-sdk-0.14.2)
-- Found dtc: /home/guo/zephyr-sdk-0.14.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.14.2 (/home/guo/zephyr-sdk-0.14.2)
-- Found BOARD.dts: /home/guo/zephyrproject/zephyr/boards/arm/rpi_pico/rpi_pico.dts
-- Generated zephyr.dts: /home/guo/zephyrproject/zephyr/twister-out/rpi_pico/tests/subsys/logging/log_syst/logging.log_syst.mipi_syst/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/guo/zephyrproject/zephyr/twister-out/rpi_pico/tests/subsys/logging/log_syst/logging.log_syst.mipi_syst/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/guo/zephyrproject/zephyr/twister-out/rpi_pico/tests/subsys/logging/log_syst/logging.log_syst.mipi_syst/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /home/guo/zephyrproject/zephyr/twister-out/rpi_pico/tests/subsys/logging/log_syst/logging.log_syst.mipi_syst/zephyr/dts.cmake
/home/guo/zephyrproject/zephyr/twister-out/rpi_pico/tests/subsys/logging/log_syst/logging.log_syst.mipi_syst/zephyr/zephyr.dts:60.34-65.5: Warning (simple_bus_reg): /soc/peripheral-clk: missing or empty reg/ranges property
/home/guo/zephyrproject/zephyr/twister-out/rpi_pico/tests/subsys/logging/log_syst/logging.log_syst.mipi_syst/zephyr/zephyr.dts:66.26-71.5: Warning (simple_bus_reg): /soc/system-clk: missing or empty reg/ranges property
/home/guo/zephyrproject/zephyr/twister-out/rpi_pico/tests/subsys/logging/log_syst/logging.log_syst.mipi_syst/zephyr/zephyr.dts:81.36-112.5: Warning (unique_unit_address_if_enabled): /soc/pin-controller@40014000: duplicate unit-address (also used in node /soc/gpio@40014000)
Parsing /home/guo/zephyrproject/zephyr/tests/subsys/logging/log_syst/Kconfig
Loaded configuration '/home/guo/zephyrproject/zephyr/boards/arm/rpi_pico/rpi_pico_defconfig'
Merged configuration '/home/guo/zephyrproject/zephyr/tests/subsys/logging/log_syst/prj.conf'
Merged configuration '/home/guo/zephyrproject/zephyr/twister-out/rpi_pico/tests/subsys/logging/log_syst/logging.log_syst.mipi_syst/twister/testsuite_extra.conf'
Configuration saved to '/home/guo/zephyrproject/zephyr/twister-out/rpi_pico/tests/subsys/logging/log_syst/logging.log_syst.mipi_syst/zephyr/.config'
Kconfig header saved to '/home/guo/zephyrproject/zephyr/twister-out/rpi_pico/tests/subsys/logging/log_syst/logging.log_syst.mipi_syst/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/guo/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/guo/zephyrproject/zephyr/twister-out/rpi_pico/tests/subsys/logging/log_syst/logging.log_syst.mipi_syst
[  1%] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
[  2%] Built target parse_syscalls_target
[  2%] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h, include/generated/otype-to-size.h
[  2%] Built target kobj_types_h_target
[  2%] Generating include/generated/syscall_dispatch.c, include/generated/syscall_list.h
[  2%] Built target syscall_list_h_target
[  3%] Generating include/generated/driver-validation.h
[  3%] Built target driver_validation_h_target
[  3%] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
[  3%] Built target offsets
[  4%] Generating include/generated/offsets.h
[  4%] Built target offsets_h
[  5%] Generating include/generated/version.h
-- Zephyr version: 3.1.0-rc2 (/home/guo/zephyrproject/zephyr), build: v3.1.0-rc2-18-gea41f4ec8795
[  5%] Built target version_h
[  5%] Built target zephyr_generated_headers
[  5%] Building C object CMakeFiles/app.dir/src/main.c.obj
[  6%] Building C object CMakeFiles/app.dir/src/mock_backend.c.obj
[  6%] Linking C static library app/libapp.a
[  6%] Built target app
[  6%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/cbprintf.c.obj
[  7%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/cbprintf_packaged.c.obj
[  8%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc32c_sw.c.obj
[  8%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc32_sw.c.obj
[  9%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc16_sw.c.obj
[  9%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc8_sw.c.obj
[ 10%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc7_sw.c.obj
[ 10%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/dec.c.obj
[... 111 lines cut ...]
/home/guo/zephyrproject/zephyr/include/zephyr/drivers/reset.h: In function 'z_impl_reset_assert':
/home/guo/zephyrproject/zephyr/include/zephyr/drivers/reset.h:213:28: error: macro "assert" passed 2 arguments, but takes just 1
  213 |  return api->assert(dev, id);
      |                            ^
In file included from /home/guo/zephyr-sdk-0.14.2/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/reent.h:503,
                 from /home/guo/zephyr-sdk-0.14.2/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/errno.h:11,
                 from /home/guo/zephyr-sdk-0.14.2/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/errno.h:9,
                 from /home/guo/zephyrproject/zephyr/include/zephyr/drivers/uart.h:23,
                 from /home/guo/zephyrproject/zephyr/drivers/serial/uart_rpi_pico.c:8:
/home/guo/zephyr-sdk-0.14.2/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/assert.h:16: note: macro "assert" defined here
   16 | # define assert(__e) ((__e) ? (void)0 : __assert_func (__FILE__, __LINE__, \
      | 
In file included from /home/guo/zephyrproject/zephyr/drivers/serial/uart_rpi_pico.c:9:
/home/guo/zephyrproject/zephyr/include/zephyr/drivers/reset.h:213:12: error: returning 'reset_api_assert' {aka 'int (* const)(const struct device *, unsigned int)'} from a function with return type 'int' makes integer from pointer without a cast [-Werror=int-conversion]
  213 |  return api->assert(dev, id);
      |         ~~~^~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/build.make:76: zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/uart_rpi_pico.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:3276: zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain 0.14.2

stephanosio, 2022-05-25:
@yonsch The particular test that fails is `tests/subsys/logging/log_syst/logging.log_syst.mipi_syst`:
https://github.com/zephyrproject-rtos/zephyr/blob/25e6279493d673ec76ef5cb72bbfb0b651274144/tests/subsys/logging/log_syst/testcase.yaml#L2-L17

You can reproduce by running:
```
west build -b rpi_pico tests/subsys/logging/log_syst -DCONFIG_LOG_MIPI_SYST_ENABLE=y -DCONFIG_LOG_BACKEND_MOCK_OUTPUT_SYST=y
```

fabiobaltieri, 2022-05-25:
Hacking an `#undef` into the code still does not fix the build for an unrelated issue related to the pico second stage bootloader, opened #46027 for it.

closed 2022-05-26 by mbolivar-nordic