zephyrproject-rtos/zephyr #45933: webusb sample code linking error for esp32 board

enthunilu, 2022-05-24:
**Describe the bug**
I am trying to build webusb sample code for my esp32 dev kit.
https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/subsys/usb/webusb

I am using below command,
west build -p auto -b esp_wrover_kit samples/subsys/usb/webusb

while execution I am getting linking error for usb API's.

Steps:
1. Open command prompt
2. Go to zephyr directory
3. execute: west build -p auto -b esp_wrover_kit samples/subsys/usb/webusb
4. Check logs on console

**Logs and console output**

D:/zephyr/zephyrproject/zephyrproject/zephyr/subsys/usb/device/usb_transfer.c:111: undefined reference to `usb_dc_ep_read_wait'
d:/esp32/esp-idf-install/.espressif/tools/zephyr/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/zephyr/zephyrproject/zephyrproject/zephyr/subsys/usb/device/usb_transfer.c:125: undefined reference to `usb_dc_ep_mps'
d:/esp32/esp-idf-install/.espressif/tools/zephyr/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/zephyr/zephyrproject/zephyrproject/zephyr/subsys/usb/device/usb_transfer.c:125: undefined reference to `usb_dc_ep_read_continue'
d:/esp32/esp-idf-install/.espressif/tools/zephyr/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: zephyr/libzephyr.a(usb_transfer.c.obj): in function `usb_transfer_ep_callback':
D:/zephyr/zephyrproject/zephyrproject/zephyr/subsys/usb/device/usb_transfer.c:180: undefined reference to `usb_dc_ep_read_wait'
d:/esp32/esp-idf-install/.espressif/tools/zephyr/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: zephyr/libzephyr.a(usb_transfer.c.obj): in function `usb_transfer':
D:/zephyr/zephyrproject/zephyrproject/zephyr/subsys/usb/device/usb_transfer.c:244: undefined reference to `usb_dc_ep_mps'
d:/esp32/esp-idf-install/.espressif/tools/zephyr/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/zephyr/zephyrproject/zephyrproject/zephyr/subsys/usb/device/usb_transfer.c:244: undefined reference to `usb_dc_ep_mps'
d:/esp32/esp-idf-install/.espressif/tools/zephyr/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/zephyr/zephyrproject/zephyrproject/zephyr/subsys/usb/device/usb_transfer.c:201: undefined reference to `usb_dc_ep_read_continue'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'D:\zephyr\zephyrproject\zephyrproject\zephyr\build'

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: Zephyr SDK 

Note: I am able to build same example for nrf board i.e. bl653_dvk

I am not sure is there any setting or something need to be done for esp32 device before building code.
Please suggest if I am missing anything.

Thanks!!

mbolivar-nordic, 2022-05-24:
@enthunilu please open an enhancement issue for adding USB support to this platform if you would like

closed 2022-05-24 by mbolivar-nordic

enthunilu, 2022-05-25:
Thank you so much @mbolivar-nordic for the quick update.
Sure, I'll raise enhancement ticket for the support.

If possible can you please share any example code or documentation on USB RNDIS support ?

Thanks!!

mbolivar-nordic, 2022-05-25:
> If possible can you please share any example code or documentation on USB RNDIS support ?

@sylvioalves

enthunilu, 2022-05-26:
Hello @sylvioalves ,
Sorry, I missed to update.
I am not expecting example code or documentation for ESP32 device. Any device supporting example code or documentation will be very helpful.

enthunilu, 2022-06-06:
Hello @sylvioalves and @mbolivar-nordic,

Can anyone please help me with any example code or documentation on USB RNDIS support in zephyr ?

Thanks!!