zephyrproject-rtos/zephyr #45936: flash_sam0: Driver does not support RWW sections

bamkrs, 2022-05-24:
**Is your enhancement proposal related to a problem? Please describe.**
Some sam0 devices have a special RWW flash section. For example a `SAMD21E17D` has a 4Kbytes section at `0x00400000`.
The current `flash_sam0.c` implementation does not know about these special sections and complains with an `EINVAL`:
 
> [    0.000000] <wrn> flash_sam0: 0x4003fc: ends past the end of flash

**Describe the solution you'd like**
Implement necessary config-options and runtime-checks to allow the usage of the RWW section.

**Describe alternatives you've considered**
One has to fall back to the normal flash region which reduces the usable amount of flash for the application.

**Additional context**
[image]
**Notes:**
1. x = G, J or E.
2. Only applicable for device variants B, C, D,and L.

maksimdrachov, 2024-05-29:
@bamkrs Any chance you managed to solve this? (Could you share some pointers if so?)

[image]

maksimdrachov, 2024-05-31:
https://microchip-mplab-harmony.github.io/csp_apps_sam_c20_c21/apps/nvmctrl/nvmctrl_flash_read_write/readme.html

For those stuck on the same, check out the above, it provides example code on how to interact with NVMCTRL.