zephyrproject-rtos/zephyr #45955: stm32h7 i2s support

endian-benjamin, 2022-05-24:
Currently I2S is supported for some STM32:s but not for H7.

I figured the difference between F4 and H7 can't be THAT significant, so I considered porting it but with my somewhat limited experience with the low-level ST HAL, it's proven difficult.

From what I can tell, there are some differences that makes the current driver incompatible. After hacking in the (hopefully correct) clock source, I was left with:

```
undefined reference to `LL_I2S_DisableIT_ERR'
undefined reference to `LL_I2S_EnableIT_ERR'
undefined reference to `LL_SPI_DMA_GetRegAddr'
undefined reference to `LL_RCC_PLL_GetMainSource'
undefined reference to `LL_RCC_SetI2SClockSource'
```

- `DisableIT_ERR` seems to not be implemented, despite a macro being available (`__HAL_I2S_DISABLE_IT(__HANDLE__, __INTERRUPT__)`  from `stm32h7xx_hal_i2s.h`)
- `GetRegAddr` seems to not exist because there are now separate buffers for RX and TX (if I'm reading the HAL headers correctly)
- The RCC stuff seems to be due to clock being configured slightly differently

All in all, it seems hard to imagine that support for H7 would be all that tricky... I'd absolutely be willing to give it a try, but I wanted to throw the question out there first - is anyone else working on this? Are there any pitfalls to know about (I recall reading something about certain restrictions on which RAM regions that are usable for DMA for instance)? Any helpful advice in general? :D

erwango, 2022-05-24:
About providing series abstraction, you can look at what was done on spi_stm32, which is somewhat close to I2S:
https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/spi/spi_ll_stm32.h#L130-145

About the RCC stuff, I suggest to first convert i2s driver to use new device tree alternate clock source to simplify the source clk selection: https://github.com/zephyrproject-rtos/zephyr/pull/45053

emillindq, 2022-05-31:
I have done this and it works great, however very sadly I'm not able to contribute to Zephyr about this or share any code because employment policy states all code I produce on the clock belongs to the company. But I can share a few insights I made. Please note my diff I'm looking at is quite outdated so maybe the current state of the existing driver is slightly different
* You have to use DCACHE, for me the existing outcommented macro definitions for __DCACHE_PRESENT=1 worked
* LL_RCC_SetI2SClockSource is LL_RCC_SetSPIClockSource
* I didn't use the CONFIG_I2S_STM32_USE_PLLI2S_ENABLE, you can and need to get the in clock frequency from HAL_RCCEx_GetPeriphCLKFreq
* instead of LL_SPI_DMA_GetRegAddr use &(cfg->i2s->TXDR)
* in i2s_stm32_isr you have to explicitly clear the UDR flag using if (LL_I2S_IsActiveFlag_UDR) and corresponding clear function
* LL_I2S_EnableIT_ERR doesn't exist, just enable the errors OVR, UDR and FRE using their respectively clear functions
* Above also for disable
* I don't remember why but in the config struct there's .i2s_clk_sel, which I hard-coded to LL_RCC_SPI123_CLKSOURCE_PLL1Q

In addition, the DMA works differently, a DMAMUX1 and DMA1 must be instantiated and a reference to the DMAMUX1 must be inserted into the "dmas" property of the i2s1 node in the dts with appropriate line numbers. Also note that in your code that feeds the i2s api (i2s_write and such) you have to clean the dcache buffer of the samples and align them to the MCU's dcache boundary.

I think I got all of it, good luck and please ask questions if you run into problems.

closed 2022-08-14 by github-actions[bot]

deyro, 2023-08-28:
Hello,

Is there any update on this? I am looking to use the i2s on stm32h753. I am getting the same errors as mentioned by @endian-benjamin. Any pointers will be very helpful.

Thanks