zephyrproject-rtos/zephyr #45956: net: socket: dtls: handle multiple fragments in sendmsg()

mniestroj, 2022-05-24:
**Is your enhancement proposal related to a problem? Please describe.**
Using `sendmsg()` API does not work for DTLS connection, as fragmented data is not properly sent as a single DTLS record.

**Describe the solution you'd like**
Ideally, there should be a 'gather' write/send API in mbedTLS or alternatively an explicit "send flush" request (as implemented in https://github.com/Mbed-TLS/mbedtls/pull/4188).

**Describe alternatives you've considered**
Currently mbedTLS does not support 'gather' write, so as a workaround an intermediate buffer can be allocated to "linearize" all data passed to `sendmsg()` and then request single mbedTLS write.

**Additional context**
Related bug report: #45948

boaks, 2022-07-14:
If an application-message (or dtls-record) is split in several UDP packets, I'm not sure about the reliability of the outcome. That opens also a too large DoS door, sending a lot of data (upd-packets), but not that last upd-packet keeping the other peer holding data, but can't verify the MAC in order to withdraw it.
In my opinion, such an approach doesn't make too much sense for so much trouble.

mniestroj, 2022-08-10:
@boaks It is about sending fragmented data (many application chunks) as single dtls-record, not the other way. The problem that we have right now is that sending multiple chunks with single `sendmsg()` API call results in each chunk being sent in separate dtls-record. What we want is sending all the chunks together (linearized and authenticated/encrypted) in single dtls-record.

boaks, 2022-08-13:
We will see, if that pays off.
In my experience, the really benefit of the UDP style is to let an upper layer do, what could be done better there. I don't know, why the application emit small chunks, but maybe there is also a better application specific way to group them.

boaks, 2022-08-18:
Some additional considerations:

In difference to TCP, UDP includes the application payload length into the UDP messsage (see [RFC 768](https://www.rfc-editor.org/rfc/rfc768). Therefore, in difference to TCP, is you send

UDP-1 8 bytes, UDP-2 16 bytes, and , UDP-3 6 bytes.

you will always receive/read on the peer exactly one of these messages, may be in different order (in my experience really rare) or some dropped (in my experience 2%). But you never read assembled messages, e.g. UDP-2+3 22 bytes.

This also reflect the usage of the API, the sequence of bytes passed to `sendto` is considered to be the same as `recvfrom` (also different to TCP).

Maybe, the idea is to "relax" that to the behavior of TCP. In my experience, that doesn't work to well, UDP based protocols are frequently using that UDP message length.   

That will raise the question, where the length of the fragments are going to be included. The UDP message will only contain the overall length (including the DTLS header). If you put multiple DTLS records in one UDP message, then the length of the fragments will be encoded in the DTLS record. Such a solution may just encrypt the fragments and collect the resulting DTLS records in one UDP message. That solution will then result in
1x UDP-header: IPv4 20 bytes + UDP 8 bytes. => 28 bytes
Nx DTLS-header + MAC, e.g. AES-CCM: header 13 bytes + nonce 8 bytes + MAC 16 bytes. => 37 bytes

Assuming 4 fragments, the saving will be 3x 28 bytes.

Alternative solutions will in my opinion either change the grants of  `sendto`/`recvfrom`  or will add an additional "application side layer" (top of dtls).

So, please, do not only describe your idea on a high level, please add more details about the intended implementation.

boaks, 2022-08-18:
Just an additional stackoverflow link about [sendmsg](https://stackoverflow.com/questions/4258834/how-sendmsg-works). With that, it's  just intended to accumulates the data into one message, so one dtls record and one udp packet.

mniestroj, 2022-08-18:
>Just an additional stackoverflow link about [sendmsg](https://stackoverflow.com/questions/4258834/how-sendmsg-works). With that, it's just intended to accumulates the data into one message, so one dtls record and one udp packet.

That is exactly what this issue is about!

rlubos, 2024-02-09:
This is still a valid request, so keeping it open. Maybe I could find some time to look into this closer.

PR #69280 (2024-03-06): net: socket: dtls: handle multiple fragments in sendmsg()
DTLS socket should be able to combine more than one message into a single datagram when calling sendmsg().

Resolves #45956
closed 2024-03-06 by aescolar