zephyrproject-rtos/zephyr #45968: Bluetooth: OTS: Rename `struct bt_ots_init` to `struct bt_ots_init_param`

Thalley, 2022-05-24:
**Is your enhancement proposal related to a problem? Please describe.**
The `bt_ots_init` creates a duplicate warning as we also have a function called `bt_ots_init`. Since the struct actual contains parameters to that function, we should rename it. 

**Describe the solution you'd like**
Rename `struct bt_ots_init` to `struct bt_ots_init_param` to avoid duplicating the name `bt_ots_init`. 

**Describe alternatives you've considered**
N/A

**Additional context**
Was added as a known issue in https://github.com/zephyrproject-rtos/zephyr/pull/45908, and should be removed as part of the fix.

Thalley, 2022-05-24:
> Note that this warning is due to a Sphinx/Breathe bug, so there's no need to adjust valid C code. This is why we have the warnings_filter Sphinx extension.

I agree, but IMO it's still bad practice to have structs and functions with the same name, and we also have a case (which perhaps is also causing warnings) where we have a function and a local variable with the same name:
```c
static int ots_init(void)
{
	int err;
	struct bt_ots *ots;
	struct object_creation_data obj_data;
	struct bt_ots_init ots_init;
```

jaiiarora, 2023-06-30:
Is this issue is still open, I would like to work on it

Thalley, 2023-06-30:
> If this issue is still open, I would like to work on it

It is still open yes :)

jaiiarora, 2023-06-30:
Please take a look at my pull [request](https://github.com/zephyrproject-rtos/zephyr/pull/59870). @Thalley 
Thank you
https://github.com/zephyrproject-rtos/zephyr/pull/59870

closed 2023-06-30 by carlescufi