zephyrproject-rtos/zephyr #45993: Matter(CHIP) support

ck-telecom, 2022-05-25:
## Origin

Matter(CHIP)
https://github.com/project-chip/connectedhomeip

## Purpose

Matter(It don't matter) is an Application Layer Protocol for home IoT

## Mode of integration

module? 

## Pull Request

Not yet

## Description

For now to integrate to Zephyr based project is a little hard, user
could only integrate Matter only add as External Project and it does
not make sense of much flexible for configuration

What problem are you trying to solve? 
Hum... I don't know how to say

Why is this the right component to solve it (e.g., SQLite is small,
easy to use, and has a very liberal license.)

# Dependencies

Lower protocol such as Thread and etc

Will the Zephyr project have a direct dependency on the component, or
will it be included via an abstraction layer with this component as a
replaceable implementation?

## Revision

The latest version

## License

Please use an SPDX identifier (https://spdx.org/licenses/), such as
``BSD-3-Clause``

carlescufi, 2022-05-25:
@ck-telecom let me understand, you'd like Matter to be part of the Zephyr distribution by default by integrating it as a module upstream?

ck-telecom, 2022-05-26:
Yes, But not sure, the build system is not same with Zephyr, I think it is worth talking about this topic.

LuDuda, 2022-05-31:
Hello @ck-telecom 

Well, it is not true that Matter SDK only supports nRF5x serires with Zephyr.

Matter SDK takes approach of the "main" repository, and apart of the Matter stack it contains also platforms integration. You can find three related to Zephyr platforms already:

1. [Zephyr platform](https://github.com/project-chip/connectedhomeip/tree/master/src/platform/Zephyr)- it implements generic functionality for all Zephyr platforms e.g. BLE Host, Settings, Thread integration and so on.
2. [nRF Connect platform](https://github.com/project-chip/connectedhomeip/tree/master/src/platform/nrfconnect) - it implements functionality specific to nRF Connect SDK (i.e. nRF SoCs) e.g. NFC, DFU, Configuration management. This platform inherits from Zephyr platform and just adds and overrides some functionality.
3. [Telink platform](https://github.com/project-chip/connectedhomeip/tree/master/src/platform/telink) - it implements functionality specific to Telink platform. This platform also inherits from Zephyr platform and just adds and overrides some functionality.

We've done some effort at the beginning to separate Zephyr platform from nRF specific functionality (nRF Connect), which was reused later on by Zephyr-based Telink platform.

ck-telecom, 2022-06-01:
@LuDuda Thanks for your reply!
>We've done some effort at the beginning to separate Zephyr platform from nRF specific functionality (nRF Connect), which was reused later on by Zephyr-based Telink platform.

I do think that your effort on the separate Zephyr platform coz I can not configure my <chip_root>/zephyr/module.yml file as the following, that only depends on zephyr platform:
```
name: connectedhomeip

build:
    cmake: config/zephyr/chip-module
    kconfig: config/zephyr/Kconfig
    depends:
    - openthread
```
after separate from zephyr, nrf connect and telink could *import* the basic zephyr configuration for their addtion and overrides

ck-telecom, 2022-06-06:
```
name: connectedhomeip

build:
    cmake-ext: Ture
    kconfig: config/zephyr/Kconfig
    depends:
    - openthread
```
I think if we integrate Matter(CHIP) to Zephyr as a module to upstream, and add ${ZEPHY_BASE}/modules/matter/CMakeLists.txt that like openthread

Damian-Nordic, 2022-06-06:
@ck-telecom I think the best way to add support for standalone Zephyr platform in Matter is:
1. Create a similar directory structure in `MATTER-REPO/config/Zephyr` as we have in `MATTER-REPO/config/nrfconnect` and `MATTER-REPO/config/telink`. `Kconfig` file for Zephyr already exists, but `CMakeLists.txt` is missing. Also, since Matter uses GN build system, a GN project definition for Zephyr is needed.
2. Add missing files in `src/platform/Zephyr`, such as `CHIPDeviceConfig.h` or `CHIPDevicePlatformConfig.h`
3. Review Matter's `BUILD.gn` and `*.gni` files. Verify that conditions such as `chip_device_platform == "nrfconnect"` don't need to be extended on other Zephyr platforms.
4. Create a fork of the Matter repo in zephyrproject-rtos.
5. On the fork, add `zephyr/module.yml` that uses `CMakeLists.txt` and `Kconfig` from `MATTER-REPO/config/Zephyr/chip-module`.

That way, the only conflict we will have between Zephyr's and nRF Connect's Matter fork will be the Zephyr module manifest.

ck-telecom, 2022-06-07:
@Damian-Nordic Thanks for your tips!

I have started in my repo: https://github.com/ck-telecom/connectedhomeip/tree/zephyr_standalone

ck-telecom, 2022-06-13:
PR for Matter: https://github.com/project-chip/connectedhomeip/pull/19504

ck-telecom, 2022-06-15:
> Create a fork of the Matter repo in zephyrproject-rtos.

@Damian-Nordic how about this?

Damian-Nordic, 2022-06-15:
I guess we need someone with higher privileges. @carlescufi are you in charge of creating repos in zephyrproject-rtos?

carlescufi, 2022-07-06:
@ck-telecom or @Damian-Nordic for the TSC to approve this we need additional details in the issue body (eg. clear statement about module and maintainership) and also a PR showing how the integration would work.

closed 2022-09-20 by github-actions[bot]