zephyrproject-rtos/zephyr #46008: stm32h7: gptp sample does not work at all

gleichdick, 2022-05-25:
**Describe the bug**
Compiling and running the gPTP example does not work at all with a Nucleo H743 board.
The other PTP device is an Intel i210 with `ptp4l -4 -i eth1 -S -s` which was tested with another i210 card.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/net/gptp
2. west build -b nucleo_h743zi .
3. flash

**Expected behavior**
gPTP works as described in the [docs](https://docs.zephyrproject.org/latest/samples/net/gptp/README.html).

**Impact**
No PTP synchronization between a Nucleo board and a Linux PC.

**Logs and console output**

The board says the following:
```
uart:~*** Booting Zephyr OS build v3.1.0-rc2-64-gf3d460ff5ae6  ***
[00:00:00.007,000] <inf> net_config: Initializing network
[00:00:00.007,000] <inf> net_config: Waiting interface 1 (0x24001228) to be up...
[00:00:00.500,000] <inf> net_config: Interface 1 (0x24001228) coming up
[00:00:00.501,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:00.601,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:01.504,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:02.508,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:03.512,000] <wrn> net_gptp: Reset Pdelay requests
uart:~$ net gptp
Port Interface          Role
 1   0x24001228 [1]     DISABLED

SiteSyncSync state machine variables:
        Current state                  : INITIALIZING
        A PortSyncSync struct is ready : no
ClockSlaveSync state machine variables:
        Current state                  : SEND_SYNC_IND
        A PortSyncSync struct is ready : no
        The local clock has expired    : no
PortRoleSelection state machine variables:
        Current state                  : ROLE_SELECTION
ClockMasterSyncReceive state machine variables:
        Current state                  : SOURCE_TIME
        A ClockSourceTime              : no
        The local clock has expired    : no
```
Zephyr log with `CONFIG_NET_GPTP_LOG_LEVEL_DBG=y`:
```
uart:~*** Booting Zephyr OS build v3.1.0-rc2-64-gf3d460ff5ae6  ***
[00:00:00.007,000] <inf> net_config: Initializing network
[00:00:00.007,000] <inf> net_config: Waiting interface 1 (0x24001228) to be up...
[00:00:00.500,000] <inf> net_config: Interface 1 (0x24001228) coming up
[00:00:00.501,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:00.601,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:05.002,000] <dbg> net_gptp: gptp_thread: (gptp): Starting PTP thread
[00:00:05.002,000] <dbg> net_gptp: gptp_change_port_state_debug: (gptp): [1] state INITIALIZING -> DISABLED (gptp_thread():561)
[00:00:05.003,000] <dbg> net_gptp: gptp_change_pa_info_state_debug: (gptp): [1] PA info state DISABLED -> POST_DISABLED (gptp_mi_port_announce_information_state_machine():1445)
[00:00:05.003,000] <dbg> net_gptp: gptp_change_port_state_debug: (gptp): [0] state INITIALIZING -> SLAVE (gptp_updt_roles_tree():1876)
[00:00:05.004,000] <dbg> net_gptp: gptp_send_pdelay_req: (gptp): Sending PDELAY_REQ seq 12625 pkt 0x2400cdf0
[00:00:06.006,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:06.008,000] <dbg> net_gptp: gptp_send_pdelay_req: (gptp): Sending PDELAY_REQ seq 12626 pkt 0x2400cdf0
[00:00:07.010,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:07.012,000] <dbg> net_gptp: gptp_send_pdelay_req: (gptp): Sending PDELAY_REQ seq 12627 pkt 0x2400cdf0
[00:00:08.013,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:08.015,000] <dbg> net_gptp: gptp_send_pdelay_req: (gptp): Sending PDELAY_REQ seq 12628 pkt 0x2400cdf0
```

The network setup seems to have problems as well:
```
uart:~$ net ping 192.0.2.2 
PING 192.0.2.2
28 bytes from 192.0.2.2 to 192.0.2.1: icmp_seq=0 ttl=64 time=1 ms
28 bytes from 192.0.2.2 to 192.0.2.1: icmp_seq=1 ttl=64 time=0 ms
28 bytes from 192.0.2.2 to 192.0.2.1: icmp_seq=2 ttl=64 time=1 ms
```
Pinging from the host has some initial package losses:
```
~> ping stm32
PING stm32 (192.0.2.1) 56(84) bytes of data.
64 bytes from stm32 (192.0.2.1): icmp_seq=9 ttl=64 time=1.02 ms
64 bytes from stm32 (192.0.2.1): icmp_seq=10 ttl=64 time=0.518 ms
64 bytes from stm32 (192.0.2.1): icmp_seq=11 ttl=64 time=0.652 ms
64 bytes from stm32 (192.0.2.1): icmp_seq=12 ttl=64 time=0.541 ms
^C
--- stm32 ping statistics ---
12 packets transmitted, 4 received, 66.6667% packet loss, time 11211ms
rtt min/avg/max/mdev = 0.518/0.682/1.017/0.199 ms
```

But surprisingly the telnet example works flawless...

**Environment (please complete the following information):**
 - OS: OpenSuSE Tumbleweed
 - Toolchain: CROSS_COMPILE (gcc version 10.3.1)
 - v3.1.0-rc2-64-gf3d460ff5ae6
 - ptp4l version 3.1.1

**Additional context**

erwango, 2022-05-25:
@alsrgv Would you mind having a look ? As you're the one that pushed https://github.com/zephyrproject-rtos/zephyr/pull/40187
Thanks

alsrgv, 2022-05-26:
I'm traveling next two weeks without access to STM dev kit, but I'll take a look week of 6/6.

gleichdick, 2022-05-30:
Thanks!
I think my fault was using ptp4l instead of [gptp](https://github.com/Avnu/gptp), and not that there is an issue with zephyr.
The sample works when plugging two STM32 together with an crossover cable as well as with a linux box running gptp.
Only ptp4l does not work.
Sorry for the noise

alsrgv, 2022-05-30:
I haven't been able to make ptp4l with gptp before as well. We have code for PTPv2 master and slave that uses ethernet transport (-2 option in ptp4l) that we're happy to publish.

erwango, 2022-05-30:
> Thanks! I think my fault was using ptp4l instead of [gptp](https://github.com/Avnu/gptp), and not that there is an issue with zephyr. The sample works when plugging two STM32 together with an crossover cable as well as with a linux box running gptp. Only ptp4l does not work. Sorry for the noise

@gleichdick Thanks for the heads up. Can you close the ticket when possible then ?

gleichdick, 2022-06-09:
@alsrgv That'd be nice if you could point me into the right direction regarding ptp4l.

closed 2022-06-09 by gleichdick

ypearson-bdai, 2024-08-01:
does PPS out work?