zephyrproject-rtos/zephyr #46032: Enforce C99 inttypes usage

stephanosio, 2022-05-26:
**Is your enhancement proposal related to a problem? Please describe.**

Zephyr currently overrides the toolchain default C99 integer types (`[u]intN_t`) to fixed types such that `int32_t` can always be processed without a size modifier, and `int64_t` with `ll` size modifier.

This "hack," introduced in #16645, has caused various toolchain compatibility issues (e.g. https://github.com/zephyrproject-rtos/zephyr/pull/37712#discussion_r689578180) and will not be sustainable as we start adding more non-GNU toolchains as supported toolchains for Zephyr -- note that `__INT32_TYPE__` and its friends defined by `zephyr_stdint.h` are non-standard and the toolchain default C library is not obliged to set up the type system using these.

**Describe the solution you'd like**

Stop re-defining the toolchain default types (i.e. remove `zephyr_stdint.h`) and always use the C99 inttypes macros with the C99 integer types.

For this, the following changes need to be implemented:

- [ ] Rework the minimal libc `inttypes.h` to not assume the fixed types defined by `zephyr_stdint.h`; the size modifiers need to be resolved based on the native toolchain types.
- [ ] Fix all discrete size modifier used with the C99 integer types to use the C99 inttypes macros.
- [ ] Remove `zephyr_stdint.h`.

**Additional context**

It may be useful to go over the following related issues:

* https://github.com/zephyrproject-rtos/zephyr/issues/37718
* https://github.com/zephyrproject-rtos/zephyr/issues/44199
* https://github.com/zephyrproject-rtos/zephyr/issues/45336

stephanosio, 2022-05-26:
cc @npitre @dcpleung @carlescufi

andyross, 2022-05-26:
Just to record for posterity: I hate the ISO `PRI*` macro nonsense with a fiery passion.  It's ridiculous-looking and hard to read, it takes a perfectly working API with perfectly working error reporting and introduces line noise syntax **not** for correctness[1], but for portability.  And that's just the Wrong Thing.  Let code manage the build problems that result, if there are platform-dependent sizes in printf format strings then it's the programmer's job to make sure they aren't used[2].

I won't -1 this.  I won't stand in the way of a merge.  But this is just awful.  We're now committing to explaining to generations of new developers why Zephyr's printf and logging layers are awful.  And they're all going to ask "Why doesn't Linux have this problem?!".  What's our answer going to be?

[1] There's no typesafety problem with passing e.g. a "int64_t" to a "%ld" on arm64, for example.  The compiler knows the types, it knows how to pass a 64 bit argument, it knows how big a "long" is, and it can trivially (and does!) emit a printf format error on i386 builds.  Why is that a problem?  If the user wrote non-portable code they should fix it.  They don't need new (and bad!) syntax in the language to help with that.

[2] e.g. By using `%lld` in the first place to ensure the full 64 bit value is printed, or by deliberately downconverting a value to 32 bits, etc...  Those decisions are **application** decisions!  The printf library can't make them for us, and it shouldn't try.

stephanosio, 2022-05-26:
> "But why doesn't Linux need this?" :)

Because they use their own type system? (`u32`, ...)

We can go back to doing that again; but, as long as we are using the C99 integer types, we will have to use the corresponding C99 integer type format specifiers too.

npitre, 2022-05-26:
> as long as we are using the C99 integer types, we will have to use the 
> corresponding C99 integer type format specifiers too.

C99 integer types aren't the problem.  The basic types to which they're 
mapped is. Make int32_t onto an int and int64_t onto a long long at the 
toolchain level and you'll be fine. That's trivial to accomplish there. 
The alternative is way more painful and far more work for everyone.

andyross, 2022-05-26:
Or we can commit a silly hack that we've been living with just fine.  I mean, I get that clang picked some different defaults (or whatever, I'm no expert on the linked bug), but we already have a toolchain abstraction where we could synchronize this.

It's **far** simpler for us to have a layer that manages some platform-specific typedefs than it is to inflict this weird non-C/non-printf alien syntax on app developers.

That's the core of the problem here.  This PRI nonsense just "isn't C code".  It's a mistake.  The printf() API has a four decade tradition of hackers (some of them even older than me!) who understand it in their bones.  It's not the only way to format strings, but it's the C way.

Or consider it this way: if we really did want to change our string formatting syntax, why on earth would we pick THIS one?  Why not move to something that looked more like Rust std::fmt, for example, which I think is quite nice personally?  I mean, your answer is going to be "because that's rust and not C", right?  Well... this isn't C either.  No one uses these things in practical code, they're a wart to paper over a mistake in the standard.

stephanosio, 2022-05-26:
> The basic types to which they're mapped is

Yes, and that is allowed by the C standard -- that is the real problem. As long as that is allowed, we will have to use the proper abstraction provided by the standard.

> Make int32_t onto an int and int64_t onto a long long at the
toolchain level and you'll be fine. That's trivial to accomplish there.

For the GNU toolchains and maybe Clang, yes; for others like Arm Compiler, IAR, and whatever we add support for in the future, likely not.

stephanosio, 2022-05-26:
> Or we can commit a silly hack that we've been living with just fine.

I have been saying we cannot do that as I noted above "`__INT32_TYPE__` and its friends defined by zephyr_stdint.h are non-standard and the toolchain default C library is not obliged to set up the type system using these."

> It's far simpler for us to have a layer that manages some platform-specific typedefs

For GCC, yes; for others, not really.

> if we really did want to change our string formatting syntax, why on earth would we pick THIS one?

Because it is a part of the C99 standard. As I said above, "it is as simple as, if you want to use C99 integer types (int32_t, ...), you will have to use the corresponding C99 integer type format specifier macros (PRId32, ...) too."

stephanosio, 2022-05-26:
If you are really unhappy with this, we can consider going back to the days of `u32_t` and its friends, but we have had plenty of problems with that too.

I will emphasise again: "it is as simple as, if you want to use C99 integer types (int32_t, ...), you will have to use the corresponding C99 integer type format specifier macros (PRId32, ...) too."

andyross, 2022-05-26:
> I have been saying we cannot do that as I noted above "__INT32_TYPE__ and its friends defined by zephyr_stdint.h are non-standard and the toolchain default C library is not obliged to set up the type system using these."

Right, so just make our headers produce the same types as the toolchain/platform expects (again, we already have a framework for doing this).  That fixes the build errors, and leaves the portability issue as a task for the developer, which is where it should be anyway.

npitre, 2022-05-26:
> > Make int32_t onto an int and int64_t onto a long long at the toolchain level and you'll be fine. That's trivial to accomplish there.

> For the GNU toolchains and maybe Clang, yes; for others like Arm Compiler, IAR, and whatever we add support for in the future, likely not.

Do we really have to support all compilers under the sun?

If so, shouldn't we try to ask them to create a "Zephyr" mode first?

This is typically defined in a single header file therefore very easy to do.

stephanosio, 2022-05-26:
> Right, so just make our headers produce the same types as the toolchain/platform expects

That may not be feasible for all compilers. Some compilers may not even have something like `__INT32_TYPE__` which we can use to override the compiler-default C library-defined C99 fixed-width types.

> Do we really have to support all compilers under the sun?

Maybe not "all compilers," but we are committed to supporting multiple toolchains.

> If so, shouldn't we try to ask them to create a "Zephyr" mode first?

Ideally, yes; but, not everything is ideal in real life ...

stephanosio, 2022-05-26:
As a side note, if we, as project, want to introduce a non-standard string formatting syntax (and that is probably warranted, if I may add), we can; but, that is a whole another issue and does not change the fact that the use of standard C functions need to comply with the C standard.

stephanosio, 2022-05-26:
> Having int32_t sometimes map to an int, sometimes to a long is a
complete abomination.

That is indeed an abomination. But for `int64_t`, it may very well be `long` or `long long` depending on whom you ask.

npitre, 2022-05-26:
> As a side note, if we, as project, want to introduce a non-standard string formatting syntax (and that is probably warranted, if I may add), we can; but, that is a whole another issue and does not change the fact that the use of standard C functions need to comply with the C standard.

It is possible for the toolchain to be stricter in its type association 
and still be within the C standard.

stephanosio, 2022-05-26:
> It is possible for the toolchain to be stricter in its type association and still be within the C standard.

I am inclined to argue that something like below is not C99-compliant and is indeed wrong:

```
int32_t val;

printf("val = %d", val);        /* Incorrect */
```

`int32_t` is only meant to be printed out using the corresponding C99 format specifier macro, which is `PRI{d,i,o}32` and that is how it ought to be in order to be fully portable.

At least, for `printf`, which is a standard C function, I strongly believe that is how it needs to be handled.

For other similar non-standard/Zephyr-specific functions like `printk` and `LOG_*`, we could potentially introduce custom format specifiers like `%I32d` or somewhere along that line to avoid the "`PRIxN` abominations."

keith-packard, 2022-05-26:
(I don't think anyone *likes* the C99 format specifiers; I wish they had just added more size prefixes to the printf spec instead)

Does zephyr support any platforms for which

 * sizeof(int) != 4
 * sizeof(long long) != 8

If not, then we could use %d for int32_t and %lld for int64_t. Unfortunately, this can *still* generate warnings on platforms for which sizeof(long) == 4 and int32_t is a typedef for long (sigh), but you could stick a cast to (int) for int32_t in the printf call. Not great.

It would be good to evaluate where we could be using size_t and ptrdiff_t instead of explicit bit sizes as those can be represented with %zd and %td in printf fomat strings.

Alternatively, cast everything to long long and use %lld everywhere. Easier to read than "%"PRId32"\n". That does depend on printf supporting 64-bit integers (which is optional on 32-bit platforms).

One significant benefit for using the right format specifiers is that we can use the existing code in gcc and clang which evaluate printf format strings and make sure the arguments are of the right type. I think this is worth a small amount of pain

npitre, 2022-05-26:
@keith-packard wrote:

> Does zephyr support any platforms for which
> 
> • sizeof(int) != 4
> • sizeof(long long) != 8

No.

> If not, then we could use %d for int32_t and %lld for int64_t.

That's what we do already.

Furthermore, there are 32-bit architectures that mapped int32_t to a long
or intptr_t to an int. For those the following commit was created:

https://github.com/zephyrproject-rtos/zephyr/commit/f32330b22c33

Some people like to call it a hack, but one that has worked remarcably
well so far. I prefer to call it sanity enforcement.

Then there are compilers other than gcc or clang that people want to use with
creative type mappings that conflict with what we want to enforce here. But
hey, they are in their own right to do so per the C standard.

stephanosio, 2022-05-27:
> Alternatively, cast everything to long long and use %lld everywhere. Easier to read than "%"PRId32"\n". That does depend on printf supporting 64-bit integers (which is optional on 32-bit platforms).

For C99 and on, we can cast everything to `intmax_t` and use `%jd`; but, these casts are ugly and also there are footprint and performance implications of doing so on resource constrained embedded targets, so we are probably better off using the `PRIxN` macros in this case.

> Some people like to call it a hack, but one that has worked remarcably
well so far. 

I beg to differ. I have been fixing various issues around this until now -- some fixable and others not.

Let us try to stick with what the standard says should be done for the standard functions at least. I do agree that `PRIxN` macros are syntactically horrible to use and, as I briefly mentioned above, we should probably invent our own format specifier to replace them for Zephyr-specific functions.

> For other similar non-standard/Zephyr-specific functions like `printk` and `LOG_*`, we could potentially introduce custom format specifiers like `%I32d` or somewhere along that line to avoid the "`PRIxN` abominations."

For example, `%I32x` for `[u]int32_t`, `%I64x` for `[u]int64_t` and so on.

**Ultimately, we can stop using the standard C functions like `printf` in our codebase to avoid the `PRIxN` macro hell, and replace them with `[f,vf,sn,vsn]printfcb` from the cbprintf core OS library, which then can be updated to support the non-standard C99 fixed-width integer type specifiers like `%I32x`.** However, for the standard C functions, we should stick with what the standard recommends for the sake of portability.

stephanosio, 2022-05-27:
@nordic-krch FYI

stephanosio, 2022-05-27:
I have renamed the issue to reflect the scope of this proposal more clearly:

> Enforce C99 inttypes usage **for standard C functions**

npitre, 2022-05-27:
> For other similar non-standard/Zephyr-specific functions like `printk` and
> `LOG_*`, we could potentially introduce custom format specifiers like `%I32d`
> or somewhere along that line to avoid the "`PRIxN` abominations."

Can this scheme work with the format attribute for the compiler to
validate the argument type against the format string? That's an
important tool, and the warnings it generates is actually the very
reason why we are overriding integer types in the first place.

If "%I32" cannot be validated at compile time with the format attribute
then we could just drop the __printf_like from printk() and friends, and
still continue to use "%d" for an int32_t and "%lld" for an int64_t.
It won't matter at that point if int32_t is mapped to a long if an int
and a long are the same size.

Or, wording this differently:

- Keep the current integer type override for those targets that do
accommodate it. They will be the format validating targets.

- Those targets that can't do that, or do conflict (e.g. native_posix)
or use of external C++ libs with incompatible type mangling, then drop
the override and make __printf_like a no-op.

- This could even be made into a Kconfig option ("Enforce Zephyr integer
type convention" vs "Use native toolchain integer types (with caviats)".
The main caviat being that no string format would be validated at
compile time for the later. But that's still better than "%I32" which
would never be validated.

To be extra sure, we could use something like:

```
BUILD_ASSERT(sizeof(int32_t) == sizeof(int));
BUILD_ASSERT(sizeof(int64_t) == sizeof(long long));
BUILD_ASSERT(sizeof(void *) == sizeof(long));
```

As long as this is true we'll be fine with the above scheme.

What do you think?

keith-packard, 2022-05-27:
I think all this would require is explicit (int) casts for all int32_t printf parameters and things would work fine. Otherwise, silly systems where sizeof(int) == sizeof(long) but which typedef int32_t as long instead of int will generate compiler warnings. We could also create a zephyr/stdint.h which #include's stdint.h and then makes sure int32_t is 'int' and not 'long'.

stephanosio, 2022-05-27:
> Can this scheme work with the format attribute for the compiler to
validate the argument type against the format string?

Obviously, the format checking for `%Nx` is not going to work in the vanilla GCC because it is not standard and GCC supports no such thing, but we can relatively easily add support for these in the Zephyr GCC fork used to build the Zephyr SDK.

The parameter checking is handled here:
https://github.com/gcc-mirror/gcc/blob/b2cb88fb45a68fc91bda1c69d9486e824c4a559e/gcc/c-family/c-format.cc#L4065-L4070

> If "%I32" cannot be validated at compile time with the format attribute
then we could just drop the __printf_like from printk() and friends, and
still continue to use "%d" for an int32_t and "%lld" for an int64_t.
It won't matter at that point if int32_t is mapped to a long if an int
and a long are the same size.

We will definitely need to remove `__printf_like` (aka. `__attribute__((format (printf, f, a)))`) from non-standard functions like `printk` because they are not really standard C `printf`; instead, we will need to introduce something like `__attribute__((format (cbprintf, f, a)))` that properly handles the (future) cbprintf-specific format checks such as `%I32`.

Using `%d` for `int32_t` and `%lld` for `int64_t` is an abomination in my opinion because:

1. `%d` represents `int`, and `int32_t` is not guaranteed to be `int` as per the C standard.
2. `%lld` represents `long long int`, and `int64_t` is not guaranteed to be `long long` as per the C standard.

Hence, the rationale for introducing the proper format specifiers/abstractions for printing the C99 fixed-width integer types:

* `PRIxN` macros for standard C library functions
* `%INx` (e.g. `%I32d`) format specifiers for Zephyr-specific print functions (i.e. cbprintf)

stephanosio, 2022-05-27:
> I think all this would require is explicit (int) casts for all int32_t printf parameters and things would work fine. Otherwise, silly systems where sizeof(int) == sizeof(long) but which typedef int32_t as long instead of int will generate compiler warnings.

That would be the simplest solution indeed, but that cast sounds like as much of an abomination as the `PRIxN` macro to me; I would like to hear what others think of this approach.

> We could also create a zephyr/stdint.h which #include's stdint.h and then makes sure int32_t is 'int' and not 'long'.

That is sort of what we are doing at the moment:
https://github.com/zephyrproject-rtos/zephyr/blob/735ddeb4a8e26c6ce07d6aa5a0b77fffc559b2a8/include/zephyr/toolchain/zephyr_stdint.h#L23-L47

`zephyr_stdint.h` gets globally included using `-imacros`:
https://github.com/zephyrproject-rtos/zephyr/blob/3122a2c8b885b5f0f033898df09a7fe7e424ef37/CMakeLists.txt#L286-L291

stephanosio, 2022-05-27:
Just as a reminder, the goal of this proposal is to ensure portability.

If we are only going to support GCC and Clang, we might as well keep the status quo; but, that is not the direction we are headed unfortunately (at least, from the TSC/Toolchain WG perspective) and we need C standard conformance in order to fundamentally ensure the compatibility with other compilers.

keith-packard, 2022-05-27:
hrm. I don't see a huge conflict here with creating a zephyr_stdint.h that insists that int32_t be 'int' and not 'long' (for ILP32 ABIs). Code which uses zephyr_stdint.h can use %d while code that uses bare stdint.h would need to use PRId32. That is about the only difference visible to applications.

keith-packard, 2022-05-27:
> That is sort of what we are doing at the moment:

That's a much bigger hammer than I think we need -- all we really need is for int32_t to be 'int' and not 'long' so that printf can use %d instead of %ld. Everything else can use the regular stdint.h defines, I think.

stephanosio, 2022-05-27:
> Code which uses zephyr_stdint.h can use %d while code that uses bare stdint.h would need to use PRId32.

The problem is that they are the same code.

keith-packard, 2022-05-27:
> We will definitely need to remove `__printf_like` (aka. `__attribute__((format (printf, f, a)))`) from non-standard functions like `printk` because they are not really standard C `printf`; 

I'm hoping to see the duplicate format scanning code in cbprintf and printk replaced by calls to printf. Having a well-tested and POSIX-compliant formatted output implementation seems like a good way to save code, developer pain and long-term support costs.   [lib/os: Use picolibc printf as a replacement for printk](https://github.com/zephyrproject-rtos/zephyr/pull/44096/commits/10d6ce57f65c55fda7b6701ff5458b80317f286f)

npitre, 2022-05-27:
> I'm hoping to see the duplicate format scanning code in cbprintf and printk replaced by calls to printf.

Won't happen.

Just look at the binary size inflation you'll get from standard libc printf. 
Don't forget that size matters here. Otherwise you may as well just use Linux.

npitre, 2022-05-27:
> Just as a reminder, the goal of this proposal is to ensure portability.

Sure.

> 1. `%d` represents `int`, and `int32_t` is not guaranteed to be `int` as per the C standard.

> 2. `%lld` represents `long long int`, and `int64_t` is not guaranteed to be `long long` as per the C standard.

Indeed. And in practice they aren't all the time either. Hence the override.

But if they're the same _size_ then it doesn't matter that much. And in 
practice they always are so far.

I'll experiment with the idea I just proposed and report back. That 
should provide both portability and very minimal code change. If that 
doesn't work then very little would be wasted.

stephanosio, 2022-05-27:
> I'm hoping to see the duplicate format scanning code in cbprintf and printk replaced by calls to printf. Having a well-tested and POSIX-compliant formatted output implementation seems like a good way to save code, developer pain and long-term support costs.

@keith-packard Since we have not evaluated the picolibc `printf` in comparison to the Zephyr `cbprintf`, it might be worth looking into. Let us continue this discussion in the picolibc PR later (after 3.1 release). FYI, the rationale behind the cbprintf is described in https://github.com/zephyrproject-rtos/zephyr/commit/33103828dc2f48aecda932f38f4f4198213f7efc.

stephanosio, 2022-05-27:
> But if they're the same _size_ then it doesn't matter that much.

Sure, but that leaves a bad taste in my mouth.

> And in practice they always are so far.

Fortunately (I guess ...), Zephyr does not support any 16-bit targets, in which 16-bit `int` is very common.

Just to be pedantic, if we ever need to support a 32-bit target with I16LP32 data model (not unheard of), this will be problematic -- though, given the current trend, that is probably not going to happen in the foreseeable future (but who knows, some shiny new insanely-optimised-for-embedded RISC-V target might just do that).

Also C standard defines `long long` to be _at least_ 64-bit, meaning there is no reason it cannot be a 128-bit type and that is, in fact, a natural choice for the targets that natively support 128-bit data types.

GCC even says ["There is no support in GCC for expressing an integer constant of type __int128 for targets with long long integer less than 128 bits wide."](https://gcc.gnu.org/onlinedocs/gcc/_005f_005fint128.html), implying `long long` may very well be a 128-bit data type -- who says the RISC-V RV128 will not adopt a data model that defines `long long` as a 128-bit data type?

**Conforming to the C standard naturally prepares our codebase for such (although currently obscure, may be feasible in the future) targets, whereas the current `zephyr_stdint.h` hack is very fragile and may need a complete overhaul at some point in the future when such a target needs to be supported.**

Another major issue with the current `zephyr_stdint.h` hack, besides what I have already explicitly stated above, is that it creates problems for linking the libraries that are compiled outside the Zephyr build system (e.g. pre-built libraries provided by a third-party, independent components that have their own build system).

While that is not so much of a problem for C (because C does not mangle identifiers based on types), this is a very real problem for C++ as reported in https://github.com/zephyrproject-rtos/zephyr/issues/45248 -- for instance, if you have a function in an external library that accepts `int32_t` argument, where the toolchain default type for `int32_t` is `long`, the header that gets processed on the Zephyr side will expect the function symbol to be mangled for `int` argument, whereas the actual mangled symbol in the library will be for `long` argument.

Moreover, C may not be completely unaffected by this either -- who says there will not be some obscure architecture-specific ABI handling convention that passes `int` and `long`, or `long` and `long long` data differently despite being the same size? 

For the reasons stated above, I do not think the current `zephyr_stdint.h` hack is a sustainable solution let alone a portable one. Yes, some of the cases mentioned above are hypothetical, but as an "operating system" project, we need to be prepared for those.

npitre, 2022-05-27:
> Just to be pedantic, if we ever need to support a 32-bit target with
> I16LP32 data model (not unheard of), this will be problematic --
> though, given the current trend, that is probably not going to [...]

It'd better preserve the ilp32 model to minimize the pain on developers.
Especially since a short is already widely used as a 16-bit quantity.
On such architectures it is often the char type becoming 16 bits.

In any case, let's not over-engineer things unless necessary. Designing
for hypothetical cases often turns out to be wasted efforts if such
cases never materialize and the added cost will remain on us forever.

I do recognize the c++ mangling issue though. But a simple solution
might work fine.

keith-packard, 2022-05-27:
> > Code which uses zephyr_stdint.h can use %d while code that uses bare stdint.h would need to use PRId32.
> 
> The problem is that they are the same code.

I'm not sure you understood my suggestion. The *only* issue with using %d for int32_t is on ILP32 platforms where the typedef for int32_t is 'long' instead of 'int' in that it will generate compiler warnings. If the file using %d with int32_t values had used zephyr_stdint instead of stdint, then the int32_t typedef would be mapped to 'int' and no warning would be generated. 

Pulling in existing code which uses stdint.h which then uses int32_t values to these functions would require the use of PRId32 to avoid compiler warnings in this ILP32-with-long-int32_t environment.

npitre, 2022-05-27:
> Fortunately, the printf code in picolibc is smaller than cbprintf or printk, so you save even more space if applications use a mixture of APIs.

Note that we already funnel various APIs through the cbprintf 
implementation, so no duplication.

And then we have 2 such implementations: the full fledged and a tiny 
barebone one.

Duplication comes about only if an external stdio library is used.

nordic-krch, 2022-05-30:
as for logging/printk. What about utilizing `_Generic` to cast to proper type that match standard format specifier. If `_Generic` is not supported but there are exotic types,  `__printf_like` can removed using ifdefed.

cfriedt, 2022-05-30:
I would +1 removing `zephyr_stdint.h` in favour of that being the toolchain's responsibility.

Not crazy about using `%" PRId32 "` instead of `%d` everywhere though just to print an `int`. Also would prefer to not *always* need to specify `int32_t` rather than `int`.

I find `PRIu64`, `PRIx64`, and `PRId64` mainly helpful as a common way to print 64-bit values on both 32 and 64-bit machines.

npitre, 2022-10-11:
> I believe this is an inevitable decision for us.

> While I am not a very big fan of this either, this is the only way to be portable across different compilers and we have no option but to do this, especially as we start introducing more non-GNU compiler support to the Zephyr.

Why do we have to bow to them?

Zephyr is becoming the most important RTOS project out there. That 
should be sufficient incentive for those other compilers to create a 
config with sane defaults for Zephyr.

stephanosio, 2022-10-11:
https://github.com/zephyrproject-rtos/zephyr/pull/46708#issuecomment-1162701856

> **TL;DR: If you want to use `int32_t` and its friends in the `printf`, use `PRId32` and its friends too. Do not use `int32_t` in the `printf` and drop the accompanying `PRId32` because that is going to create a big mess.**

> I too agree with you that the `PRIxN` macros in the `printf` are horrible and we do not want to live with this wart in the Zephyr codebase. **So let us drop it all. Let us switch over to using the `printk` and the Zephyr-native cbprintf API** with the alternative C99 fixed-width integer type format specifiers like `%I32d` for `int32_t` as suggested in https://github.com/zephyrproject-rtos/zephyr/issues/46032#issuecomment-1139228035.

keith-packard, 2022-10-11:
> I too agree with you that the `PRIxN` macros in the `printf` are horrible and we do not want to live with this wart in the Zephyr codebase. **So let us drop it all. Let us switch over to using the `printk` and the Zephyr-native cbprintf API** with the alternative C99 fixed-width integer type format specifiers like `%I32d` for `int32_t` as suggested in [#46032 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/46032#issuecomment-1139228035).

I think I slightly disagree with this. The (significant) benefit of using printf is that the C compiler will typecheck arguments against the format strings. Any printf-like-function which doesn't match the printf formatting conventions cannot use the compiler's printf typechecking, so you lose that benefit. Even if you call it `printk`, or `cbprintf`, I'd still encourage you to consider the checking benefits vs the convenience of simpler usage.

If you don't like `PRId32` (and, frankly, who does), you could use `%ld` and insert a `(long)` cast for the argument. On 32-bit systems using ILP32 and 64-bit systems using LP64, this is zero cost -- `long` is the native machine word size in both cases.

Using the standards also lowers the barrier to entry for new Zephyr developers; they can leverage knowledge gained in other environments without having to constantly second-guess where Zephyr differs. There are other benefits, like being able to validate the implementation by using existing test suites and using printf to implement printk and cbprintf (which Zephyr already does when using picolibc).

jgl-meta, 2023-04-26:
@stephanosio What is the status of this? Will this make it into the 3.4 release?

keith-packard, 2023-04-26:
Looking forward to c23, we should expect to have %w format specifiers in the standard libraries at some point soon, which means we can use %w32d instead of PRId32.

marc-hb, 2023-04-26:
> Why do we have to bow to them?
>
> Zephyr is becoming the most important RTOS project out there. That
> should be sufficient incentive for those other compilers to create a
> config with sane defaults for Zephyr.

This does not seem realistic. Some of Linus Torvalds' complaints about gcc have fallen on deaf ears. Kernels are a small "market" for compilers.

Also, Linux supports just one toolchain (+ clang, a... gcc emulator on top of LLVM) whereas Zephyr bends over backwards to support unknown and proprietary toolchains thanks to an elaborate CMake indirection layer. Zephyr also has a licensing model welcoming closed source libraries and even a tool for binary blobs. Expecting to rally all these in the crusade against PRIu64 is not realistic.

> In any case, let's not over-engineer things unless necessary. 

Everyone agrees PRIu64 is incredibly ugly and a huge mistake. Probably for backwards compatibility, always a huge problem with C/C++. But why is it **necessary** to (over)engineer a solution that avoids PRIu64 at all costs? Code style and readability matter, a lot. But "necessary" means something different.

If there is a simple and reliable solution that avoids PRIu64 for C and C++ across a reasonable range of unmodified toolchains then perfect; "send patches!"  However @stephanosio's experience and expertise + the length of this thread and of a good few other  before make me feel very sceptical. Speaking of patches, does Zephyr already have some build-time test case(s) for this?

> Looking forward to c23, we should expect to have %w format specifiers in the standard libraries at some point soon, which means we can use %w32d instead of PRId32.

Great news! So the compilers "bowed" after all. Now we can get rid of PRIu64 around 2030 ;-)

stephanosio, 2023-04-27:
> @stephanosio What is the status of this? Will this make it into the 3.4 release?

@jgl-meta This is still under discussion and is unlikely to make into the 3.4 release.

> Looking forward to c23, we should expect to have %w format specifiers in the standard libraries at some point soon, which means we can use %w32d instead of PRId32.

@keith-packard That is a great news; but, using those format specifiers will require us to set the minimum required C standard version for Zephyr to C23, which I do not think is going to happen in the foreseeable future.

> Great news! So the compilers "bowed" after all. Now we can get rid of PRIu64 around 2030 ;-)

@marc-hb Given that [we are still debating whether it is reasonable to _require_ C11](https://github.com/zephyrproject-rtos/zephyr/issues/30105) in 2023, maybe 2040 is more realistic :)

stephanosio, 2023-04-27:
Another option could be to add the C23 `%w` format specifier support to our `printk` and `cbprintf` libraries.

As long as we use the `%w` format specifier with only `printk` and `cbprintf` (i.e. not in the C standard library functions such as `printf`), we do not need set the minimum required C standard version to C23.

This will effectively allow us to not use the wart that `PRIxN` is in the Zephyr codebase.

cfriedt, 2023-04-27:
> @marc-hb Given that [we are still debating whether it is reasonable to _require_ C11](https://github.com/zephyrproject-rtos/zephyr/issues/30105) in 2023, maybe 2040 is more realistic :)
🥲

keith-packard, 2026-02-20:
> I see both approaches as complementary: yours reduces how often the override is needed, mine gives a way out when it gets in the way.

Sure, yours provides a solution for people using toolchains they can't fix and using headers they can't fix. I'm glad I will probably never be in that situation :-)