zephyrproject-rtos/zephyr #46034: subsys settings: should check the return value of function cs->cs_itf->csi_load(cs, &arg).

CV-Bowen, 2022-05-26:
**Is your enhancement proposal related to a problem? Please describe.**
I see the return value of function `cs->cs_itf->csi_load(cs, &arg)` was ignored in function `settings_load_subtree()` and `settings_load_subtree_direct()`, i think the return value should be processed instead ignored.

**Describe the solution you'd like**
If the `csi_load()` failed, a error number should be returned.

Laczen, 2022-05-28:
> **Is your enhancement proposal related to a problem? Please describe.** I see the return value of function `cs->cs_itf->csi_load(cs, &arg)` was ignored in function `settings_load_subtree()` and [...]

Hi @CV-Bowen, the problem is that `csi_load` is returning multiple entries, and one of them fails. Should the routine then stop, or continue ?

CV-Bowen, 2022-05-30:
> Hi @CV-Bowen, the problem is that `csi_load` is returning multiple entries, and one of them fails. Should the routine then stop, or continue ?

I think the routine should stop immediately and return the error code once `csi_load()` failed.

Laczen, 2022-05-30:
@CV-Bowen, stopping immediately doesn't seem a good idea. The error code from `csi_load()` would be the result of the set handler. The application already knows that there is some problem with loading a specific entry (in the set handler), why should it stop loading other entries.

Laczen, 2022-05-30:
> @CV-Bowen, stopping immediately doesn't seem a good idea. The error code from `csi_load()` would be the result of the set handler. The application already knows that there is some problem with loading a specific entry (in the set handler), why should it stop loading other entries?

CV-Bowen, 2022-05-30:
> @CV-Bowen, stopping immediately doesn't seem a good idea. The error code from `csi_load()` would be the result of the set handler. The application already knows that there is some problem with loading a specific entry (in the set handler), why should it stop loading other entries?

Do you mean application has known the error code in the callback function (`struct settings_load_arg: cb`) called in `settings_call_set_handler()` ?

CV-Bowen, 2022-05-30:
OK, i understood, thank you for detailed explanation! I will close this issue.

closed 2022-05-30 by CV-Bowen