zephyrproject-rtos/zephyr #46044: ESP32: Web USB support request

enthunilu, 2022-05-26:
**Is your enhancement proposal related to a problem? Please describe.**
I was trying to build webusb sample code for my esp32 dev kit.
https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/subsys/usb/webusb

I was using below command,
west build -p auto -b esp_wrover_kit samples/subsys/usb/webusb

while execution I am getting linking error for usb API's as mentioned in below ticket
https://github.com/zephyrproject-rtos/zephyr/issues/45933

Steps:

Open command prompt
Go to zephyr directory
execute: west build -p auto -b esp_wrover_kit samples/subsys/usb/webusb
Check logs on console

**Describe the solution you'd like**
USB support should be there for ESP32 device. 

**Describe alternatives you've considered**
I am unable to find any alternative at this moment.

sylvioalves, 2022-05-26:
@enthunilu This sample requires some USB support to work. However, there is no support for ESP32. In fact, ESP32 has no USB port. So you would need to work with ESP32SS/S3/C3 when/if USB support is implemented.

sylvioalves, 2022-05-26:
Sorry, I should not have closed this.

rftafas, 2022-05-27:
Just to note that ESP32-C3 contains only an USB to Serial converter. It is not a general USB peripheral/host controller and the RTOS running on C3 should expect it as another serial channel.

ddavidebor, 2024-04-13:
+1 from me: please add USB support to Zephyr (new USB Stack) in particular regarding USB HS support on ESP32-P4