zephyrproject-rtos/zephyr #46049: Usage faults on semaphore usage in driver (stm32l1)

sappigeninja, 2022-05-26:
Me and my team of students are working on a driver for an Ublox SARA N310.
We are planning on merging this driver into Zephyr when the driver is complete.

We currently have a critical issue where a semaphore causes a usage fault deep into Zephyr itself.
When we use a response semaphore for the `modem_cmd_send()` function it gives a usage fault.
We call this function via a wrapper function called `send_at_command()` its does nothing but forward the arguments to the `modem_cmd_send()` function.
The `send_at_command()` function gets used by a number of our public functions defined in drivers header file.

The catch is that the function does not fail on the first few calls.
It fails after a set amount of calls to the function in rapid succession.
The amount can vary when I added a new public function that used `send_at_command()` function the problem shifted a few functions down (my guess is that something is going wrong in the stack).
On some functions it always fails on the semaphore and gives a usage fault.
Sometimes when a function is called twice it will fail.
With a usage fault on the second call.
Different public functions give different usage faults.

The problem does not occur when slowly stepping through the code with a debugger.
I included the result of a `thread apply all bt` in a file for review.
The problem happens in `modem_cmd_send_ext()` on the second call to `k_sem_take()` (line 527).
The usage fault is invoked.

`thread apply all bt`
```
(gdb) thread apply all bt
Info : Getting thread 536874392 reg list

Thread 6 (Thread 536874392):
#0  arch_swap (key=key@entry=0) at /home/hackerman/Zephyr/zephyrproject/zephyr/arch/arm/core/aarch32/swap.c:53
#1  0x08009874 in z_swap_irqlock (key=0) at /home/hackerman/Zephyr/zephyrproject/zephyr/kernel/include/kswap.h:196
#2  z_swap (key=..., lock=0x200015ec <pending_cancels>) at /home/hackerman/Zephyr/zephyrproject/zephyr/kernel/include/kswap.h:196
#3  z_pend_curr (lock=lock@entry=0x200015ec <pending_cancels>, key=..., wait_q=wait_q@entry=0x20000b28 <mdata+832>, timeout=...) at /home/hackerman/Zephyr/zephyrproject/zephyr/kernel/sched.c:815
#4  0x08009140 in z_impl_k_sem_take (sem=sem@entry=0x20000b28 <mdata+832>, timeout=...) at /home/hackerman/Zephyr/zephyrproject/zephyr/kernel/sem.c:143
#5  0x0800f238 in k_sem_take (timeout=..., sem=0x20000b28 <mdata+832>) at /home/hackerman/Zephyr/zephyrproject/zephyr/build/zephyr/include/generated/syscalls/kernel.h:1035
#6  modem_cmd_send_ext (iface=iface@entry=0x20000814 <mdata+44>, handler=handler@entry=0x20000824 <mdata+60>, handler_cmds=handler_cmds@entry=0x0, handler_cmds_len=handler_cmds_len@entry=0, buf=buf@entry=0x20002164 <z_main_stack+124> "AT+ULGASP=2,last gasp,0,,17,192.168.0.10:5050,,1,1,1", sem=sem@entry=0x20000b28 <mdata+832>, timeout=timeout@entry=..., flags=flags@entry=0) at /home/hackerman/Zephyr/zephyrproject/zephyr/drivers/modem/modem_cmd_handler.c:527
#7  0x0800f730 in modem_cmd_send (timeout=..., sem=0x20000b28 <mdata+832>, buf=0x20002164 <z_main_stack+124> "AT+ULGASP=2,last gasp,0,,17,192.168.0.10:5050,,1,1,1", handler_cmds_len=0, handler_cmds=0x0, handler=0x20000824 <mdata+60>, iface=0x20000814 <mdata+44>) at /home/hackerman/Zephyr/zephyrproject/zephyr/drivers/modem/modem_cmd_handler.h:226
#8  send_at_command (iface=iface@entry=0x20000814 <mdata+44>, handler=handler@entry=0x20000824 <mdata+60>, handler_cmds=handler_cmds@entry=0x0, handler_cmds_len=handler_cmds_len@entry=0, buf=buf@entry=0x20002164 <z_main_stack+124> "AT+ULGASP=2,last gasp,0,,17,192.168.0.10:5050,,1,1,1", sem=sem@entry=0x20000b28 <mdata+832>, timeout=..., tx_lock=tx_lock@entry=true) at /home/hackerman/Zephyr/zephyrproject/zephyr/drivers/modem/ublox-sara-n310.c:190
#9  0x080078d6 in n310_set_last_gasp (gpio_pin=gpio_pin@entry=16, text=text@entry=0x8010ce4 "Hello there!", ip_proto=ip_proto@entry=17, ip_addr=ip_addr@entry=0x8010cd0 "192.168.0.10:5050", tx_cnt=tx_cnt@entry=1, shutdown=shutdown@entry=1) at /home/hackerman/Zephyr/zephyrproject/zephyr/include/sys/time_units.h:405
#10 0x08000632 in main () at /home/hackerman/Zephyr/zephyrproject/zephyr/samples/drivers/ublox_sara_n310/src/main.c:59
Info : Getting thread 536874224 reg list

Thread 5 (Thread 536874224):
#0  arch_cpu_idle () at /home/hackerman/Zephyr/zephyrproject/zephyr/arch/arm/core/aarch32/cpu_idle.S:108
#1  0x0800fdc0 in k_cpu_idle () at /home/hackerman/Zephyr/zephyrproject/zephyr/include/kernel.h:5659
#2  idle (unused1=<optimized out>, unused2=<optimized out>, unused3=<optimized out>) at /home/hackerman/Zephyr/zephyrproject/zephyr/kernel/idle.c:86
#3  0x0800a8b0 in z_thread_entry (entry=0x800fdab <idle>, p1=<optimized out>, p2=<optimized out>, p3=<optimized out>) at /home/hackerman/Zephyr/zephyrproject/zephyr/lib/os/thread_entry.c:36
#4  0x74726174 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
Info : Getting thread 536872520 reg list

Thread 4 (Thread 536872520):
#0  cbvprintf (out=out@entry=0x800b53b <out_func>, ctx=0x8010f2c <log_output_uart>, fp=0x8010e77 "> ", fp@entry=0x1 "$", ap=...) at /home/hackerman/Zephyr/zephyrproject/zephyr/lib/os/cbprintf_complete.c:1717
#1  0x08001f1a in print_formatted (output=<optimized out>, fmt=0x8010e74 "<%s> ") at /home/hackerman/Zephyr/zephyrproject/zephyr/subsys/logging/log_output.c:144
#2  0x080020fa in ids_print (output=output@entry=0x8010f2c <log_output_uart>, level_on=level_on@entry=true, func_on=func_on@entry=false, domain_id=domain_id@entry=0, source_id=source_id@entry=2, level=level@entry=3) at /home/hackerman/Zephyr/zephyrproject/zephyr/subsys/logging/log_output.c:265
#3  0x08002188 in prefix_print (output=output@entry=0x8010f2c <log_output_uart>, flags=flags@entry=15, func_on=func_on@entry=false, timestamp=17911, level=level@entry=3 '\003', domain_id=0 '\000', source_id=2) at /home/hackerman/Zephyr/zephyrproject/zephyr/subsys/logging/log_output.c:541
#4  0x080021e6 in log_output_msg2_process (output=0x8010f2c <log_output_uart>, msg=0x20000510 <buf32+768>, flags=15) at /home/hackerman/Zephyr/zephyrproject/zephyr/subsys/logging/log_output.c:610
#5  0x08002362 in process (backend=<optimized out>, msg=0x20000510 <buf32+768>) at /home/hackerman/Zephyr/zephyrproject/zephyr/subsys/logging/log_backend_uart.c:109
#6  0x08001d10 in log_backend_msg2_process (msg=0x20000510 <buf32+768>, backend=0x8010b38 <log_backend_uart>) at /home/hackerman/Zephyr/zephyrproject/zephyr/include/logging/log_backend.h:135
#7  msg_process (msg=..., bypass=bypass@entry=false) at /home/hackerman/Zephyr/zephyrproject/zephyr/subsys/logging/log_core.c:776
#8  0x08001d6a in z_impl_log_process (bypass=bypass@entry=false) at /home/hackerman/Zephyr/zephyrproject/zephyr/subsys/logging/log_core.c:857
#9  0x08001ea8 in log_process (bypass=false) at /home/hackerman/Zephyr/zephyrproject/zephyr/build/zephyr/include/generated/syscalls/log_ctrl.h:56
--Type <RET> for more, q to quit, c to continue without paging--
#10 log_process_thread_func (dummy1=<optimized out>, dummy2=<optimized out>, dummy3=<optimized out>) at /home/hackerman/Zephyr/zephyrproject/zephyr/subsys/logging/log_core.c:1340
#11 0x0800a8b0 in z_thread_entry (entry=0x8001e95 <log_process_thread_func>, p1=<optimized out>, p2=<optimized out>, p3=<optimized out>) at /home/hackerman/Zephyr/zephyrproject/zephyr/lib/os/thread_entry.c:36
#12 0x08010bb0 in __devicehdl_DT_N_S_soc_S_pin_controller_40020000_S_gpio_40020000 ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
Info : Getting thread 536874568 reg list

Thread 3 (Thread 536874568):
#0  arch_swap (key=key@entry=0) at /home/hackerman/Zephyr/zephyrproject/zephyr/arch/arm/core/aarch32/swap.c:53
#1  0x08009874 in z_swap_irqlock (key=0) at /home/hackerman/Zephyr/zephyrproject/zephyr/kernel/include/kswap.h:196
[... 142 lines cut ...]
### debugging ###
CONFIG_LOG=y
CONFIG_LOG_STRDUP_BUF_COUNT=6

### needed to suppress "sys_rand_get" compiling error ###
CONFIG_TEST_RANDOM_GENERATOR=y

###### DEBUGGING TOOLS ######
CONFIG_DEBUG_THREAD_INFO=y
CONFIG_THREAD_NAME=y
CONFIG_DEBUG=y
#CONFIG_ASSERT=y
CONFIG_LOG=y
CONFIG_LOG_DEFAULT_LEVEL=4
CONFIG_GPIO_LOG_LEVEL_DBG=y
CONFIG_I2C_LOG_LEVEL_DBG=y
CONFIG_SENSOR_LOG_LEVEL_DBG=y
CONFIG_MODEM_LOG_LEVEL_DBG=y

# Try this
CONFIG_MODEM_IFACE_UART=y
CONFIG_EXTRA_EXCEPTION_INFO=y
```

On request I will send the driver code (or else this post might be a little to lengthy (its 2000 LoC), to properly read).

carlescufi, 2022-05-26:
Can you please double all stack sizes and try again? You can find all stacks by looking for `_STACK_SIZE` in `build/zephyr/.config`

alissakruiniger, 2022-05-30:
@carlescufi Thank you, this fixed our issue (am also part of the team). Is there a way to define the stack sizes in the prj.conf of a program?

sappigeninja, 2022-05-31:
After enabling some (KConfig) options for MPU protection.
We quickly got the following output after testing again.

```
[00:00:16.167,000] <err> os: r0/a1:  0x00000002  r1/a2:  0x20000c68  r2/a3:  0xf0f0f0f0
[00:00:16.167,000] <err> os: r3/a4:  0x20001f48 r12/ip:  0x00000003 r14/lr:  0x08006239
[00:00:16.167,000] <err> os:  xpsr:  0x41000036
[00:00:16.167,000] <err> os: EXC_RETURN: 0x0
[00:00:16.167,000] <err> os: Faulting instruction address (r15/pc): 0x080136dc
[00:00:16.167,000] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
[00:00:16.167,000] <err> os: Fault during interrupt handling
```

We are apparently getting a stack overflow somewhere.
What would be the best way to deal with this stack overflow?
We use a lot of strings for sending AT commands and those can get quite long.
There is no way for us to easily optimize and save more memory.

Is just doubling the stack size an acceptable solution (as we are planning on merging the driver into Zephyr, so correctness matters).

sappigeninja, 2022-06-07:
We doubled the stack sizes in the `prj.conf` and accepting this as our solution as it overflows if we do not double the `CONFIG_LOG_PROCESS_THREAD` then we get a stack overflow.
So its reasonable to double the stack sizes.

closed 2022-06-07 by sappigeninja