zephyrproject-rtos/zephyr #46066: mps2_an521_ns: Failed to trigger nmi_test_isr in "tests/arch/arm/arm_runtime_nmi/"

hunterhu3000, 2022-05-27:
**Describe the bug**
When testing the case of "tests/arch/arm/arm_runtime_nmi/", the nmi_test_isr should be triggered and the log "NMI received (test_handler_isr)! Rebooting...\n" should be seen on the terminal. The test looks good for all other boards(including mps2_an521) except mps2_an521_ns (run on QEMU).

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? ->  mps2_an521_ns with QEMU
 - What have you tried to diagnose or workaround this issue? 
 The background is that we found there is a bug in the test of "tests/arch/arm/arm_runtime_nmi/", which always reported "PASS" even if the test is actually "FAIL". When the test case is FAIL, the nmi_test_isr is not triggered and "NMI received (test_handler_isr)! Rebooting...\n" will not be seen. This is why we had never found this issue before. Now we have provide a PR https://github.com/zephyrproject-rtos/zephyr/pull/45998 to fix the "always PASS" issue. When the CI tests the mps2_an521_ns with QEMU, it failed and blocks the PR merge. 

**To Reproduce**
Steps to reproduce the behavior:
1. twister --device-testing --device-serial /dev/ttyUSB0 -vv -T tests/arch/arm/arm_runtime_nmi/ -p mps2_an521_ns
2. Check the console log and see if "NMI received (test_handler_isr)! Rebooting...\n" shows up.

**Expected behavior**
"NMI received (test_handler_isr)! Rebooting...\n" will be seen

**Impact**
mps2_an521_ns cannot trigger nmi isr correctly
blocks the merge of https://github.com/zephyrproject-rtos/zephyr/pull/45998

**Logs and console output**
```
ztest@ztest-NUC11PAHi5:~/zephyrproject/zephyr$ twister --device-testing --device-serial /dev/ttyUSB0 -vv -T tests/arch/arm/arm_runtime_nmi/ -p mps2_an521_ns
Renaming output directory to /home/ztest/zephyrproject/zephyr/twister-out.6
INFO    - Zephyr version: v3.1.0-rc2-137-g735ddeb4a8e2
INFO    - JOBS: 8
DEBUG   - Reading test case configuration files under /home/ztest/zephyrproject/zephyr/tests/arch/arm/arm_runtime_nmi...
DEBUG   - Found possible test case in /home/ztest/zephyrproject/zephyr/tests/arch/arm/arm_runtime_nmi
DEBUG   - Reading platform configuration files under /home/ztest/zephyrproject/zephyr/boards...
DEBUG   - Reading platform configuration files under /home/ztest/zephyrproject/zephyr/scripts/pylib/twister/boards...
DEBUG   - Running cmake script /home/ztest/zephyrproject/zephyr/cmake/modules/verify-toolchain.cmake
DEBUG   - Calling cmake with arguments: ['-DFORMAT=json', '-P', PosixPath('/home/ztest/zephyrproject/zephyr/cmake/modules/verify-toolchain.cmake')]
DEBUG   - Finished running  /home/ztest/zephyrproject/zephyr/cmake/modules/verify-toolchain.cmake
INFO    - Using 'zephyr' toolchain.
DEBUG   - platform filter: ['mps2_an521_ns']
DEBUG   -     arch_filter: None
DEBUG   -      tag_filter: None
DEBUG   -     exclude_tag: None
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /home/ztest/zephyrproject/zephyr/twister-out/testplan.json
DEBUG   - Running cmake script /home/ztest/zephyrproject/zephyr/cmake/modules/verify-toolchain.cmake
DEBUG   - Calling cmake with arguments: ['-DFORMAT=json', '-P', PosixPath('/home/ztest/zephyrproject/zephyr/cmake/modules/verify-toolchain.cmake')]
DEBUG   - Finished running  /home/ztest/zephyrproject/zephyr/cmake/modules/verify-toolchain.cmake
INFO    - Using 'zephyr' toolchain.

Device testing on:

| Platform      | ID   | Serial device   |
|---------------|------|-----------------|
| mps2_an521_ns |      | /dev/ttyUSB0    |

INFO    - 1 test scenarios (1 configurations) selected, 0 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
DEBUG   - adding mps2_an521_ns/tests/arch/arm/arm_runtime_nmi/arch.interrupt.arm.nmi
INFO    - Added initial list of jobs to queue
DEBUG   - Launch process 0
DEBUG   - Launch process 1
DEBUG   - Launch process 2
DEBUG   - Launch process 3
DEBUG   - Launch process 4
DEBUG   - Running cmake on /home/ztest/zephyrproject/zephyr/tests/arch/arm/arm_runtime_nmi for mps2_an521_ns
DEBUG   - Calling cmake with arguments: ['-B/home/ztest/zephyrproject/zephyr/twister-out/mps2_an521_ns/tests/arch/arm/arm_runtime_nmi/arch.interrupt.arm.nmi', '-S/home/ztest/zephyrproject/zephyr/tests/arch/arm/arm_runtime_nmi', '-DTC_RUNID=d4214fd5d62d74fefaee506bba5c4109', '-DEXTRA_CFLAGS=-Werror', '-DEXTRA_AFLAGS=-Werror -Wa,--fatal-warnings', '-DEXTRA_LDFLAGS=-Wl,--fatal-warnings', '-DEXTRA_GEN_DEFINES_ARGS=--edtlib-Werror', '-GUnix Makefiles', '-DQEMU_PIPE=/home/ztest/zephyrproject/zephyr/twister-out/mps2_an521_ns/tests/arch/arm/arm_runtime_nmi/arch.interrupt.arm.nmi/qemu-fifo', '-DBOARD=mps2_an521_ns']
DEBUG   - Launch process 5
DEBUG   - Launch process 6
DEBUG   - Launch process 7
DEBUG   - Finished building /home/ztest/zephyrproject/zephyr/tests/arch/arm/arm_runtime_nmi for mps2_an521_ns
DEBUG   - build test: mps2_an521_ns/tests/arch/arm/arm_runtime_nmi/arch.interrupt.arm.nmi
DEBUG   - Building /home/ztest/zephyrproject/zephyr/tests/arch/arm/arm_runtime_nmi for mps2_an521_ns
DEBUG   - run test: mps2_an521_ns/tests/arch/arm/arm_runtime_nmi/arch.interrupt.arm.nmi
[... 50 lines cut ...]
INFO    - In total 1 test cases were executed, 0 skipped on 1 out of total 456 platforms (0.22%)
INFO    - 1 test configurations executed on platforms, 0 test configurations were only built.

Hardware distribution summary:

| Board         | ID   |   Counter |
|---------------|------|-----------|
| mps2_an521_ns |      |         0 |
INFO    - Saving reports...
INFO    - Writing JSON report /home/ztest/zephyrproject/zephyr/twister-out/twister.json
DEBUG   - Running cmake script /home/ztest/zephyrproject/zephyr/cmake/modules/verify-toolchain.cmake
DEBUG   - Calling cmake with arguments: ['-DFORMAT=json', '-P', PosixPath('/home/ztest/zephyrproject/zephyr/cmake/modules/verify-toolchain.cmake')]
DEBUG   - Finished running  /home/ztest/zephyrproject/zephyr/cmake/modules/verify-toolchain.cmake
INFO    - Using 'zephyr' toolchain.
INFO    - Writing xunit report /home/ztest/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/ztest/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): Ubuntu 20.04
 - Toolchain (e.g Zephyr SDK, ...): 0.14.1
 - Commit SHA or Version used: 735ddeb4a8e26c6ce07d6aa5a0b77fffc559b2a8

**Additional context**

LixinGuoX, 2022-05-30:
Try add `    filter: CONFIG_ARMV6_M_ARMV8_M_BASELINE or CONFIG_ARMV7_M_ARMV8_M_MAINLINE and not CONFIG_NULL_POINTER_EXCEPTION_DETECTION_NONE
` in yaml file.
For reference: https://theotherjimmy.github.io/blog/blog/2022-01-24-bug-notes-str-alloc-copy.html

mbolivar-nordic, 2022-05-31:
pinging @joerchan @microbuilder

joerchan, 2022-06-01:
I can reproduce with nRF9160 as well. I think this might be a general TF-M problem.

joerchan, 2022-06-01:
The issue is that `SCB.AIRCR.BFHFNMINS` bit is not set, so the non-secure write to `SCB.ICSR.NMIPENDSET` is ignored.

I can't see any handling in TF-M for NMI interrupt so maybe this is overlooked. However I'd like to double-check that with the TF-M project.

joerchan, 2022-06-03:
Feel free to send a PR for documentation :)

We may want to add additional documentation for other fault handlers though. The ones that are in use by TF-M and that may halt the system in TF-M on failure. There are platform specific handlers in addition though, and the setup of the fault handlers are done in the HAL part for the platforms, so maybe there are differences.
We may of course handle this separately. Or maybe we can rely on TF-Ms documentation?

hunterhu3000, 2022-06-17:
We use the " not CONFIG_BUILD_WITH_TFM" to by pass the mps2_an521_ns board.

closed 2022-06-17 by hunterhu3000