zephyrproject-rtos/zephyr #46075: BT HCI Raw on STM32WB55RG

sebachm94, 2022-05-27:
**Describe the bug**
Hi,
I tried to build Zephyr hci_raw module to run one of prepared example from Zephyr: /zephyr/samples/bluetooth/hci_uart/.
It seems that there are some kind of dependencies and it's impossible to build HCI_Raw on STM32WB55RG.
I used sources and configuration from example and tried to build it, but it failed while linking, due to undefined references to:
'bt_hci_cmd_create' and 'bt_hci_cmd_send_sync' in /zephyr/drivers/bluetooth/hci/ipm_stm32wb.c, looks like this driver depends on HCI Core, which excludes HCI Raw.
I wasn't able to use other drivers too (maybe my configuration for other drivers wasn't full - I tried: CONFIG_BT_H4 instead of CONFIG_BT_STM32_IPM, which is taken by default). 
Is there any possibility to build HCI Raw with STM32WB55RG and run this example?

**To Reproduce**
Build example: /zephyr/samples/bluetooth/hci_uart/ on STM32WB55RG.

**Expected behavior**
HCI Raw build will pass and example will work on WB55RG.

erwango, 2022-05-30:
See https://docs.zephyrproject.org/latest/connectivity/bluetooth/bluetooth-arch.html

Using ble on stm32wb involves a Host-only build. This configuration is not compatible with hci_* samples which are only compatible with respective controller builds.

The configuration you need involves a new type of host build, which would transport hci packets from usart to existing ipm_stm32wb driver. A kind of transparent host build.

erwango, 2022-05-30:
@sebachm94 For my information, one question:
In your 'Direct Test Mode for BT', do you intent to test anything else than the BLE controller and RF subsytem ?
In a pure transparent mode application, HCI messages would go from UART to HCI Zephyr driver and quite a little of Zephyr comes into play.
For that purpose, a STM32Cube sample application is already provided in STM32WB STM32Cube package, cf https://github.com/STMicroelectronics/STM32CubeWB/tree/master/Projects/P-NUCLEO-WB55.Nucleo/Applications/BLE/BLE_TransparentMode
Isn't it sufficient for your need ?
you should be able to use https://github.com/STMicroelectronics/STM32CubeWB/tree/master/Projects/P-NUCLEO-WB55.Nucleo/Applications/BLE/BLE_TransparentMode which

PR #48350 (2022-08-09): drivers: bluetooth: stm32wb: Set HCI driver compatible with HCI_RAW mode
Current implementation of stm32wb HCI driver uses functions of hci_core driver to send specific messages.
This makes the driver incompatible with BT_HCI_RAW mode.

Provide some adaptations to allow use of BT_HCI_RAW mode on STM32WB
- Remove the HCI_RESET sending as this is available in hci_core.
- Put sending of ACI_HAL_* messages under !BT_HCI_RAW as these messages are not needed for test mode.

Add a c2h chosen to allow building hci_uart sample.

Fixes #46075
closed 2022-08-09 by carlescufi