zephyrproject-rtos/zephyr #46088: [backport v2.7-branch] backport of #44102 failed

zephyrbot, 2022-05-27:
This issue was created automatically because the backport of #44102 to `v2.7-branch` failed.

The backport to `v2.7-branch` failed:
```
The process '/usr/bin/git' failed with exit code 1
```
To backport manually, run these commands in your terminal:
```bash
# Fetch latest updates from GitHub
git fetch
# Create a new working tree
git worktree add .worktrees/backport-v2.7-branch v2.7-branch
# Navigate to the new working tree
cd .worktrees/backport-v2.7-branch
# Create a new branch
git switch --create backport-44102-to-v2.7-branch
# Cherry-pick the merged commits of this pull request and resolve the conflicts
git cherry-pick -x 192e2a8f9479c920aa6d349dbe74e0c18da41915~2..192e2a8f9479c920aa6d349dbe74e0c18da41915
# Push it to GitHub
git push --set-upstream origin backport-44102-to-v2.7-branch
# Go back to the original working tree
cd ../..
# Delete the working tree
git worktree remove .worktrees/backport-v2.7-branch
```
Then, create a pull request where the `base` branch is `v2.7-branch` and the `compare`/`head` branch is `backport-44102-to-v2.7-branch`.

cfriedt, 2022-08-02:
@cvinayak - was this resolved already via manual backport?

cfriedt, 2022-08-30:
manually backported

closed 2022-08-30 by cfriedt