zephyrproject-rtos/zephyr #46097: b_l072z_lrwan1 usart dma doesn't work

M1cha, 2022-05-29:
**Describe the bug**
DMA doesn't seem to work with UART on this board.
It looks like data is neither received nor sent.

Please also mention any information which could help others to understand
the problem you're facing:
 - `b_l072z_lrwan1`
 - If you enable UART debugging, you can see that incoming data does generate interrupts, but the DMA buffer status doesn't show any change.
 - I've tried both usart1 and usart2
 - I don't understand the `dmas` config 100% and copied it from `disco_l475_iot1.overlay`, so I wouldn't exclude a user error.

**To Reproduce**
Steps to reproduce the behavior:
1. Apply patch
2. connect PA9 and PA10 to each other
3. `west build -b b_l072z_lrwan1 zephyr/tests/drivers/uart/uart_async_api`
4. `west flash -r openocd`
5. tests fail

**Expected behavior**
The tests should succeed. More specifically, the DMA should be able to send and receive data.

**Impact**
It prevents using the async UART API with this board.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: `Linux`
 - Toolchain: `Zephyr SDK`
 - Version: `fa2054f93e4e80b079a4a6a9e84d642e51912042 (tag: v3.1.0-rc3)`

**Additional context**
- A log from a different test setup where I simply did an infinite async read and sent data to the board. i added more logs to the driver to see what's going on with the DMA buffer. The idle interrupt happens when I send a character.

```
[00:00:05.620,000] <dbg> uart_stm32: uart_stm32_isr: idle interrupt occurred
[00:00:05.620,000] <dbg> uart_stm32: async_timer_start: async timer started for 3333 us
[00:00:05.623,000] <dbg> uart_stm32: uart_stm32_async_rx_timeout: rx timeout
[00:00:05.624,000] <dbg> uart_stm32: uart_stm32_dma_rx_flush: rx: flush
[00:00:05.624,000] <dbg> uart_stm32: uart_stm32_dma_rx_flush: rx: flush: buflen=64 pending=64 len: 0
```

- Patch file for adding support to `b_l072z_lrwan1` to the test:
```patch
diff --git a/tests/drivers/uart/uart_async_api/boards/b_l072z_lrwan1.overlay b/tests/drivers/uart/uart_async_api/boards/b_l072z_lrwan1.overlay
new file mode 100644
index 0000000000..f0a25833cf
--- /dev/null
+++ b/tests/drivers/uart/uart_async_api/boards/b_l072z_lrwan1.overlay
@@ -0,0 +1,11 @@
+/* SPDX-License-Identifier: Apache-2.0 */
+
+&usart1 {
+       dmas = <&dma1 3 2 0x20440>,
+               <&dma1 5 2 0x20480>;
+       dma-names = "tx", "rx";
+};
+
+&dma1 {
+       status = "okay";
+};
diff --git a/tests/drivers/uart/uart_async_api/src/test_uart.h b/tests/drivers/uart/uart_async_api/src/test_uart.h
index b52176fe43..4d7f6398b1 100644
--- a/tests/drivers/uart/uart_async_api/src/test_uart.h
+++ b/tests/drivers/uart/uart_async_api/src/test_uart.h
@@ -67,6 +67,8 @@
 #elif defined(CONFIG_BOARD_MIMXRT1010_EVK) || \
        defined(CONFIG_BOARD_MIMXRT1015_EVK)
 #define UART_DEVICE_NAME DT_LABEL(DT_NODELABEL(lpuart4))
+#elif defined(CONFIG_BOARD_B_L072Z_LRWAN1)
+#define UART_DEVICE_NAME DT_LABEL(DT_NODELABEL(usart1))
 #else
 #define UART_DEVICE_NAME DT_LABEL(DT_CHOSEN(zephyr_console))
 #endif
```

M1cha, 2022-05-29:
I've updated the patch to use DMA channels which actually support usart1. Unfortunately, that didn't help.

M1cha, 2022-05-29:
Okay I managed to get it working by setting the dma slot to 3:
```
       dmas = <&dma1 2 2 0x20440>,
                    <&dma1 3 2 0x20480>;
```

I found that by trial & error and have no idea where I'd find that  information. I thought you're free to choose the slot when using a dmamux (which I'm not using).

erwango, 2022-06-03:
> Okay I managed to get it working by setting the dma slot to 3:
>
> ```
>        dmas = <&dma1 2 2 0x20440>,
>                     <&dma1 3 2 0x20480>;
> ```
>
> I found that by trial & error and have no idea [...]

If I'm correct this information could be found in ref man (RM076.pdf), Table 51. Summary of the DMA requests for each channel