zephyrproject-rtos/zephyr #46117: Kernel events can’t be manipulated without race conditions

ajackson-xlnx, 2022-05-30:
The recently introduced kernel events module allows tasks to wait for events to be delivered to an event object.  Events can be cleared from an event object either by call k_event_set or before waiting by the call to k_event_wait (or k_event_wait_all).  The k_event_wait semantics are problematic because a task may signal an event immediately followed by another task entering a wait condition and requesting that events be cleared.  The result is that events are lost.

It would be more robust if the clearing of events took place after the wait completed successfully.  The clearing would be of those events that had been signalled.  This would avoid the race.

So perhaps the clearing argument could be changed in the wait API to allow no clearing, before waiting or after waiting.  This would permit users to retain the existing behaviour.

peter-mitsis, 2022-05-31:
A couple of points.

1. The sequence of events that lead to lost events as described above has always been a known trade-off, and one that has been deliberately pushed on to the user/developer to ensure correct behaviour. It is alluded to in the documentation with the following text.

> Furthermore, threads making a wait request have the option of resetting the current set of events tracked by the event object prior to waiting. Care must be taken with this option when multiple threads wait on the same event object.

2. Having both pre- and post-clear wait options was actually something that was considered in an early draft of this feature, but was ultimately rejected at the time. I am trying to recall the rationale surrounding that decision. As it has been many months, some details are admittedly a little hazy.  That being said, complexity definitely played a role as there are numerous other associated corner cases to consider--behaviour on timeout, behaviour with multiple waiters, .... 

The above being said, I am not inherently opposed to this enhancement request. There is some definite value in having this post-clear behaviour. It will take some more time to figure it out.

ajackson-xlnx, 2022-07-04:
The problem is that, even in the simplest single producer/consumer case, events can be lost.  Consider two tasks A (producer) and B (consumer).  A can generate either or both events of value 1 and 2.

1. B calls `k_event_wait`, clearing events.
2. A calls `k_event_post` with 1.
3. B starts processing event 1.
4. A calls `k_event_post` with 2.
5. B calls `k_event_wait` again having finished its processing, clearing events: event 2 is now lost.

B can't easily clear the events that it is servicing with `k_event_set` because that function offers no masking type API.  It is possible that one could use other synchronisation primitives to handle this but that would seem to defeat the simplicity of the event interface.

Having a primitive like `k_event_set_masked` which took a `mask` parameter would help but I think that clear-after-read would be better in most circumstances.

closed 2022-07-25 by nashif

rclabs, 2024-09-04:
```
​isr_handler() {}
  k_event_post(button_event, 0x01);
}

​consumer() {
	uint32_t events = k_event_wait(button_event, 0x01, false, K_FOREVER);
	k_event_clear(button_event);

	​if (events) {
		// do some work
	}
}
```

### Sequence
1. isr_handler() posts an event
2. consumer thread wakes up to the event and subsequently clear the event

In between the consumer waiting for the event and clearing it, there could be another call to `isr_handler()`, which is cleared immediately.

This issue addresses it by introducing `k_event_set_masked()` which can ensure that only a specific event can be cleared, however, this wouldn't prevent a missing update for that same event.

Am I missing something here? would having an atomic `wait and clear` work?

Please link to the original proposal / discussions around events if avaialble to the public.

commit e1836718 (2022-07-25): kernel: Add k_event_set_masked primitive

There is no easy way to clear event bits without
the potential for a race to exist between producer(s)
and consumer(s). The result of this race is that events
can be lost through the various resetting mechanisms
available (flag to k_event_wait(), or k_event_set()).

Add k_event_set_masked() which permits bits to be set or cleared.
This allows consumers to clear just the bits that they have read
without (accidentally) discarding any new bits.

Update unit tests to verify the functionality.

Partly Fixes #46117.