zephyrproject-rtos/zephyr #46121: Bluetooth: Controller: hci: Wrong periodic advertising report data status

ppryga, 2022-05-31:
**Describe the bug**
The periodic advertising data status may be set to wrong value when
preparing a report event in HCI layer.

In case there periodic advertiser:

sends a data with length that is equal to `CONFIG_BT_CTLR_SCAN_DATA_LEN_MAX`.
the periodic advertising has included CTE
the number of PDUs in a chain is larger than required for transport
of advertising data (e.g. for max AD payload of 1650 bytes, minimum
required number of PDUs is 7). In other words there are empty PDUs
added to the chain to transport CTE.
the event generated in HCI layer has data status set to `BT_HCI_LE_AD_EVT_TYPE_DATA_STATUS_INCOMPLETE`.
That is caused by wrong condition checking if data_len_total is less
than `CONFIG_BT_CTLR_SCAN_DATA_LEN_MAX`.

**Expected behavior**
In such condition a report event should have data status set to `BT_HCI_LE_AD_EVT_TYPE_DATA_STATUS_COMPLETE`

**Impact**
Related with periodic advertising only.

jgl-meta, 2023-06-20:
@carlescufi Do you know what the status of this one is? It has been marked as stale many times.

carlescufi, 2023-09-06:
> @carlescufi Do you know what the status of this one is? It has been marked as stale many times.

Unfortunately still no ETA for this one yet.

cvinayak, 2024-01-12:
Marked as known issue.

Controller HCI encoding of reports needs implementation to report correct data status.

closed 2025-01-26 by github-actions[bot]