zephyrproject-rtos/zephyr #46126: pm_device causes assertion error in sched.c with lis2dh

volodyaleo, 2022-05-31:
**Describe the bug**

When pm_power_state_force set to OFF, the idle thread tries to suspend sensors.

`kernel/idle.c`:
```c
if (pm_system_suspend(_kernel.idle) == false) {
	k_cpu_idle();
}
```

But lis2dh has pm action handler that uses i2c bus.

Call sequence:
- `subsys/pm/pm.c`: function pm_suspend_devices
- `subsys/pm/device.c`: function pm_device_action_run
- `drivers/sensor/lis2dh/lis2dh.c`: lis2dh_pm_action (lis2dh->hw_tf->write_reg)
- `drivers/sensor/lis2dh/lis2dh_i2c.c`: lis2dh_i2c_write_reg

The i2c driver for nordic has semaphore inside to wait operation done:

`drivers/i2c/i2c_nrfx_twi.c`:
```c
ret = k_sem_take(&(get_dev_data(dev)->completion_sync),
				 I2C_TRANSFER_TIMEOUT_MSEC);
```

Then function `k_sem_take` calls function `z_pend_curr` in `sched.c`.

This call sequence in `sched.c` causes assertion error:
- z_pend_curr
- pend
- add_to_waitq_locked
- z_priq_wait_add
- z_priq_dumb_add (here error: `__ASSERT_NO_MSG(!z_is_idle_thread_object(thread));`)

**To Reproduce**
Steps to reproduce the behavior:

Set the configs:
```
CONFIG_I2C=y
CONFIG_SENSOR=y
CONFIG_PM=y
CONFIG_PM_DEVICE=y
CONFIG_LIS2DH=y
```
Call the `pm_power_state_force(0, { PM_STATE_SOFT_OFF, 0, 0 });` from delayed work.

**Expected behavior**

Sensors and CPU halted without error.

Expected suspending sensors not from idle thread.

Probably, pm needs to be improved:
- Suspend devices in separate thread/work
- When devices suspended, switch to the idle thread

**Logs and console output**

```
[00:00:30.289,337] <inf> event_manager: e: power_down_event
[00:00:30.289,398] <inf> event_manager: e:module_state_event module:buttons state:STANDBY
ASSERTION FAIL [!z_is_idle_thread_object(thread)] @ WEST_TOPDIR/zephyr/kernel/sched.c:186
[00:00:30.336,364] <err> os: r0/a1:  0x00000004  r1/a2:  0x000000ba  r2/a3:  0x200007e0
[00:00:30.336,395] <err> os: r3/a4:  0x20000e20 r12/ip:  0x00000000 r14/lr:  0x0000d4d5
[00:00:30.336,395] <err> os:  xpsr:  0x61000000
[00:00:30.336,395] <err> os: Faulting instruction address (r15/pc): 0x0000efe6
[00:00:30.336,395] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:30.336,395] <err> os: Current thread: 0x200007e0 (unknown)
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: gnuarmeabi
 - board: thingy52_nrf52832

gmarull, 2022-05-31:
The solution is really to make use of the device runtime PM framework. Right now, system suspends devices but it comes with limitations, see the note in https://docs.zephyrproject.org/latest/services/pm/device.html#device-power-management-operations.

gmarull, 2022-05-31:
As soon as a driver has device runtime PM enabled, system will actually ignore it. Drivers should be responsible to enable device runtime PM, so that they are left suspended after being initialized. However, I guess nothing prevents you to enable/disable it later  in the application side.

mbolivar-nordic, 2022-05-31:
pinging @avisconti as well

lairdjm, 2022-06-08:
Hitting this issue also, would be good to have a fix for it. Use case: sensors with irrecoverable errors or devices that need to maintain a very low sleep current/long read period

gmarull, 2022-06-30:
Note: if using `CONFIG_PM_DEVICE_RUNTIME=y` this problem should not occur anymore, since `CONFIG_PM_DEVICE_RUNTIME_EXCLUSIVE` will default yo `y`. Now drivers should be updated to make use of the PM device runtime APIs.

MaureenHelm, 2022-08-15:
@gmarull Do you expect to fix this for the v3.2 release?

gmarull, 2022-08-16:
I'm afraid the only solution here is to implement runtime PM for that driver, the problem listed here is a known limitation.

closed 2022-10-31 by github-actions[bot]