zephyrproject-rtos/zephyr #46127: RFC: seeking solutions for the issues of enabling the hardware stack guard for ARM64

povergoing, 2022-05-31:
## Introduction

I am working on enabling the stack guard for ARM64, and the work is not complicated, just adding a read-only region or digging a hole at the top of the stack. The hardware stack guard is backed by MMU for v8a or MPU for v8r. It would be something like:

High memory addresses
[image]
Low Memory addresses

But there is an issue that I wanted to discuss.

### Problem description

1. the user stack overflow is easy to detect and handle because we handle it at the privileged stack.
2. the issue is happening when the privileged stack overflows since the `z_arm64_serror`is still using the privileged stack but the privileged stack has already overflowed!

### Proposed change

If it is better if we add another stack like **error_stack/exception_stack** which is designed to handle this problem?

## Detailed RFC

I am adding the hardware stack guard for arm64. The issue happens when the privileged stack overflows. Because the routine of handling the data abort exception is using the privileged stack but the privileged stack is corrupted at that time. For example:
[image]
When the stack overflows the exception generates. It will call the `z_arm64_serror` then `z_arm64_fatal_error`, but this is still using the privileged stack.

So, I suggest that we create a new stack called error_stack or exception_stack for `z_arm64_serror` when  `SError / vSError` happens.

### Proposed change (Detailed)

- create an asm macro to switch stack
- switch stack for every `SError / vSError` exception then call the `z_arm64_enter_exc`.
- restore the sp with the original stack if needed.

### Dependencies

N/A

### Concerns and Unresolved Questions

I may miss something, and any suggestion is important.

## Alternatives

N/A

povergoing, 2022-05-31:
kindly ping @carlocaione @stephanosio @npitre @microbuilder

povergoing, 2022-05-31:
@SgrrZhf

carlocaione, 2022-05-31:
One possible solution here is to have the kernel threads also using `SP_EL0` instead of `SP_ELx`. On Exception / SError you would switch to `SP_ELx` that would always be pointing to some valid location into the stack.

povergoing, 2022-06-01:
> One possible solution here is to have the kernel threads also using `SP_EL0` instead of `SP_ELx`. On Exception / SError you would switch to `SP_ELx` that would always be pointing to some valid location into the stack.

Good idea, but the issue also exists? IIUC, when the user thread blows up its kernel stack, such as https://github.com/zephyrproject-rtos/zephyr/blob/5a7b0fb616dc73d85aa09c94c94491b32072030d/tests/kernel/fatal/exception/src/main.c#L408-L409, the issue happens. 

Or I misunderstand what you mean? Do you mean that we still use SP_EL0 when a user thread does a syscall ( EL0 to EL1), which means we use the `EL1t` mode instead of `EL1h` and need to switch the `sp` to the privileged stack manually for whatever exception (sync, irq, err, .etc) generates?

carlocaione, 2022-06-01:
> Good idea, but the issue also exists? IIUC, when the user thread blows up its kernel stack, such as

Uhm, indeed that would be an issue. At this point having a separate stack for `SError / vSError` could make sense, not sure how difficult is to implement that TBH.

povergoing, 2022-06-01:
> > Good idea, but the issue also exists? IIUC, when the user thread blows up its kernel stack, such as
> 
> Uhm, indeed that would be an issue. At this point having a separate stack for `SError / vSError` could make sense, not sure how difficult is to implement that TBH.

Let me have a try. Hope it won't create the domino effect

povergoing, 2022-07-07:
> One possible solution here is to have the kernel threads also using SP_EL0 instead of SP_ELx. On Exception / SError you would switch to SP_ELx that would always be pointing to some valid location into the stack.

@carlocaione, inspired by your idea, the only solution I can come up with to solve this issue may be that we go back to using EL1t mode. That is, both EL0 and EL1 use SP_EL0, and we switch to SP_ELx to do the check as soon as we enter the Exception and before we touch the sp.
https://github.com/zephyrproject-rtos/zephyr/blob/c4cdba3833e6cfe8b3832c8e8705354c16a8da8a/arch/arm64/core/vector_table.S#L37
Because the Exception caused by stack overflow can happen anywhere. After the check, we then decide to switch SP_EL0 to privileged stack or not depending on whether EL change.

That is to say, in terms of the user stack pointer and privileged stack pointer, we can "manually" switch SP_EL0 between these 2 stacks under the stack pointed by SP_ELx.

Basically, the whole things would be:
1) after the boot, we use SP_EL0(`SPSel` is 1, the EL1t), SP_ELx pointing to some always-valid location
2) when exception, switch to SP_ELx (set `SPSel`, no corrupt any registers), then check if stack overflow, oops when failing with check
3) assume the check pass, we save current SP_EL0 and set it to privileged stack if the exception from lower EL, or do nothing without EL change. Then switch back to SP_EL0. In this solution, the SP_EL1 is only for checking.
4) reverse the above steps when eret.

We have to do this because we need to handle both EL0  and EL1 stack overflow. In the current state (EL1h + EL0t), when EL0 and EL1 overflow at the same time, we have no stack to use to check. Besides, we can neither touch sp nor any registers at the beginning of Exception, much less switch stack.

Using SP_EL0 as both user stack pointer and privileged stack pointer can make it possible that we can switch stacks without touching neither sp nor registers when an Exception happens. Otherwise, I cannot find a better solution to introduce the third stack. 
Any comments?

npitre, 2022-07-07:
What about `at S1E0W, sp` upon entering the exception before using the stack?
I'm not sure it is possible to retrieve the result without clobbering at 
least one register though. But if that is possible then you wouldn't 
have to impact the common "no stack overflow" case with significant
overhead.

povergoing, 2022-07-11:
> What about `at S1E0W, sp` upon entering the exception before using the stack? I'm not sure it is possible to retrieve the result without clobbering at least one register though. But if that is possible then you wouldn't have to impact the common "no stack overflow" case with significant overhead.

I think it's a good idea, so I spent some time on it. I also would like to not change the current stack strategy. But I found that
`at S1E0W, sp` doesn't work, should be `at S1E0W, x0`. This is easy to solve by exchanging the sp and x0, an idea from the [Linux kernel](https://elixir.bootlin.com/linux/latest/source/arch/arm64/kernel/entry.S#L63).
The issue is that we still need the PAR_EL1 to check the result, right? But reading PAR_EL1 requires at least one Xt register. When checking is successful, the address saved to PAR_EL1, we could clobber the X0 (it is now sp), then recover it. However, when fail, there is no address info in PAR_EL1, in this case, we could not recover the sp. What's worth, even in success, PAR_EL1 only keeps the address bit[47:12], which is also unacceptable in terms of restoring.

Seems we need at least clobber one Xt register to do the check. This may not work.
Am I missing something? Any suggestions?

povergoing, 2023-11-27:
close it as #48297 solve this

closed 2023-11-27 by povergoing