zephyrproject-rtos/zephyr #46130: net: lwm2m: Object Update Callbacks

broglep-work, 2022-05-31:
# Problem

Currently there is no good way to trigger some business logic that depends on multiple IPSO resources changes. The only way the application can be notified about changes on IPSO resources is via `lwm2m_engine_register_post_write_callback`. However the application has no means to identify if there will be further writes to other resources, so it has no means to know when it should consider the lwm2m request to be fully processed and start with the business logic. 

# Possible Enhancement: Object Update Callbacks
We could introduce object update callbacks: 
`int lwm2m_engine_register_update_callback(uint16_t obj_id, lwm2m_engine_user_cb_t cb)`

(Or maybe better explicit naming `lwm2m_engine_register_object_update_callback`, considering further possible improvements from below)

This would complete the already existing [lwm2m_engine_register_create_callback](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/lwm2m/lwm2m_engine.c#L2935) and [lwm2m_engine_register_delete_callback](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/lwm2m/lwm2m_engine.c#L2950) and with that offer callbacks for the whole lifecycle of the object (instance). 

# Alternatives / Additional Enhancements

## Generic update callbacks for arbitrary paths
Extend the 
[`lwm2m_engine_register_post_write_callback(const char *pathstr,
					 lwm2m_engine_set_data_cb_t cb)`](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/lwm2m/lwm2m_engine.c#L2905) to support arbitrary paths (and maybe rename it to `lwm2m_engine_register_update_callback`) . This would allow for a single generic mechanism to receive a callback when something changes. It could support also update callbacks for all objects instances (path `/0`) or single resource instance. Implementation however could be more complex in this case. 

It would need to be discussed however if those pre/post write callbacks are conceptionally on the same abstraction level as a the suggested `resource update` / `object instance update` callbacks. Based on the current architecture, it feels to me that the pre/post write callbacks (along with validate, read, exec callbacks) are one level lower where it is all about one individual resources and its data & representation (data tier in multi-tier architecture terms), and you shouldn't really use those callbacks to trigger any kind of higher level business logic (application/logic tier). If that is the case, it would be better to keep `lwm2m_engine_register_post_write_callback` as is and introduce a new method `lwm2m_engine_register_update_callback`. 

Would be interesting to hear from others about the tiering aspect (that might have some more background on how the current architecture evolved and what the motives/intentions were)

## LWM2M Operation Callbacks (or Request/Response Processing Callbacks)

There would be a benefit if there are callbacks about the lifecycle of a lwm2m operation / message (like pre & post [`handle_request`](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/lwm2m/lwm2m_engine.c#L4594) callbacks). In combination with `lwm2m_engine_register_post_write_callback` it could be used to achieve something similar as the object update callback would offer. Each post write callback could record the written resource path, and once the lwm2m request processing finished callback would be called, could read out the modified paths and to do the necessary work. 
This would however still pose the question about the tiering aspect from above

# Discussion / Other Ideas?

Would be great to get some thought on this and into what direction the lwm2m engine could be extend. Or learn how other have solved that challenge (as I assume, almost everybody must face that challenge eventually when using the lwm2m engine)

rlubos, 2024-02-09:
CC @SeppoTakalo

SeppoTakalo, 2024-02-09:
I have been thinking this a bit... We even have a task in our backlog, but haven't started it.

Here is a one way to do this:
* When `lwm2m_set_*()` or `lwm2m_write_handler()` modified the data, it should not call the `post_write_cb()` but instead just mark `res_inst->data_flags |= FLAG_MODIFIED`
* Once the full packet is processed, or for example when we return to engine loop, there would be a loop that goes through all resource instances that have been modified, call the post_write_cb() and remove the MODIFIED flag.

Another improvements would be that we could allow post_write callbacks to be registered to Object Instance, or even Object level. API would allow this, but implementation would not.