zephyrproject-rtos/zephyr #46180: Add GitHub app for Googler notifications

yperess, 2022-06-02:
Several folk here at Google find it difficult to keep up with our code reviews on GitHub as it doesn't play nicely with our "normal" workflow. There's a GitHub app that integrates with our notification system which should allow us to be more responsive. Would it be possible to install it for this repo?

App: https://github.com/apps/g3n-github

stephanosio, 2022-06-02:
There are two problems:

1. We have no way of verifying that the author of the app is trustworthy (e.g. the author should be affiliated with Google or other verifiable third-party sources).
2. The app source code does not seem to be public and we have no way of verifying that the app does not contain any malicious code.

yperess, 2022-06-03:
I'll reach out to the G3N team at Google and see if they can provide a means for authenticating the app.

chases2, 2022-06-06:
Hi! I'm new to this community, but I helped to write the app in question.

What would be an acceptable way of demonstrating number 1? An email address with a google.com domain? Membership in a particular org?

As for number 2, the application only requests read permissions, so I'm not sure what malicious action it could take that would disrupt a user of the application (ie. the Zephyr team). If you have specific security concerns, let's try to find a way to address them.

stephanosio, 2022-06-06:
Having a "marketplace badge" added to the app would help:
https://docs.github.com/en/developers/github-marketplace/github-marketplace-overview/about-marketplace-badges#for-github-apps

See https://docs.github.com/en/developers/github-marketplace/github-marketplace-overview/applying-for-publisher-verification-for-your-organization

henrikbrixandersen, 2022-06-14:
Having to notify individual organizations through their respective, internal communication channels doesn't scale well.

yperess, 2022-06-15:
> Having to notify individual organizations through their respective, internal communication channels doesn't scale well.

Sure, manually, but in this case Google already took care of the mechanism to if every org wants to provide the tools I don't see why not. It leads to much higher productivity since there's less overhead for checking and following up on two different systems.

yperess, 2022-06-15:
@chases2 any update on this?

stephanosio, 2022-06-16:
Installed in the `zephyrproject-rtos` organisation with read-only access to the `zephyr` repository.

Security risks are minimal since it only has read access to a public repository.

closed 2022-06-16 by stephanosio