zephyrproject-rtos/zephyr #46186: ISO Broadcaster fails silently on unsupported RTN/SDU_Interval combination

GatCode, 2022-06-02:
When a user selects a specific RTN/SDU_Interval combination in the ISO_BROADCAST sample, broacaster fails without an error message. The user doesn't know what has happened.

As @Thalley suggested, a short error message would be nice to have.

Note: tested on two nRF52840DK boards with the code of the following draft PR: 

[image]

Thalley, 2022-06-02:
The error seen is that no ISO data is seemingly received on the sync receiver, right @GatCode ?

GatCode, 2022-06-02:
Exactly, only the BigInfo is printed on the receiver side.
Additionally the output on the sender side is exactly the screenshot above but it should print `Sending value ...` every second.

Thalley, 2022-06-02:
> Exactly, only the BigInfo is printed on the receiver side. Additionally the output on the sender side is exactly the screenshot above but it should print `Sending value ...` every second.

So at least the periodic advertisement seems to be fine. From your log, it seems that just the BIG sync never completes.

PR #49394 (2022-09-08): Bluetooth: Controller: Fix BIG Create silently fail on RTN/SDU_interval
Fix BIG Create implementation from silently failing on
values of RTN and SDU_interval that lead to BIG events that
cannot be acheived when using sequential or interleaved
subevents by the Controller Implementation.

Fixes #46186.
closed 2022-09-08 by carlescufi