zephyrproject-rtos/zephyr #46199: LIS2DW12 I2C driver uses invalid write command

jacobw56, 2022-06-02:
**Describe the bug**
LIS2DW12 I2C driver uses invalid write command

**To Reproduce**
Steps to reproduce the behavior:
1. Create LIS2DW12 I2C device and attempt to write to a register.
2. Observe the following invalid I2C write sequence and note that data has not been written:
[image]

**Expected behavior**
According to the datasheets for the LIS2DW12 and LIS2DTW12, this is the expected I2C write sequence: 
[image]

**Impact**
Currently, LIS2DW12 and LIS2DTW12 are non-functional via I2C. It also appears that the LIS2DS12 should also be broken as it uses the same invalid code.

**Logs and console output**
Here are scope screens that show the current behavior:
[image]
[image]
[image]
[image]

**Additional context**
The issue is that `lis2dw12.c` uses `stmemsc_i2c_write` from `stmemsc_i2c.c` to write to the device, which in turn calls to `i2c_burst_write` in `i2c.h`. This routine attempts to perform the write as shown in the I2C sequence listed above in the steps to reproduce.  The offending code is [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/sensor/lis2dw12/lis2dw12.c#L522).  I managed to get around it on our project with 

    static int local_i2c_write(const struct stmemsc_cfg_i2c *stmemsc, uint8_t reg_addr, uint8_t *value, uint8_t len)
    {
    	uint8_t buffer[2];
    	struct i2c_msg msg;
    
    	buffer[0] = reg_addr;
    	buffer[1] = *value;
    
    	msg.buf = (uint8_t *)buffer;
    	msg.len = 2U;
    	msg.flags = I2C_MSG_WRITE | I2C_MSG_STOP;
    
    	return i2c_transfer(stmemsc->bus, &msg, 1, stmemsc->i2c_slv_addr);
    }

and changing `ctx.write_reg` to point to this function.

carlescufi, 2022-06-03:
>  I managed to get around it on our project with

@jacobw56 could you send a Pull Request with that code please?

fabiobaltieri, 2022-06-15:
@jacobw56 can you send [3aa006c](https://github.com/jacobw56/zephyr/commit/3aa006c51e706052a13e1db548a69347bde40b57) in the form of a Pull Request?

closed 2022-08-29 by github-actions[bot]