zephyrproject-rtos/zephyr #46206: it8xxx2_evb: tests/kernel/fatal/exception/ assertion failed -- "thread was not aborted"

hunterhu3000, 2022-06-02:
**Describe the bug**
Test case "tests/kernel/fatal/exception/" failed on it8xxx2_evb

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b it8xxx2_evb tests/kernel/fatal/exception/
2. sudo -S ~/itetool/ite -f build/zephyr/zephyr.bin -s check

**Expected behavior**
The test can pass

**Impact**
daily test fail

**Logs and console output**
```
*** Booting Zephyr OS build v3.1.0-rc3-37-ga78e7c5c90d8  (delayed boot 1ms) ***
Running TESTSUITE fatal
===================================================================
START - test_fatal
test alt thread 1: generic CPU exception
===================================================================
PROJECT EXECUTION SUCCESSFUL

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/fatal/exception/src/main.c:305: test_fatal: (rv equal to TC_FAIL)
thread was not aborted
 FAIL - test_fatal in 0.26 seconds
===================================================================
TESTSUITE fatal failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): Ubuntu 20.04
 - Toolchain (e.g Zephyr SDK, ...): 0.14.1
 - Commit SHA or Version used: a78e7c5c90d8cb31ec550ce289503bd698675585

hunterhu3000, 2022-06-03:
@GTLin08 @RuibinChang it8xxxb_evb does not support this test right? I saw this in the code
```
static void entry_cpu_exception(void)
{
	expected_reason = K_ERR_CPU_EXCEPTION;

	TC_PRINT("cpu exception\n");
#if defined(CONFIG_X86)
	__asm__ volatile ("ud2");
#elif defined(CONFIG_NIOS2)
	__asm__ volatile ("trap");
#elif defined(CONFIG_ARC)
	__asm__ volatile ("swi");
#else
	/* Triggers usage fault on ARM, illegal instruction on RISCV
	 * and xtensa
	 */
	{
		volatile long illegal = 0;
		((void(*)(void))&illegal)();
	}
#endif
}
```

hunterhu3000, 2022-06-03:
> @GTLin08 @RuibinChang it8xxxb_evb does not support this test right? I saw this in the code
>
> ```
> static void entry_cpu_exception(void)
> {
> 	expected_reason = K_ERR_CPU_EXCEPTION;
>
> 	TC_PRINT("cpu [...]

Sorry, I misunderstand the comments. The test case is expected to work on riscv.
Before commit 4c31f24de721102642f1c5bccc88f5dbe9d724bf, it works well.
after efaeed6cb29079820a11043d0473a2562a2d4b16	it got a compile problem
after 46160562831b2591431bc587809c748971b772e9	the compile problem is fixed, but the test case run fail

hunterhu3000, 2022-06-05:
Fix the compile problem by changing
``` 
#include <zephyr/zephyr/types.h>
```
to
```
#include <zephyr/types.h>
```
4cf45f4770e963562524dafebda62db60e43f82b is the first bad commit

hunterhu3000, 2022-06-17:
The problem can be fixed by 86c45b7e12b38fbb1201fe65bc2f8f90c9bf7b95

But the test of 
```
west build -p always -b it8xxx2_evb tests/kernel/fatal/exception/ -DCONF_FILE=sentinel.conf 
```
still fail. Could you please check if CONFIG_STACK_SENTINEL=y on ITE board? Thanks

RuibinChang, 2022-06-17:
After running the cmd `west build -p always -b it8xxx2_evb tests/kernel/fatal/exception/`, all configurations of it8xxx2_evb board are list in `build/zephyr/.config`, and I see CONFIG_STACK_SENTINEL is not set.

hunterhu3000, 2022-06-29:
The issue is fixed by 3f8e326d1a63af7eb294c59078a80a3da7f71dfc

closed 2022-06-29 by hunterhu3000