zephyrproject-rtos/zephyr #46208: it8xxx2_evb: tests/kernel/sleep failed, elapsed_ms = 2125

hunterhu3000, 2022-06-02:
**Describe the bug**
Test case "tests/kernel/sleep" got assertion fail. The timer is not accurate. This problem is solved before, see https://github.com/zephyrproject-rtos/zephyr/issues/43513. It happens again maybe because of the SDK upgrade (to 0.14.1)

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b it8xxx2_evb tests/kernel/sleep
 2. sudo -S ~/itetool/ite -f build/zephyr/zephyr.bin -s check

**Expected behavior**
The test should pass, elapsed_ms should less than 2000

**Impact**
daily test fail

**Logs and console output**
```*** Booting Zephyr OS build v3.1.0-rc3-37-ga78e7c5c90d8  (delayed boot 1ms) ***
Running TESTSUITE sleep
===================================================================
START - test_sleep
Kernel objects initialized
Test thread started: id = 0x801010f0
Helper thread started: id = 0x80101048
Testing normal expiration of k_sleep()
Testing: test thread sleep + helper thread wakeup test
Testing: test thread sleep + isr offload wakeup test
Testing: test thread sleep + main wakeup test thread
Testing kernel k_sleep()
 PASS - test_sleep in 2.34 seconds
===================================================================
START - test_usleep
elapsed_ms = 2125

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sleep/src/usleep.c:95: test_usleep: (elapsed_ms <= UPPER_BOUND_MS is false)
overslept
 FAIL - test_usleep in 21.265 seconds
===================================================================
START - test_sleep_forever
Kernel objects initialized
 PASS - test_sleep_forever in 0.4 seconds
===================================================================
TESTSUITE sleep failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): Ubuntu 20.04
 - Toolchain (e.g Zephyr SDK, ...): 0.14.1
 - Commit SHA or Version used: a78e7c5c90d8cb31ec550ce289503bd698675585

RuibinChang, 2022-06-07:
@hunterhu3000 I test with https://github.com/zephyrproject-rtos/zephyr/commit/a78e7c5c90d8cb31ec550ce289503bd698675585 and SDK 0.13.2, the result is pass.

hunterhu3000, 2022-06-07:
@RuibinChang Could you please try 0.14.1
It may also related to the board. I found it works with 0.13.2 and does not work wiht 0.14.1 on the previous board. As the previous board is broken(no serial output), we tried it on the new board. For the new board, it does not work on either 0.13.2 or 0.14.1. The logs are the same.

RuibinChang, 2022-06-13:
@hunterhu3000  
I update the SDK to latest version v0.14.2 on PR b9e833366f1360ecbad8776d6e8587e180688e1b, the 10 times result of test_usleep are pass.

hunterhu3000, 2022-06-17:
@RuibinChang It is strange that on my side https://github.com/zephyrproject-rtos/zephyr/commit/b9e833366f1360ecbad8776d6e8587e180688e1b cannot fix the problem.

But 985bdcd0766748409d09a0d72d260d869e534506 fixes the problem. I wonder if it is related to the compiler.

hunterhu3000, 2022-06-29:
@RuibinChang The case failes again on the current commit a57dee94d68f2aa0f1a6a07512f1ceab6248c982
``
ITE-Daily@14:23:18:~/zephyrproject/zephyr$ . zephyr-env.sh
ITE-Daily@14:23:21:~/zephyrproject/zephyr$ twister  -p it8xxx2_evb --device-testing --west-flash="../../itetool/loader.sh" --device-serial=/dev/ttyUSB0 -T tests/kernel/sleep/
Renaming output directory to /home/ztest/zephyrproject/zephyr/twister-out.7
INFO    - Zephyr version: zephyr-v3.1.0-990-ga57dee94d68f
INFO    - JOBS: 8
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /home/ztest/zephyrproject/zephyr/twister-out/testplan.json
INFO    - Using 'zephyr' toolchain.

Device testing on:

| Platform    | ID   | Serial device   |
|-------------|------|-----------------|
| it8xxx2_evb |      | /dev/ttyUSB0    |

INFO    - 1 test scenarios (1 configurations) selected, 0 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue

ERROR   - it8xxx2_evb               tests/kernel/sleep/kernel.common.timing            FAILED: Failed
ERROR   - see: /home/ztest/zephyrproject/zephyr/twister-out/it8xxx2_evb/tests/kernel/sleep/kernel.common.timing/handler.log
INFO    - Total complete:    1/   1  100%  skipped:    0, failed:    1
INFO    - 0 of 1 test configurations passed (0.00%), 1 failed, 0 skipped with 0 warnings in 35.05 seconds
INFO    - In total 3 test cases were executed, 0 skipped on 1 out of total 467 platforms (0.21%)
INFO    - 1 test configurations executed on platforms, 0 test configurations were only built.

Hardware distribution summary:

| Board       | ID   |   Counter |
|-------------|------|-----------|
| it8xxx2_evb |      |         1 |
INFO    - Saving reports...
INFO    - Writing JSON report /home/ztest/zephyrproject/zephyr/twister-out/twister.json
INFO    - Using 'zephyr' toolchain.
INFO    - Writing xunit report /home/ztest/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/ztest/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
ITE-Daily@14:23:59:~/zephyrproject/zephyr$
``
The west log:
```
Terminal ready
***** delaying boot 1ms (per build configuration) *****
*** Booting Zephyr OS build zephyr-v3.1.0-990-ga57dee94d68f  (delayed boot 1ms) ***
Running TESTSUITE sleep
===================================================================
START - test_sleep
Kernel objects initialized
Test thread started: id = 0x801010f0
Helper thread started: id = 0x80101048
Testing normal expiration of k_sleep()
Testing: test thread sleep + helper thread wakeup test
Testing: test thread sleep + isr offload wakeup test
Testing: test thread sleep + main wakeup test thread
Testing kernel k_sleep()
 PASS - test_sleep in 2.34 seconds
===================================================================
START - test_usleep
elapsed_ms = 2375

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sleep/src/usleep.c:95: test_usleep: (elapsed_ms <= UPPER_BOUND_MS is false)
overslept
 FAIL - test_usleep in 23.765 seconds
===================================================================
START - test_sleep_forever
Kernel objects initialized
 PASS - test_sleep_forever in 0.4 seconds
===================================================================
TESTSUITE sleep failed.
===================================================================
PROJECT EXECUTION FAILED
```

PR #47084 (2022-07-05): ITE tests/kernel/sleep: tune CONFIG_SYS_CLOCK_TICKS_PER_SEC down
We need more time to run codes because of the performance,
so I tune CONFIG_SYS_CLOCK_TICKS_PER_SEC down to reduce
the times of running k_usleep(1), then it can pass test_usleep().

Verified by follow test pattern:
west build -p always -b it8xxx2_evb tests/kernel/sleep

fixes #46208
commit dc3f29f6 (2022-07-05): ITE tests/kernel/sleep: tune CONFIG_SYS_CLOCK_TICKS_PER_SEC down

We need more time to run codes because of the performance,
so I tune CONFIG_SYS_CLOCK_TICKS_PER_SEC down to reduce
the times of running k_usleep(1), then it can pass test_usleep().

Verified by follow test pattern:
west build -p always -b it8xxx2_evb tests/kernel/sleep

fixes #46208
closed 2022-07-05 by fabiobaltieri