zephyrproject-rtos/zephyr #46235: subsystem: Bluetooth LLL: ASSERTION FAIL [!link->next]

kuikka, 2022-06-03:
BLE link layer asserts and causes a kernel Oops. (see console log below)

HW: nRF52840-DK
SW: Zephyr v3.0.0
This unit is a BLE Central, connecting to a peripheral.

See previous case (https://github.com/zephyrproject-rtos/zephyr/issues/45314) for context.

**To Reproduce**
A previous connection is disconnected, new connection is started and the assert happens very quickly after.

**Logs and console output**
```
[00:00:46.809,112] <inf> l2cap: L2CAP disconnected. PSM = e1
[00:00:46.809,204] <inf> gatt: GATT DISCONNECTED
[00:00:46.809,204] <inf> gap: DISCONNECTED, reason 0x16
Thread analyze:
 BT RX               : STACK: unused 3256 usage 840 / 4096 (20 %); CPU: 0 %
      : Total CPU cycles used: 249
 BT RX pri           : STACK: unused 3824 usage 272 / 4096 (6 %); CPU: 0 %
      : Total CPU cycles used: 141
 BT ECC              : STACK: unused 3212 usage 884 / 4096 (21 %); CPU: 0 %
      : Total CPU cycles used: 8142
 BT TX               : STACK: unused 3576 usage 520 / 4096 (12 %); CPU: 0 %
      : Total CPU cycles used: 182
 uart_rx_thread_tid  : STACK: unused 3560 usage 536 / 4096 (13 %); CPU: 0 %
      : Total CPU cycles used: 70
 thread_analyzer     : STACK: unused 576 usage 448 / 1024 (43 %); CPU: 2 %
      : Total CPU cycles used: 38490
 0x200023f8          : STACK: unused 3472 usage 624 / 4096 (15 %); CPU: 0 %
      : Total CPU cycles used: 369
 sysworkq            : STACK: unused 1272 usage 776 / 2048 (37 %); CPU: 0 %
      : Total CPU cycles used: 113
 usbworkq            : STACK: unused 712 usage 312 / 1024 (30 %); CPU: 0 %
      : Total CPU cycles used: 222
 logging             : STACK: unused 240 usage 528 / 768 (68 %); CPU: 0 %
      : Total CPU cycles used: 13894
 idle 00             : STACK: unused 212 usage 108 / 320 (33 %); CPU: 96 %
      : Total CPU cycles used: 1625200
ASSERTION FAIL [!link->next] @ ZEPHYR_BASE/subsys/bluetooth/controller/ll_sw/ull_conn.c:1952
[00:00:52.839,294] <inf> gap: Scanning started successfully
[00:00:52.885,681] <inf> gatt: GATT CONNECTED mtu=23
[00:00:52.885,681] <inf> gap: CONNECTED err=0x0
[00:00:52.887,603] <inf> gatt: Starting discovery
[00:00:53.045,013] <err> os: r0/a1:  0x00000003  r1/a2:  0x00000000  r2/a3:  0x00000001
[00:00:53.045,013] <err> os: r3/a4:  0x00024305 r12/ip:  0xa0000000 r14/lr:  0x0001dd5f
[00:00:53.045,043] <err> os:  xpsr:  0x4100001b
[00:00:53.045,043] <err> os: Faulting instruction address (r15/pc): 0x0001dd6a
[00:00:53.045,074] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:00:53.045,104] <err> os: Fault during interrupt handling

[00:00:53.045,135] <err> os: Current thread: 0x20002548 (idle 00)
[00:00:53.140,228] <err> os: Halting system
```

I have enabled thread analyzer as you can see from the logs per instructions in previous case.
I also increased all pertinent stack sizes significantly (to 4096 bytes) but this makes no difference to this issue. As you can see in the normal case the BT and USB threads are all <25% stack usage.

Since I can reproduce this easily again, I'd be open to trying different solutions or workarounds.

cvinayak, 2022-06-03:
I do not see the main thread nor the ISR thread being listed, what is the stack sizes you are using for them. 

Please provide details on the sample used, build procedure followed and test steps. Ideally, please share your application in a branch/pull request that can be used to reproduce the assertion.

kuikka, 2022-06-03:
Main thread exists after initialization in my case. I added a while(1) k_msleep() to it to keep it around but it does not overflow and makes no difference to this issue.

I don't see a thread for ISRs but I did increase the ISR stack as well to 4096 and it made no difference.

```
CONFIG_BT_RX_STACK_SIZE=4096
CONFIG_BT_HCI_TX_STACK_SIZE_WITH_PROMPT=y
CONFIG_BT_HCI_TX_STACK_SIZE=4096
CONFIG_BT_CTLR_RX_PRIO_STACK_SIZE=4096
CONFIG_BT_HCI_ECC_STACK_SIZE=4096
CONFIG_PRIVILEGED_STACK_SIZE=4096
CONFIG_ISR_STACK_SIZE=4096
CONFIG_USB_NRFX_WORK_QUEUE_STACK_SIZE=4096
```

Unfortunately I am unable to share the application. I tried to reproduce the issue with the sample app but was unable to so far.

Can you tell me what this assert signifies? What is the error that has happened? Are you confident that it is a memory corruption issue? Is there anything in Zephyr that could further help to debug this? Is there a way to get a look at the thread usage after an assert?

The application is built against v3.0.0 with the usual `cmake -B build && make -C build` combination.

The application acts as a LE central with a USB CDC ACM serial port connection to a host PC.

This issue always appears on reconnection, sequence being:
1. Scan for peripheral
2. Stop scan and connect
3. Open an L2CAP channel to peripheral
4. Disconnect
5. Scan again
6. Stop scan and connect
7. Assert

The devices are bonded and peripheral will ensure L2 security.

```
CONFIG_ASSERT=y
CONFIG_BT=y
CONFIG_BT_BUF_ACL_RX_SIZE=251
CONFIG_BT_BUF_ACL_TX_COUNT=10
CONFIG_BT_BUF_ACL_TX_SIZE=251
CONFIG_BT_CENTRAL=y
CONFIG_BT_CTLR_DATA_LENGTH_MAX=251
CONFIG_BT_CTLR_RX_BUFFERS=6
CONFIG_BT_CTLR_RX_PRIO_STACK_SIZE=4096
CONFIG_BT_FILTER_ACCEPT_LIST=y
CONFIG_BT_GATT_CLIENT=y
CONFIG_BT_HCI_ECC_STACK_SIZE=4096
CONFIG_BT_HCI_TX_STACK_SIZE=4096
CONFIG_BT_HCI_TX_STACK_SIZE_WITH_PROMPT=y
CONFIG_BT_KEYS_OVERWRITE_OLDEST=y
CONFIG_BT_L2CAP_DYNAMIC_CHANNEL=y
CONFIG_BT_L2CAP_TX_BUF_COUNT=10
CONFIG_BT_L2CAP_TX_MTU=247
CONFIG_BT_RX_STACK_SIZE=4096
CONFIG_BT_SETTINGS=y
CONFIG_BT_SMP=y
CONFIG_BT_USER_DATA_LEN_UPDATE=y
CONFIG_BT_USER_PHY_UPDATE=y
CONFIG_FLASH=y
CONFIG_FLASH_LOAD_OFFSET=0x1000
CONFIG_FLASH_LOAD_SIZE=0xCB000
CONFIG_FLASH_MAP=y
CONFIG_FLASH_PAGE_LAYOUT=y
CONFIG_ISR_STACK_SIZE=4096
CONFIG_LOG=y
CONFIG_NVS=y
CONFIG_PRIVILEGED_STACK_SIZE=4096
CONFIG_RING_BUFFER=y
CONFIG_SERIAL=y
CONFIG_SETTINGS=y
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048
CONFIG_THREAD_ANALYZER=y
CONFIG_THREAD_ANALYZER_AUTO=y
CONFIG_THREAD_ANALYZER_AUTO_INTERVAL=5
CONFIG_THREAD_ANALYZER_RUN_UNLOCKED=y
CONFIG_THREAD_ANALYZER_USE_PRINTK=y
CONFIG_THREAD_NAME=y
CONFIG_THREAD_NAME=y
CONFIG_UART_CONSOLE=y
CONFIG_UART_INTERRUPT_DRIVEN=y
CONFIG_UART_LINE_CTRL=y
CONFIG_USB_CDC_ACM=y
CONFIG_USB_DEVICE_PRODUCT="foobar"
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_NRFX_WORK_QUEUE_STACK_SIZE=4096
```

kuikka, 2022-06-03:
I was able to reduce the application to minimum and still get a repro. Attached here.
[repro.zip](https://github.com/zephyrproject-rtos/zephyr/files/8835180/repro.zip)
.

Log:
```
*** Booting Zephyr OS build zephyr-v3.0.0-1-g1bda9129f27b  ***
[00:00:00.264,923] <inf> fs_nvs: 4 Sectors of 4096 bytes
[00:00:00.264,923] <inf> fs_nvs: alloc wra: 0, fd0
[00:00:00.264,923] <inf> fs_nvs: data wra: 0, 1c
[00:00:00.270,355] <inf> bt_hci_core: No ID address. App must call settings_load()
[00:00:00.270,385] <inf> main: Bluetooth initialized
Thread analyze:
 BT RX               : STACK: unused 3936 usage 160 / 4096 (3 %); CPU: 0 %
      : Total CPU cycles used: 1
 BT RX pri           : STACK: unused 3944 usage 152 / 4096 (3 %); CPU: 0 %
      : Total CPU cycles used: 0
 BT ECC              : STACK: unused 3212 usage 884 / 4096 (21 %); CPU: 78 %
      : Total CPU cycles used: 8573
 BT TX               : STACK: unused 3712 usage 384 / 4096 (9 %); CPU: 0 %
      : Total CPU cycles used: 50
 thread_analyzer     : STACK: unused 608 usage 416 / 1024 (40 %); CPU: 11 %
      : Total CPU cycles used: 1388
 sysworkq            : STACK: unused 3440 usage 656 / 4096 (16 %); CPU: 0 %
      : Total CPU cycles used: 41
 logging             : STACK: unused 336 usage 432 / 768 (56 %); CPU: 7 %
      : Total CPU cycles used: 965
 idle 00             : STACK: unused 288 usage 32 / 320 (10 %); CPU: 0 %
      : Total CPU cycles used: 0
 main                : STACK: unused 448 usage 576 / 1024 (56 %); CPU: 4 %
      : Total CPU cycles used: 562
[00:00:01.272,155] <inf> main: state=0
[00:00:02.272,277] <inf> main: state=1
[00:00:02.273,193] <inf> main: bt_conn_le_create() ret=0
[00:00:02.343,841] <inf> main: Connected, err=0
[00:00:03.273,315] <inf> main: state=4
[00:00:04.273,559] <inf> main: bt_conn_disconnect() ret=0
[00:00:05.273,681] <inf> main: state=5
[00:00:05.274,444] <inf> main: bt_conn_le_create() ret=0
ASSERTION FAIL [!link->next] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_conn.c:1952
[00:00:05.344,970] <inf> main: Connected, err=0
[00:00:05.504,974] <err> os: r0/a1:  0x00000003  r1/a2:  0x00000000  r2/a3:  0x00000001
[00:00:05.504,974] <err> os: r3/a4:  0x0001f06d r12/ip:  0xa0000000 r14/lr:  0x00019b7f
[00:00:05.504,974] <err> os:  xpsr:  0x4100001b
[00:00:05.505,004] <err> os: Faulting instruction address (r15/pc): 0x00019b8a
[00:00:05.505,035] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:00:05.505,035] <err> os: Fault during interrupt handling

[00:00:05.505,065] <err> os: Current thread: 0x20001a68 (idle 00)
[00:00:05.903,594] <err> os: Halting system
```

cvinayak, 2022-06-04:
Please try with `CONFIG_LOG=n` and use `printk` in your application. I tried your application under babblesim simulation and I do not observe any crashes. 
[log.txt](https://github.com/zephyrproject-rtos/zephyr/files/8837058/log.txt)

kuikka, 2022-06-07:
I will try that but it seems to me this may be related to timing on the physical layer. I will see if I can similarly reduce the peripheral code and share that as well. It is on nRF52840 but with S140 soft device.

cvinayak, 2022-06-08:
> It is on nRF52840 but with S140 soft device.

You are not using the S140 SoftDevice as the assertion reported is from a source file in the Open Source Bluetooth Low Energy Controller implementation in Zephyr Project. If you intend to use S140 softdevice, and face the assertion, you have to get assistance at https://devzone.nordicsemi.com/

carlescufi, 2022-06-08:
@kuikka can you please try to reproduce the issue with the latest `main` (or 3.1.0, which was just released). Thanks!

kuikka, 2022-06-10:
@cvinayak I think there was a misunderstanding here. The central device that shows the assert runs on Zephyr, the peripheral it connects to during this assert runs S140 soft device.

@carlescufi Certainly.

kuikka, 2022-06-15:
I am still seeing an issue with v3.1.0 but it looks different:

```
[00:01:24.080,230] <err> os: ***** USAGE FAULT *****
[00:01:24.080,261] <err> os:   Illegal use of the EPSR
[00:01:24.080,291] <err> os: r0/a1:  0x20001a38  r1/a2:  0x2000255c  r2/a3:  0x00000000
[00:01:24.080,322] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x0002f7c9
[00:01:24.080,322] <err> os:  xpsr:  0x60000000
[00:01:24.080,352] <err> os: Faulting instruction address (r15/pc): 0x00000000
[00:01:24.080,352] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:01:24.080,383] <err> os: Current thread: 0x20001fc0 (BT RX)
[00:01:24.158,721] <err> os: Halting system
```

I need to dig more into this.

kuikka, 2022-06-15:
This turned out to be a memory issue as you suspected. 

I was bitten by the 2nd argument to `k_mem_slab_free()` needing to be a `void**`. In this case a bogus "buffer" got put into the free list of the slab and then returned for the next allocation.

Thank you for your help!

closed 2022-06-15 by kuikka

kuikka, 2022-06-15:
Just to clarify, I still see the original issue in v3.0.0 but no longer in v3.1.0 so it has been fixed on the way.

cvinayak, 2022-06-15:
Thank you for following up. If the issue recurs, feel free to open a new github issue.