zephyrproject-rtos/zephyr #46241: Bluetooth: Controller: ISO: Setting CONFIG_BT_CTLR_ISO_TX_BUFFERS=4 breaks non-ISO data

Thalley, 2022-06-03:
**Describe the bug**
If `CONFIG_BT_CTLR_ISO_TX_BUFFERS=4` and CONFIG_BT_CTLR_ADV_ISO=y then non-ISO TX buffer overflow occurs (e.g. when doing GATT operations). It also fails for larger values than 4. 

**To Reproduce**
Run e.g. `tests/bluetooth/bsim_bt/bsim_test_audio/test_scripts/vcs.sh` with `CONFIG_BT_CTLR_ISO_TX_BUFFERS=4` (instead of the CONFIG_BT_CTLR_ISO_TX_BUFFERS=3 which is the default/current at the time of this issue) with https://github.com/zephyrproject-rtos/zephyr/pull/46204.

**Expected behavior**
That the test passes with `CONFIG_BT_CTLR_ISO_TX_BUFFERS=4` as it does with `CONFIG_BT_CTLR_ISO_TX_BUFFERS=3`.

**Impact**
Minor

**Logs and console output**
Running the test will result in the following:
```
d_01: @00:00:00.835740  [00:00:00.835,723] <dbg> bt_vcs_client: vcs_discover_func: Setup complete for VCS
d_00: @00:00:01.086560  [00:00:01.086,547] <err> bt_ctlr_hci: Tx Buffer Overflow
d_00: @00:00:01.086560  [00:00:01.086,547] <wrn> bt_hci_core: Data buffer overflow (link type 0x01)
d_00: @00:00:01.086560  [00:00:01.086,547] <err> bt_conn: Unable to send to driver (err -105)
d_01: @00:00:30.000000 ERROR: (WEST_TOPDIR/zephyr/tests/bluetooth/bsim_bt/bsim_test_audio/src/common.c:73): test failed (not passed after 30 seconds)
```

**Environment (please complete the following information):**
 - Commit SHA or Version used: https://github.com/zephyrproject-rtos/zephyr/pull/46204

**Additional context**N/A

cvinayak, 2022-06-03:
[image]

if CONFIG_BT_ISO=y  then the wrong bt_dev.le.iso_pkts  will be returned for ACL packet transmission, right?

Thalley, 2022-06-03:
> [image]
> 
> if CONFIG_BT_ISO=y then the wrong bt_dev.le.iso_pkts will be returned for ACL packet transmission, right?

It would appear so due to the `||` instead of a `&&`

PR #46246 (2022-06-03): Bluetooth: Host: Fix incorrect ACL buffer count
Fix incorrect ACL buffer count use when Controller used
supports separate ISO buffers.

Fixes #46241.
closed 2022-06-03 by carlescufi