zephyrproject-rtos/zephyr #46251: memset broken behaviour

torres98, 2022-06-03:
After the release candidate updates i've been experiencing some completely broken behaviour regarding some C and C++ functionalities of the standard library. Most notably, memset seems to be stuck in a loop whenever it gets called (both from a C and a C++ application); i've tried investigating further but i'm not that skilled with gdb or debugging in general. Anyway here is the most simple snippet that causes this problem:

```
#include <stdio.h>
#include <string.h>

int main() {
    unsigned char message[1000];

    printf("Reachable.\n");
    memset(message, 0x41, 1000);
    printf("Unreachable.\n");

    return 0;
}
```
Furthermore, i've experienced a similar behaviour with `std::ostringstream`: just declaring a variable of that types completely locks the application indefinitely. 

I'm working on Linux Mint 20.3, and i've experienced this problem with Zephyr 3.1.0-rc3 on two different boards (nordic nRF5340DK and LaunchXL-cc1352r1).

carlescufi, 2022-06-03:
@torres98 are you building with NEWLIB or minimal libc? can you please tell us the exact steps to reproduce?

carlescufi, 2022-06-03:
Also, you are allocating 1000 bytes in the stack, are you sure your `main` stack is big enough for that? Can you try with a smaller number of moving the `message` declaration outside of `main` so it's a global?

torres98, 2022-06-03:
That seemed to be the case, even though on previous releases it wasn't an issue. I'm still bugged by a similar behaviour but related to c++ exceptions and `std::ostringstream`. Do i close this issue and open a new one or simply edit this one?

carlescufi, 2022-06-03:
> That seemed to be the case, even though on previous releases it wasn't an issue. I'm still bugged by a similar behaviour but related to c++ exceptions and `std::ostringstream`. Do i close this issue and open a new one or simply edit this one?

Please close this one and open a new one.

stephanosio, 2022-06-03:
Closing this issue since it is due to a stack overflow.