zephyrproject-rtos/zephyr #46268: Update RNDIS USB class codes for aoutomatic driver loading by Windows

msasdk, 2022-06-03:
**Is your enhancement proposal related to a problem? Please describe.**
The current implementation of RNDIS is identified by windows as a serial device (matches usbser.inf) and windows fails to load the RNDIS driver automatically. User has to manually find a working driver. (Network adapters > Microsoft > Remote NDIS Compatible Device, on Windows 10). 

**Describe the solution you'd like**
Use the [USB-IF reserved class codes ](https://www.usb.org/defined-class-codes#anchor_BaseClassEFh) for RNDIS (0xEF 0x04 0x01) instead of the current values in [function_rndis.c](https://github.com/zephyrproject-rtos/zephyr/blob/5113c1418d2b9089be39a7e7dc6601f36e468193/subsys/usb/device/class/netusb/function_rndis.c) (.iad and .if0). This makes Windows load the RNDIS driver automatically.

**Additional context**
Both current implementation and proposed change works in Linux.

enthunilu, 2022-08-16:
I also faced the same issue. And after changing class codes, my nucleo_f429zi board is getting detected as RNDIS compatible device.
Mentioned solution should get added to zephyr main branch.

msasdk, 2022-11-04:
@henrikbrixandersen don't really agree with the label "Windows Support (Related to build Zephyr on Windows)". This has nothing to do with building on Windows but rather what a Zephyr based USB RNDS gadget is detected as on Windows.

GeorgeCGV, 2022-12-09:
Host: Windows 10 Version 21H2.
Device: custom board with STM32H730.

Tried the suggestion:

```
.bFunctionClass = USB_BCC_MISCELLANEOUS,
.bFunctionSubClass = 4, 
.bFunctionProtocol = 1,
```

It is also suggested in #47865 by @enthunilu for whom it seems to work as well.

Maybe @finikorg (_author_) could provide more information regarding used class (`USB_BCC_CDC_CONTROL`), subclass (`6`), and protocol `1`.

Microsoft has the following [USB RNDIS example](https://learn.microsoft.com/en-us/azure/rtos/usbx/usbx-device-stack-supplemental-2). But it differs in other things as well.

msasdk, 2022-12-09:
@GeorgeCGV As stated in the description, it technically it works on Windows, you just need to pick the driver manually, further, it works out of the box on Linux, hence the reasoning behind the enhancement label.

jfischer-no, 2022-12-09:
I close it because Zephyr's RNDIS Class/SubClass/Protocol codes are already the same as described in https://learn.microsoft.com/en-us/windows-hardware/drivers/network/interface-descriptor-for-communication-class-interface. If someone has a need for discussion please do it here https://github.com/zephyrproject-rtos/zephyr/discussions/51765

closed 2022-12-09 by jfischer-no

finikorg, 2022-12-09:
Now I do not remember how did I select this class, probably by reading the sample above. For selecting MS driver I used OS Descriptors. I had a presentation: https://static.sched.com/hosted_files/osseu19/6a/WYOG-eng-v15.pdf

GeorgeCGV, 2022-12-09:
@msasdk, indeed.
@jfischer-no, interesting as https://learn.microsoft.com/en-us/azure/rtos/usbx/usbx-device-stack-supplemental-2 descriptors differ from https://learn.microsoft.com/en-us/windows-hardware/drivers/network/interface-descriptor-for-communication-class-interface.

However, with 
```
.bFunctionClass = USB_BCC_MISCELLANEOUS,
.bFunctionSubClass = 4, 
.bFunctionProtocol = 1,
```
and 
```
CONFIG_USB_DEVICE_NETWORK_ECM=n
CONFIG_USB_DEVICE_NETWORK_RNDIS=y
CONFIG_USB_COMPOSITE_DEVICE=y
```
everything works out of the box.

`CONFIG_USB_COMPOSITE_DEVICE=y` solves another error related to the device descriptor when IAD is used.

jfischer-no, 2022-12-09:
> @msasdk, indeed. @jfischer-no, interesting as https://learn.microsoft.com/en-us/azure/rtos/usbx/usbx-device-stack-supplemental-2 descriptors differ from [...]

@GeorgeCGV Please move your comment to https://github.com/zephyrproject-rtos/zephyr/discussions/51765