zephyrproject-rtos/zephyr #46284: ring buffer in item mode crashes

CodingGhost, 2022-06-05:
I have written a small demo to showcase the problem:

```
#define LOG_LEVEL LOG_LEVEL_DBG

#include <zephyr.h>
#include <logging/log.h>
#include <SEGGER_RTT.h>
#include <sys/ring_buffer.h>

LOG_MODULE_REGISTER(main);

struct k_timer periodictimer;

struct ringbufData
{
	struct ring_buf rb;
	uint32_t buffer[64];
} rx_ringbufdata;

void periodicISR1KHz(k_timer *timer_id)
{
	unsigned char burstBuffer[32] = {};
	static int counter = 0;
	LOG_INF("test");
	ring_buf_item_put(&rx_ringbufdata.rb, counter, burstBuffer[5], NULL, 0);
}
void main(void)
{
	LOG_INF("--ZepyrOS booted succesfully--");
	ring_buf_init(&rx_ringbufdata.rb, sizeof(rx_ringbufdata.buffer), rx_ringbufdata.buffer);
	k_timer_init(&periodictimer, periodicISR1KHz, NULL);
	k_timer_start(&periodictimer, K_MSEC(2000), K_MSEC(100));
}
```

```
#####--> C++ config <--#####
CONFIG_NEWLIB_LIBC=y
CONFIG_CPLUSPLUS=y
CONFIG_LIB_CPLUSPLUS=y
CONFIG_STD_CPP17=y
CONFIG_DEBUG_OPTIMIZATIONS=n
CONFIG_SIZE_OPTIMIZATIONS=n
#####--> log config <--#####
CONFIG_USE_SEGGER_RTT=y
CONFIG_RTT_CONSOLE=y
CONFIG_UART_CONSOLE=n
CONFIG_LOG=y
CONFIG_LOG_PROCESS_THREAD=y
CONFIG_LOG_BACKEND_UART=n
CONFIG_LOG_DEFAULT_LEVEL=3
CONFIG_LOG_STRDUP_BUF_COUNT=200
CONFIG_LOG_STRDUP_MAX_STRING=200
#####--> kernel config <--#####
CONFIG_RING_BUFFER=y

```

when I run this program, I first get about 30 seconds of "test" and then I get a precise data bus error.
the behaviour changes when disabling c++, it then no longer takes about 30 seconds but a mere 10 seconds.
I originally stumbled across this issue whem my program crashed on allocating new heap for a stl container while uart was filling a ringbuffer in the background. I couldnt figure out what was happening so I wrote this example and got it to crash even without the stl container allocation.

Im on zephyr 2.7.1 currently. that should be the LTS i think?
this bug is code breaking at the moment!
(It still could be my fault, if thats the case please tell me where my problem is and I will close the issue asap.)

andyross, 2022-06-05:
@npitre was in the ring buffer code recently and might have input.   Which platform are you seeing this on?

CodingGhost, 2022-06-05:
nRF52840

CodingGhost, 2022-06-05:
I solved it and it was mostly my fault.
ring_buf_init(&rx_ringbufdata.rb, sizeof(rx_ringbufdata.buffer), rx_ringbufdata.buffer);
should really be
ring_buf_init(&rx_ringbufdata.rb, ARRAY_SIZE(rx_ringbufdata.buffer), rx_ringbufdata.buffer);

the thing is, I only did this because its described that way in the docs:
https://docs.zephyrproject.org/3.0.0/reference/data_structures/ring_buffers.html
look at the first code snipped at Defining a ringbuffer.

so It really is a bug in the documentation, and not in Zephyr itself.

fabiobaltieri, 2022-06-15:
Closing down since it seems solved.

closed 2022-06-15 by fabiobaltieri