zephyrproject-rtos/zephyr #46297: SPI STM32 - DMA optimalization

paveldedourek-eaton, 2022-06-06:
**Is your enhancement proposal related to a problem? Please describe.**
I found timing problem with **spi_ll_stm32** driver was used and DMA was enabled. The problem is, when the same amount of data needs to be transferred periodically with short period (transfer few bytes every new transfer in few tens of us). I mean, that issue is there, when the time for initialization takes longer that DMA transfer itself.

transceive_dma() is developed in the way that SPI and DMA is always initialized from the scratch which takes time and it is not needed to do it when the same amount of data are going to transfer - spi_dma_move_buffers() takes ~20 us @ CPU=168MHz.

**Describe the solution you'd like**
Solution will be, that driver knows if initialization needs to be done inside or not. User can do initialization only once. 
I tried to fix it, but there is still room to improve it.
[spi_dma_performance_nucleo_32f429zi](https://github.com/paveldedourek-eaton/zephyr/tree/spi_dma_performance_nucleo_32f429zi)

**Describe alternatives you've considered**
NA

**Additional context**
NA

erwango, 2022-06-06:
@paveldedourek-eaton Thanks for reporting this. If you have a solution, would you be able to push a PR directly?

paveldedourek-eaton, 2022-06-07:
@erwango My code is not tested and even this, is not correctly implemented from the entire system point of view. There is missing function/parameter which tell to the DMA, that next time, initialization can be skipped. Function spi_transceive() does not have such parameter. I guess, the driver needs other function which can configure it. We cannot use init/config function, because you can decide it in runtime that which SPI transactions needs every time DMA init or which does NOT need it.

erwango, 2023-10-12:
@teburd By any chance, in the latests DMA evolutions, is there any solution to address this ?

teburd, 2023-10-12:
if I'm understanding the issue, at the moment each time spi_transcieve() is called, and for every normalized tx/rx/txrx (lengths are equal) dma_config() is done under the assumption that an entirely new transfer needs to be setup. dma_config() in this setup is considered costly. 

dma_reload() could possibly be used here in place of re-configuring entirely, there's no requirement that dma_config() be called every time before dma_start(). You could go so far as to only call dma_config() and dma_start() once for all time, and forever more call dma_reload in this case it seems.

Would dma_reload() be enough in this case?

erwango, 2024-02-07:
@paveldedourek-eaton Any feedback on above suggestion ?

theadib, 2026-11-08:
I am also stumbled about the poor performance of the STM32 SPI driver.
In my case I want to interface an radio transceiver chip. During the access you have to access individual registers.
Means more small transfers <4 bytes. usually you give an address header and then transfer data to or from that register.
I made comparison of SPI transfer of a) 4byte transmit, 2) 2byte transmit then 2byte receive, c) 4byte transmit & receive same time.
in polling mode vs DMA mode.
Zephyr does not handle pure tx differently than tx/rx so case a) and b) are always same time.
result polling: a) 23usec, b) 24,5usec
result DMA: a) 58usec, b) 106usec
System: STM32 @120MHz, SPI @3,75MHz

This is quite shocking waste of time for DMA
The DMA re-initialization in b) costs 43usec. This is more than the transfer in polling mode itself.

I also tested using 2+40bytes, but also the DMA consumes more than it wins.

In my opinion the driver:
**- should use the DMA only for larger payloads maybe configurable by kconfig,**
  currently when DMA is configured all other code path is #ifdef out
**- should make less testing in code-path to improve performance in "standard" path**
  it would be better to have separate functions for the transmit-only or half-duplex  code path.
  the polling branch could be more performant

[image]

[image]