zephyrproject-rtos/zephyr #46301: HAL Module Request: hal_hpmicro

willChuai, 2022-06-06:
## Origin

Zephyr HAL HPMicro project.
Sources to be added:
https://github.com/hpmicro/zephyr_hal_hpmicro.git

## Purpose

This software provides Hardware Abstraction Layer drivers support for HPMicro HPM6400_6700 family SoCs.

## Mode of integration

Please integrate this project as a module according to the generic Zephyr HAL modules structure.

## Pull Request

N/A (new project)

## Description

This project provides HPM6700_6400 family peripheral drivers support (such as GPIO, UART, PWM, I2C, SPI etc..), and it is going to be used by HPM6750 SoC and HPM6750EVKMINI board in Zephyr.

No other projects available, no better options.

# Dependencies

None.

## Revision

4f07ff60858add71f73955c4238508397b117759

## License

BSD-3-Clause

stephanosio, 2022-06-06:
https://github.com/hpmicro/zephyr_hal_hpmicro.git does not seem to be accessible. If it is a private repository, please make it public since we have no way of viewing it.

Also there needs to be a Zephyr pull request using this module before this module repository can be created (the `west.yml` in the pull request can point to the provided URL until the upstream repository is created).

willChuai, 2022-06-07:
Done, please review.

stephanosio, 2022-06-08:
We will review this at the next TSC meeting (next Wednesday).

fkokosinski, 2022-06-15:
Hi @willChuai!

To be honest, `arch/`, `soc/`, `boards/` and `'driver/` directories content look as if it should be merged into the main Zephyr tree and not kept separate in a HAL module. There is also a FreeRTOS (?) directory in `middleware/` - why is that the case?

willChuai, 2022-06-15:
- hpm_sdk is a development environment provided by hpmicro to complete development work on the hpmicro family of chips, so FreeRTOS and other middleware will be available in hpm_sdk. 
- By introducing it as a whole into zephyr, we are mainly concerned with maintainability, so that when hpmicro releases a version, all new changes can be easily synchronised and updated into zephyr. 
- The structure of hpm_sdk refers to zephyr, but the contents of `arch/`, `soc/`, `boards/` and `driver/` do not match zephyr, so it cannot simply be integrated into the main zephyr tree. 
 - When the first version of zephyr is complete and the development process is fixed, we plan to put more staff into improving `driver/` and `boards/`.

willChuai, 2022-06-15:
> Hi @willChuai!
> 
> To be honest, `arch/`, `soc/`, `boards/` and `driver/` directories content look as if it should be merged into the main Zephyr tree and not kept separate in a HAL module. There is also a FreeRTOS (?) directory in `middleware/` - why is that the case?

- hpm_sdk is a development environment provided by hpmicro to complete development work on the hpmicro family of chips, so FreeRTOS and other middleware will be available in hpm_sdk. 
- By introducing it as a whole into zephyr, we are mainly concerned with maintainability, so that when hpmicro releases a version, all new changes can be easily synchronised and updated into zephyr. 
- The structure of hpm_sdk refers to zephyr, but the contents of `arch/`, `soc/`, `boards/` and `driver/` do not match zephyr, so it cannot simply be integrated into the main zephyr tree. 
 - When the first version of zephyr is complete and the development process is fixed, we plan to put more staff into improving `driver/` and `boards/`.

henrikbrixandersen, 2022-06-23:
> * hpm_sdk is a development environment provided by hpmicro to complete development work on the hpmicro family of chips, so FreeRTOS and other middleware will be available in hpm_sdk.
> * By [...]

On the other hand, having Zephyr-specific code in an imported HAL module introduces the burden of maintaining this on anybody doing changes in the main Zephyr tree (e.g. if changing a Zephyr API, which is being used in your HAL, the author now needs to perform changes in two places, within Zephyr and within your HAL).

Note that the approach taken here is different to all the other vendor HALs included in Zephyr.

> * The structure of hpm_sdk refers to zephyr, but the contents of `arch/`, `soc/`, `boards/` and `driver/` do not match zephyr, so it cannot simply be integrated into the main zephyr tree.

Is the current structure of the code a requirement from HP Micro or could it be changed to match that of Zephyr, so that the code can be included directly?

> * When the first version of zephyr is complete and the development process is fixed, we plan to put more staff into improving `driver/` and `boards/`.

Sounds good.

willChuai, 2022-06-23:
> On the other hand, having Zephyr-specific code in an imported HAL module introduces the burden of maintaining this on anybody doing changes in the main Zephyr tree (e.g. if changing a Zephyr API, [...]
Note that the approach taken here is different to all the other vendor HALs included in Zephyr.

My previous description was unclear, hal_hpmicro does not contain the zephyr code and API. So zephyr related changes will only be done in zephyr. Although the entire hpm_sdk is added to zephyr, the only files that are compiled into zephyr are the chip-related content.  To remove the extra content from hpm_sdk would affect the speed of the release.

> Is the current structure of the code a requirement from HP Micro or could it be changed to match that of Zephyr, so that the code can be included directly?

We hope to keep the current form, but if zephyr insists we can delete the `middleware/` , `samples/` and other unused directories.

carlescufi, 2022-06-27:
I have looked at the [proposed HAL](https://github.com/hpmicro/zephyr_hal_hpmicro/tree/manifest-rev/hpm_sdk) and I do not see anything there that is particularly troublesome personally. As @willChuai explains, this is an SDK which contains more than just the HAL, but the actual contents used by Zephyr will be a subset. We have two options here:

- Ask HPMicro to pull only the files actually needed by Zephyr into `hal_hpmicro`
- Use their SDK repo as-is, but Zephyr will not use all of it

AFAIK Zephyr doesn't currently have a policy regarding whether we should pull only portions of existing SDKs, but it is true that most vendors do pull only the files that are needed.

willChuai, 2022-06-27:
> @willChuai can you and/or HPMicro commit to maintaining the integration of this HAL with Zephyr for the foreseeable future?

Of course, HPMicro and I will keep the integration of hal_hpmicro and zephyr.

henrikbrixandersen, 2022-06-28:
> I have looked at the [proposed HAL](https://github.com/hpmicro/zephyr_hal_hpmicro/tree/manifest-rev/hpm_sdk) and I do not see anything there that is particularly troublesome personally. As [...]

The HAL contains Zephyr-specific files, e.g. Zephyr board YAML files (https://github.com/hpmicro/zephyr_hal_hpmicro/blob/manifest-rev/hpm_sdk/boards/hpm6360evk/hpm6360evk.yaml), Zephyr-dependent CMake function calls (https://github.com/hpmicro/zephyr_hal_hpmicro/blob/d652edd06e71286be560b9a918c0db9d47962521/CMakeLists.txt#L14), a lot of sample code unrelated to Zephyr (https://github.com/hpmicro/zephyr_hal_hpmicro/tree/manifest-rev/hpm_sdk/samples), and quite a lot of middleware unused by Zephyr (https://github.com/hpmicro/zephyr_hal_hpmicro/tree/manifest-rev/hpm_sdk/middleware).

Personally, I think this is too much overhead to accept as a Zephyr HAL module.

carlescufi, 2022-06-29:
@willChuai In the TSC there was an issue raised with your HAL: It contains too much beyond the actual HAL, especially the `middleware`. Can you try an approach similar to what STM does, with an importer script that filters out the parts that are not needed for Zephyr? https://github.com/zephyrproject-rtos/hal_stm32/blob/main/scripts/update_stm32_package.py

A script of some sort that takes the latest SDK release and pulls the files that you are interested in.

willChuai, 2022-06-30:
> @willChuai In the TSC there was an issue raised with your HAL: It contains too much beyond the actual HAL, especially the `middleware`. Can you try an approach similar to what STM does, with an [...]

OK, we'll remove the unused content.This will take some time.

willChuai, 2022-06-30:
> The HAL contains Zephyr-specific files, e.g. Zephyr board YAML files (https://github.com/hpmicro/zephyr_hal_hpmicro/blob/manifest-rev/hpm_sdk/boards/hpm6360evk/hpm6360evk.yaml), Zephyr-dependent [...]

Okay, revising now.

willChuai, 2022-07-21:
Revision completed, please review, thanks.

fkokosinski, 2022-07-21:
Thanks, it looks much better without e.g. the `middleware` directory.

I'm wondering - why can't much of the HAL be moved and merged directly into the main Zephyr tree? Some parts, i.e. files in [hpm_sdk](https://github.com/hpmicro/zephyr_hal_hpmicro/tree/manifest-rev/hpm_sdk)/[drivers](https://github.com/hpmicro/zephyr_hal_hpmicro/tree/manifest-rev/hpm_sdk/drivers/src) look like they should be moved to drivers directly, instead of being in a HAL that is stored in a separate repo. That would make e.g. bisecting much easier in the future.

I know that you've responded to this suggestion before with:
> By introducing it as a whole into zephyr, we are mainly concerned with maintainability, so that when hpmicro releases a version, all new changes can be easily synchronised and updated into zephyr.

But that's generally how open source works: you want to update something that you had contributed in the past, you just contribute again. If you only have Zephyr-related stuff in your HAL, I don't really see how _synchronizing_ your work would be made more difficult (if you follow Zephyr's coding guidelines etc.) if hpmicro-related drivers would be directly in Zephyr.

willChuai, 2022-07-21:
> Thanks, it looks much better without e.g. the `middleware` directory.
>
> I'm wondering - why can't much of the HAL be moved and merged directly into the main Zephyr tree? Some parts, i.e. files in [...]

The hpm_sdk/drivers package is already very thin, it corresponds directly to the mcu registers, without this layer it would do the register operations directly, which should be consistent with most vendors. So the final structure is that zephyr/drivers calls the hpm_sdk/drivers interface.

> I know that you've responded to this suggestion before with:
>
>
> But that's generally how open source works: you want to update something that you had contributed in the past, you just contribute [...]

We are active in the open source community and hpm_sdk itself is open source. HPMicro does not support many zephyr drivers and we plan to first complete an initial version and then invest more staff in adding drivers, as without a proper version we think we may have ineffective work. Within the next two months we will be adding adc, pwm, spi, i2c and other basic drivers.

willChuai, 2022-11-26:
Hi all, any new progress on this issue?

willChuai, 2023-04-28:
Can you add TSC tags?

Cherish-Gww, 2023-04-28:
@fkokosinski 
The reason why hpmicro needs an independent repository is because hpmicro has already produced multiple models and series of MCUs, not only HPM6300 and HPM6750.

fkokosinski, 2023-04-28:
> Can you add TSC tags?

Hi, I took a look at [TSC meeting notes from 2022](https://docs.google.com/document/d/1_sFM1BTLHrwRA9pbNoD7XPHE3OuqXT0P04AlAS2dvTk/edit#) and it looks like the motion passed (page 124) on 3 August 2022, and that's why the `TSC` label was removed:
```
HAL Module Request: hal_hpmicro
Cleaned up from before
Vote to approve HAL Module Request: hal_hpmicro #46301
Motion: Carles
Second: Keith
No objections raised. Motion passes.
```

Cherish-Gww, 2023-04-28:
> Hi, I took a look at [TSC meeting notes from 2022](https://docs.google.com/document/d/1_sFM1BTLHrwRA9pbNoD7XPHE3OuqXT0P04AlAS2dvTk/edit#) and it looks like the motion passed (page 124) on 3 August [...]

what`s the next?

stephanosio, 2023-05-02:
@willChuai Can you confirm that the content of https://github.com/hpmicro/zephyr_hal_hpmicro should be used to create the Zephyr `hal_hpmicro` module repository?

Once you confirm, I will go ahead and create the repository for it.

willChuai, 2023-05-04:
> @willChuai Can you confirm that the content of https://github.com/hpmicro/zephyr_hal_hpmicro should be used to create the Zephyr `hal_hpmicro` module repository?
> 
> Once you confirm, I will go ahead and create the repository for it.

Thanks, I confirm.

stephanosio, 2023-05-04:
https://github.com/zephyrproject-rtos/hal_hpmicro has been created.

closed 2023-05-04 by stephanosio