zephyrproject-rtos/zephyr #46312: sample: bluetooth: ipsp - TCP not running over IPSP

icatorze, 2022-06-06:
I am running IPSP sample on a nucleo_f207zg board and a nRF52833-dk as BLE hci_uart. 

I made every configuration discussed at [IPSP Sample README](https://docs.zephyrproject.org/latest/samples/bluetooth/ipsp/README.html) and everything works as expected but TCP connection NOT.

There are some old issues about tcp connection problems but no one has a clear aswser.

I reproduce the problem with other hardware sets with same result.

**To Reproduce**
- I made an overlay file to configure USARTs properly.
- I made a conf file to include shell, log, increase bt_rx_stack and disable onboard ethernet:
CONFIG_SHELL=y
CONFIG_NET_LOG=y
CONFIG_NET_SHELL=y
CONFIG_BT_RX_STACK_SIZE=4096
CONFIG_NET_IF_MAX_IPV6_COUNT=2
CONFIG_ETH_STM32_HAL=n

- west build -p -b nucleo_f207zg -d build/ipsp_207_3.1 samples/bluetooth/ipsp/ -- -G"Eclipse CDT4 - Ninja"
- west flash -d build/ipsp_207_3.1

On the other side I am running a Debian 11 (5.10.106-1 (2022-03-17) x86_64 GNU/Linux) with a BLE 4.2 dongle:
- hcitool lescan|grep Test
After take MAC address I used it in:
- echo "connect __mac:get:in:last:step__ 2" > /sys/kernel/debug/bluetooth/6lowpan_control
- ip address add 2001:db8::2/64 dev bt0

Now i can ping from Linux to Nucleo board and vice-versa.

UDP connection works fine:
$ ncat -6 -u 2001:db8::1 4242
teste
teste

TCP connection never complete and goes out with timeout:
$ ncat -6  2001:db8::1 4242
Ncat: TIMEOUT.

**Expected behavior**
ncat or telnet command must connect and change echo messages.

**Logs and console output**
With telnet or ncat, i can see in wireshark TCP packet SYN trying to reach device but never goes there. 

On other hand, when I try a tracert6 SYN packet reach device as expected.
$ tracert6 -i bt0 -S -n -p 4242 2001:db8::1
traceroute to 2001:db8::1 (2001:db8::1) from 2001:db8::2, port 4242, from port 55537, 30 hops max, 60 bytes packets
 1  2001:db8::1  363.427 ms [open]  100.416 ms  99.730 ms

**Environment (please complete the following information):**
 - OS: Debian 11 (5.10.106-1 (2022-03-17) x86_64 GNU/Linux) 
 - Zephyr version: 3.1.99
 - Toolchain zephyr 0.14.1

**Attachments**
[Wireshark_ble_not_connect.pcapng.gz](https://github.com/zephyrproject-rtos/zephyr/files/8845886/Wireshark_ble_not_connect.pcapng.gz)
[Wireshark_ble_tracert6_connect.pcapng.gz](https://github.com/zephyrproject-rtos/zephyr/files/8845887/Wireshark_ble_tracert6_connect.pcapng.gz)

fabiobaltieri, 2022-06-15:
@jhedberg any chance you could take a look or reassign this one?

jhedberg, 2022-06-15:
This sounds like a likely network stack issue, since Bluetooth shouldn’t even be aware of what’s going on in these higher protocol layers.

rlubos, 2022-06-15:
There was a conclusion in a similar report some time ago, that the problem with TCP connection over IPSP lies on the linux side (https://github.com/zephyrproject-rtos/zephyr/issues/33185#issuecomment-799002558).

carlescufi, 2022-06-15:
> There was a conclusion in a similar report some time ago, that the problem with TCP connection over IPSP lies on the linux side ([#33185 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/33185#issuecomment-799002558)).

@icatorze closing as duplicate of #33185, the issue seems to be on the Linux side so please report it there.

closed 2022-06-15 by carlescufi