zephyrproject-rtos/zephyr #46322: Time units in shtcx sensor

androbi-com, 2022-06-06:
I have been having problems reading the shtc3 sensor. It would work from the debugger but not when running without debugger. I suspected some timing issue and noted in file https://github.com/zephyrproject-rtos/zephyr/blob/f875d027a04374fcf07661809f81a42c2c646629/drivers/sensor/shtcx/shtcx.c#L33 that time units don't seem to be consistent. If 14400 is 14.4ms, 1210 is _not_ 12.1 ms as indicated in the comments, but 1.21 ms.

When changing the constant to 12100 my program works so I suspect a little bug here. Or should the delay only be be 1.21ms??

fabiobaltieri, 2022-06-15:
@str4t0m I think you wrote this one, can you look into this?

androbi-com, 2022-06-15:
In the meantime I checked with the datasheet, the delay should indeed be 12100 (12.1ms)

str4t0m, 2022-06-15:
Thanks, this should be indeed `12100`. I have opened a PR that fixes it (#46574).

PR #46574 (2022-06-15): drivers: shtc3: fix normal mode measurement time.
The measurement time in normal mode was too short.
With this commit the maximum value of the datasheet is applied.

Fixes #46322
closed 2022-06-15 by MaureenHelm