zephyrproject-rtos/zephyr #46324: it8xxx2_evb: tests/kernel/sleep/ fail due to k_sleep(K_MSEC(100)) not correct

hunterhu3000, 2022-06-07:
**Describe the bug**
The daily test case tests/kernel/sleep/ failed due to k_sleep(K_MSEC(100)) not correct

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?: it8xxx2_evb
 - What have you tried to diagnose or workaround this issue? Add log before and after k_sleep(K_MSEC(100)), the log after it is never seen

**To Reproduce**
Two ways to reproduce the behavior:
Method 1
1. twister -v -p it8xxx2_evb --device-testing --west-flash="../../itetool/loader.sh" --device-serial=/dev/ttyUSB0 -T tests/kernel/sched/schedule_api/
Method 2
1. west build -p always -b it8xxx2_evb tests/kernel/sleep/ -DCONFIG_TIMESLICE_PER_THREAD=y -DCONFIG_TIMESLICING=y
2. sudo -S ~/itetool/ite -f build/zephyr/zephyr.bin -s check

**Expected behavior**
The test can pass

**Impact**
Blocks all the daily tests under tests/kernel/sleep/

**Logs and console output**
Logs for west:
```
*** Booting Zephyr OS build zephyr-v3.1.0-179-g774a3cd756b4  (delayed boot 1ms)                                                                               ***
Running TESTSUITE sleep
===================================================================
START - test_sleep
Kernel objects initialized
Test thread started: id = 0x80101100
Helper thread started: id = 0x80101048
Testing normal expiration of k_sleep()
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): ubuntu
 - Toolchain (e.g Zephyr SDK, ...): 0.14.1
 - Commit SHA or Version used: f875d027a04374fcf07661809f81a42c2c646629

RuibinChang, 2022-06-07:
Why do you enable CONFIG_TIMESLICE_PER_THREAD?
it8xxx2_evb default isn't set this config.

hunterhu3000, 2022-06-07:
> Why do you enable CONFIG_TIMESLICE_PER_THREAD? On it8xxx2_evb, default doesn't set this config.

One of the test case defined in the testcase.yaml contains this parameter
```
  kernel.scheduler.slice_perthread:
    filter: not CONFIG_SCHED_MULTIQ
    extra_configs:
      - CONFIG_TIMESLICING=y
      - CONFIG_TIMESLICE_PER_THREAD=y
    tags: kernel threads sched userspace ignore_faults
```
If you use west build without -DCONFIG_TIMESLICE_PER_THREAD=y, the case also fails.

RuibinChang, 2022-06-13:
@hunterhu3000
I update the SDK to latest version v0.14.2 on commit https://github.com/zephyrproject-rtos/zephyr/commit/b9e833366f1360ecbad8776d6e8587e180688e1b, the 10 times result of kernel k_sleep() are pass.

RuibinChang, 2022-06-17:
Running `west build -p always -b it8xxx2_evb tests/kernel/sched/schedule_api` on SDK v0.14.2 and commit f232db463078ff3e8a85ceeb9e9fcc0c14e141e3, the 10 times result of schedule_api are pass.

hunterhu3000, 2022-06-29:
@RuibinChang
Sorry for late reply
Twist and west are different. When using twister to do the test, it will test many different configurations, but for west, it conly contains 1 most common config. For example the test case of tests/kernel/sched/schedule_api, there are 8 test cases defined in tests/kernel/sched/schedule_api/testcase.yaml. There are 7 test cases suitable for it8xxx2_evb. Among these 7, only 1 is failed. I copy the failed test below
```
  kernel.scheduler.slice_perthread:
    filter: not CONFIG_SCHED_MULTIQ
    extra_configs:
      - CONFIG_TIMESLICING=y
      - CONFIG_TIMESLICE_PER_THREAD=y
    tags: kernel threads sched userspace ignore_faults
```
As you can see, this case defined two more macros, so if you want to test it in west you need to add these macros, like this:
```
west build -p always -b it8xxx2_evb tests/kernel/sched/schedule_api -DCONFIG_TIMESLICE_PER_THREAD=y -DCONFIG_TIMESLICING=y
```

If you think any of these macros does not fit for it8xxx2_evb, or we do not need to test these macros, we can skip this test for it8xxx2_evb.

RuibinChang, 2022-07-01:
If we enable CONFIG_TIMESLICE_PER_THREAD, the kernel operation 
seems that doesn't fit for it8xxx2_evb. Since we don't support CONFIG_TIMESLICE_PER_THREAD in default, so I think we can skip this test.
If we support CONFIG_TIMESLICE_PER_THREAD in the future, then we do this test.

Dino-Li, 2022-07-07:
@RuibinChang can you investigate that why we can't enable the CONFIG_TIMESLICE_PER_THREAD option on it8xxx2_evb?

hunterhu3000, 2022-07-13:
@RuibinChang any update for this issue? Thanks

RuibinChang, 2022-07-13:
Because we do not support CONFIG_TIMESLICE_PER_THREAD, curr->base.slice_ticks of slice_time() is a large number (should be zero), then _current_cpu->slice_ticks of z_reset_time_slice() will be set a large ticks remaining in current time slice.

hunterhu3000, 2022-07-13:
@RuibinChang So the test of 
```
west build -p always -b it8xxx2_evb tests/kernel/sched/schedule_api -DCONFIG_TIMESLICE_PER_THREAD=y -DCONFIG_TIMESLICING=y
```
is invalid on it8xxx2_evb, right?

RuibinChang, 2022-07-13:
Yes

RuibinChang, 2022-07-13:
@hunterhu3000  Do others riscv platform are default DCONFIG_TIMESLICE_PER_THREAD=y? Or do you know which riscv platform is default DCONFIG_TIMESLICE_PER_THREAD=y?

hunterhu3000, 2022-07-13:
According to the current code, all the riscv board will test with DCONFIG_TIMESLICE_PER_THREAD=y.
But we only have 1 riscv board in the daily test, that is it8xxx2_evb. So we do not know if other riscv board can pass this test.

hunterhu3000, 2022-07-27:
z_tick_sleep->z_add_thread_timeout->z_add_timeout->next_timeout->_current_cpu->slice_ticks

The _current_cpu->slice_ticks is not initialized in the test of test_priority_cooperative().  So the swap of the thread got problem.
I'll provide a PR to fix it.

RuibinChang, 2022-07-28:
Re# Could you please help to check? The function "riscv_idle" is in soc/riscv/riscv-ite/it8xxx2/soc.c.

Sorry for the late reply.
I checked that event timer interrupt is enabled. I also compared the test_priority_cooperative() test between CONFIG_TIMESLICE_PER_THREAD = y/n, and found that kernel never sets event timer count when running test_priority_cooperative() with CONFIG_TIMESLICE_PER_THREAD = y, so EC won't be wakeup by event timer interrupt in riscv_idle().

hunterhu3000, 2022-07-28:
> next_timeout

The problem happens in next_timeout()
```
#ifdef CONFIG_TIMESLICING
	if (_current_cpu->slice_ticks && _current_cpu->slice_ticks < ret) {
		ret = _current_cpu->slice_ticks;
	}
#endif
```
_current_cpu->slice_ticks is not initialized

So similar to test "test_time_slicing_preemptible", I add 
```
#ifdef CONFIG_TIMESLICING
        k_sched_time_slice_set(200, 0); /* 200 ms */
#endif
```

PR #48425 (2022-08-04): kernel: Init the base.slice_ticks for dummy_thread
Fixes #46324
Set dummy_thread->base.slice_ticks to 0 when
CONFIG_TIMESLICE_PER_THREAD is set. To avoid
_current_cpu->slice_ticks be a big number.
commit 57487622 (2022-08-04): kernel: Init the base.slice_ticks for dummy_thread

Fixes #46324
Set dummy_thread->base.slice_ticks to 0 when
CONFIG_TIMESLICE_PER_THREAD is set. To avoid
_current_cpu->slice_ticks be a big number.
closed 2022-08-04 by nashif