zephyrproject-rtos/zephyr #46325: ESP32 strcmp error while enable MCUBOOT and NEWLIB_LIBC

sam131208, 2022-06-07:
**Describe the bug**
Get the strcmp error while enable MCUBOOT and NEWLIB_LIBC.
It seems that there are still some issues with libc that have not been resolved。

**Logs and console output**
=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2022.06.07 10:28:39 =~=~=~=~=~=~=~=~=~=~=~=
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff8598,len:4908
ho 0 tail 12 room 4
load:0x40093000,len:13792
load:0x4009b800,len:6004
entry 0x4009343c
[esp32] [INF] Enabling RNG early entropy source...
[esp32] [INF] *** Booting MCUboot build v1.9.0-1617-ge7415b5 ***
[esp32] [INF] Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[esp32] [INF] Scratch: magic=bad, swap_type=0x0, copy_done=0x2, image_ok=0x2
[esp32] [INF] Boot source: primary slot
[esp32] [INF] Swap type: none
[esp32] [INF] Disabling RNG early entropy source...
[esp32] [INF] br_image_off = 0x10000
[esp32] [INF] ih_hdr_size = 0x20
[esp32] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32] [INF] DRAM segment: start=0x11d86, size=0x2b4c, vaddr=0x3ffbdb60
[esp32] [INF] IRAM segment: start=0x148d4, size=0x183e4, vaddr=0x40080000
[esp32] [INF] start=0x4008ffd8
*** Booting Zephyr OS build zephyr-v3.1.0-165-g90537fe447d6  ***
Starting Bluetooth Peripheral example

Backtrace:0x40091003:0x3FFEB210 0x400E3643:0x3FFEB220 0x400D5936:0x3FFEB240 0x400D23D7:0x3FFEB280 0x4008C03F:0x3FFEB2A0 0x400E2607:0x3FFEB2C0 

 .text.strchr   0x0000000040090f2c       0xd4 /home/sam/zephyr-sdk-0.14.2/xtensa-espressif_esp32_zephyr-elf/xtensa-espressif_esp32_zephyr-elf/lib/./libc.a(lib_a-strchr.o)
                0x0000000040090f2c                strchr
 .text          0x0000000040091000      0x123 /home/sam/zephyr-sdk-0.14.2/xtensa-espressif_esp32_zephyr-elf/xtensa-espressif_esp32_zephyr-elf/lib/./libc.a(lib_a-strcmp.o)
                0x0000000040091000                strcmp
 *fill*         0x0000000040091123        0x1 
 .text          0x0000000040091124       0x90 /home/sam/zephyr-sdk-0.14.2/xtensa-espressif_esp32_zephyr-elf/xtensa-espressif_esp32_zephyr-elf/lib/./libc.a(lib_a-strcpy.o)
                0x0000000040091124                strcpy
 .text          0x00000000400911b4       0x63 /home/sam/zephyr-sdk-0.14.2/xtensa-espressif_esp32_zephyr-elf/xtensa-espressif_esp32_zephyr-elf/lib/./libc.a(lib_a-strlen.o)
                0x00000000400911b4                strlen
 *fill*         0x0000000040091217        0x1 
 .text.strncmp  0x0000000040091218       0xa4 /home/sam/zephyr-sdk-0.14.2/xtensa-espressif_esp32_zephyr-elf/xtensa-espressif_esp32_zephyr-elf/lib/./libc.a(lib_a-strncmp.o)
                0x0000000040091218                strncmp
 .text          0x00000000400912bc      0x113 /home/sam/zephyr-sdk-0.14.2/xtensa-espressif_esp32_zephyr-elf/xtensa-espressif_esp32_zephyr-elf/lib/./libc.a(lib_a-strncpy.o)
                0x00000000400912f4                strncpy
 *fill*         0x00000000400913cf        0x1

sylvioalves, 2022-06-07:
@sam131208 Can you share the sample code or prj.conf?

sam131208, 2022-06-08:
@sylvioalves Under file is the test project.

[DemoBoard_ESP32.zip](https://github.com/zephyrproject-rtos/zephyr/files/8858093/DemoBoard_ESP32.zip)

sam131208, 2022-06-10:
@sylvioalves Do you have updates?

sylvioalves, 2022-06-22:
@sam131208 Can you retry using latest main?

sam131208, 2022-06-23:
@sylvioalves  I build the project DemoBoard_ESP32 with latest main，get a build error。

     mcuboot_hdr:          32 B         32 B    100.00%
        metadata:          32 B         32 B    100.00%
             ROM:      390512 B    4194240 B      9.31%
     iram0_0_seg:      100456 B       128 KB     76.64%
     irom0_0_seg:      193968 B      3264 KB      5.80%
     dram0_0_seg:       83348 B     124580 B     66.90%
     dram0_1_seg:       60216 B     110032 B     54.73%
     drom0_0_seg:       25432 B    4194240 B      0.61%
    rtc_iram_seg:          0 GB         8 KB      0.00%
    rtc_slow_seg:          0 GB         4 KB      0.00%
        IDT_LIST:          0 GB         8 KB      0.00%
Usage: imgtool sign [OPTIONS] INFILE OUTFILE
Try "imgtool sign -h" for help.

Error: Invalid value for "--align": invalid choice: 32. (choose from 1, 2, 4, 8)
FATAL ERROR: command exited with status 2: /home/sam/.local/bin/imgtool sign --version 0.0.0+0 --align 32 --header-size 32 --slot-size 1048576 /home/sam/zephyrproject/zephyr/samples/sam/DemoBoard_ESP32/build/zephyr/zephyr.bin /home/sam/zephyrproject/zephyr/samples/sam/DemoBoard_ESP32/build/zephyr/zephyr.signed.bin
ninja: build stopped: subcommand failed.

sylvioalves, 2022-06-23:
@sam131208, please, can you update your imgtool: `pip3 install imgtool`. The version you are using does not support signing images aligned with 32 bytes.

sam131208, 2022-06-24:
Thanks for your help, it's already working fine.

closed 2022-06-24 by sam131208