zephyrproject-rtos/zephyr #46326: Async UART for STM32 U5 suppport

JiiZR, 2022-06-07:
**Is your enhancement proposal related to a problem? Please describe.**
Is is possible to add STM32 U575/U585 support for tests/drivers/uart/uart_async_api. I'am aware that U5 DMA support is not yet merged, https://github.com/zephyrproject-rtos/zephyr/pull/45350  but after that would be nice time.

**Describe the solution you'd like**
To be able to build and run test: west build -p auto -b b_u585i_iot02a_ns tests/drivers/uart/uart_async_api

**Describe alternatives you've considered**

 
**Additional context**

FRASTM, 2022-06-07:
Thanks for raising this PR.
The async UART for stm32u5 requires DMA : purpose of the #45350
Then I have a branch here https://github.com/FRASTM/zephyr/tree/uart_dma_u5 for running the uart_async_api. not yet finalized because of a interrupt raising unexpectedly.

FRASTM, 2022-06-08:
There is a device errata for the in the stm32U575/585
[ES0499](https://www.st.com/resource/en/errata_sheet/es0499-stm32u575xx-and-stm32u585xx-device-errata-stmicroelectronics.pdf)
that prevent the use of the DMA for USART transfer :  "USART does not generate DMA requests after setting/clearing DMAT bit
_Description
If the DMA is used for data transmission (DMAT = 1 in USART_CR3 register), and the software clears DMAT bit
and sets it again to prepare the next transmission, then the peripheral does not generate DMA requests anymore.
As a result, data are not transmitted.
Workaround
• Avoid clearing DMAT.
• If clearing DMAT is needed after the end of DMA transfers, once DMAT is cleared, disable and reenable the
peripheral through UE bit of USART_CR1 register. This workaround is acceptable only if the peripheral is not
used in receiver mode._

This is the case in the tests/drivers/uart/uart_async_api/, when test_write_abort and test_forever_timeout : the DMA request (DMAT of the USART CR3 bit) is disabled and re-enabled to start a next transmission.

FRASTM, 2022-08-30:
Please see  branch https://github.com/FRASTM/zephyr/tree/uart_async_u5 as a first step.

erwango, 2022-08-31:
@JiiZR Conclusion of this ticket is that unfortunately UART DMA RX is not functional on U5 and this is a hardware limitation.
A workaround is available on TX as described above by @FRASTM.

Closing.

JiiZR, 2022-09-05:
Thanks for the info. So this means that there will not be async UART support for U5 series. or did I missed something ?

erwango, 2022-09-05:
> Thanks for the info. So this means that there will not be async UART support for U5 series or did I miss something ?

At least it would require some heavy workaround. This is unlikely this would go upsteam.

FRASTM, 2022-10-05:
Could you please have a look at PR https://github.com/zephyrproject-rtos/zephyr/pull/50968

closed 2022-11-23 by carlescufi