zephyrproject-rtos/zephyr #46335: For ESP32, initialization of static object during declaration with derived class type doesn't work.

vermaabhishek-eaton, 2022-06-07:
- For esp32 board, if we are initializing static object with derived class type during declaration, it doesn't work.
- Same code is working with nucleo_f429zi board.
- Refer below example code to reproduce the issue.

**1) Header file content Test_Code.h**
`/* Parent class- test1 */
class test1
{
	public:
		virtual void Available( void );
};

/*  Derived class- test2 */
class test2 : public test1
{
	public:
		test2();
		~test2();
		void Available( void );
};

/* Standalone class- xyz */
class xyz
{
	public:
		xyz();
		~xyz();
		static void Allocate(void);
	private:
		static test2 m_debug_obj;
		static test1* m_abc_ptr;
};`

**2) Source file content Test_Code.cpp**
#include <zephyr.h>
#include "Test_Code.h"

test2 xyz::m_debug_obj;
test1* xyz::m_abc_ptr = &m_debug_obj;  // this initialization doesn't work for ESP32 target.

/* Derived Class function*/
test2::test2()
{}

test2::~test2()
{}

void test2::Available( void )
{
	unsigned char a = 0;
	printk("m_abc_ptr initialized correctly\n");
	a++;

}

/* Static function of Class xyz */
void xyz::Allocate( void )
{
	static unsigned char temp = 0;
/* If we remove initialization during declaration and enable below line
 * then initialization works */
//	m_abc_ptr = (&m_debug_obj);
	if ( m_abc_ptr != nullptr )
		{
			m_abc_ptr->Available();
			temp ++;
		}
}

/* Main function */
void main( void )
{
	xyz::Allocate();
	while ( 1 )
	{
		printk( "In loop\n" );
		k_msleep( 1000 );
	}
}

As a workaround for this issue we are initializing **m_abc_ptr** object through function call.

**To Reproduce**
1. Build Test_Code.cpp with Zephyr v3.0.0 for nucleo_f429zi and esp32 board.
2. Connect putty with configuration ( 115200 baudrate, 8bit, Parity-none, 1Stop bit ) and observe output logs.
3. For nucleo_f429zi, you will get "m_abc_ptr initialized correctly" log. 
4. For esp32, you will get Backtrace logs.

stephanosio, 2022-06-07:
@vermaabhishek-eaton which toolchain and version are you using?

@sylvioalves does this ring a bell?

sylvioalves, 2022-06-07:
@vermaabhishek-eaton Would you retry using v3.1.0? Please, note that ESP32 toolchain now uses Zephyr-SDK. Check [this](https://docs.zephyrproject.org/latest/boards/xtensa/esp32/doc/index.html) again and make sure `export ZEPHYR_TOOLCHAIN_VARIANT="zephyr"` is properly set.

vermaabhishek-eaton, 2022-06-07:
> @vermaabhishek-eaton which toolchain and version are you using?
> 
> @sylvioalves does this ring a bell?

espressif toolchain I have installed through "west espressif install" command. Its version is **esp-2020r3-8.4.0**

vermaabhishek-eaton, 2022-06-07:
> @vermaabhishek-eaton Would you retry using v3.1.0? Please, note that ESP32 toolchain now uses Zephyr-SDK. Check [this](https://docs.zephyrproject.org/latest/boards/xtensa/esp32/doc/index.html) again and make sure `export ZEPHYR_TOOLCHAIN_VARIANT="zephyr"` is properly set.

Yes I can try on it.

stephanosio, 2022-06-07:
Assigning this issue to @sylvioalves since this is an Espressif toolchain issue.

Please try with Zephyr 3.1.0 using the Zephyr SDK. If this is still an issue with the Zephyr SDK, you can assign it back to me.

sylvioalves, 2022-06-15:
@vermaabhishek-eaton, were you able to check this?

vermaabhishek-eaton, 2022-06-21:
> @vermaabhishek-eaton, were you able to check this?

yes, it is working with zephyr v_3.1.0 using zephyr sdk.

closed 2022-06-21 by vermaabhishek-eaton