zephyrproject-rtos/zephyr #46339: Zephyr message queue: add LIFO operation

quantum-leaps, 2022-06-07:
Most RTOSes support both FIFO and LIFO operations for the message queues. For example, ThreadX has: `tx_queue_send()` and `tx_queue_front_send()`. EmbOS has: `OS_PutMailCond()` and `OS_PutMailFrontCond()`. Currently, Zephyr has only `k_msgq_put()`.

This feature request is for adding `k_msgq_put_front()`, for LIFO operation.

quantum-leaps, 2024-02-08:
The feature request is still relevant. I'm not sure what kind of additional information is needed, as the request is straightforward and the feature is already supported by most other RTOS kernels. A compelling use case for the LIFO policy for message queues is the self-posting of messages. Typically, a thread that posts a message to self is in a certain state and is ill-prepared to handle other messages (ahead of the self-posted one) because these other messages might change the internal state. This is very common for deferring messages, for example.

andyross, 2024-02-09:
Am I missing something?  k_lifo/k_fifo are special cases of k_queue, which does indeed support operations at both ends of the list.   Closing; please reopen if the k_queue_append/prepend/remove() APIs don't meet the requirements here.

andyross, 2024-02-09:
Oh, never mind.  Reopen.  The comment above used the words LIFO and FIFO, but it was actually talking about the k_msgq API, not queues.

quantum-leaps, 2024-02-09:
This feature request is about supporting both message-posting policies (FIFO and LIFO) in the same RTOS primitive (message queue). If I understand correctly, the `k_lifo RTOS` primitive is a _different_ than `k_fifo`, which is NOT what I mean.
If it would be helpful, my feature request for the Zephyr message queue corresponds to the two policies supported by the FreeRTOS queue. These policies are documented as [xQueueSendToBack()](https://www.freertos.org/xQueueSendToBack.html) and [xQueueSendToFront()](https://www.freertos.org/xQueueSendToFront.html). I hope that this clarifies the requested semantics.

closed 2025-07-25 by cfriedt