zephyrproject-rtos/zephyr #46341: Zephyr scheduler lock: add selective locking up to a given priority ceiling

quantum-leaps, 2022-06-07:
Currently, Zephyr only supports unconditional scheduler locking `k_sched_lock()`. This is often too crude. It would be much better to lock the scheduler only up to a given priority ceiling.

For example, when implementing publish/subscribe it is necessary to lock the scheduler up to the highest-priority subscriber (otherwise the ordering of posted messages might be weird since the threads might preempt the publisher and post their own messages). A global scheduler lock is overkill and might unnecessarily impact high-priority, hard real-time threads. A selective lock with a priority ceiling would be much more appropriate.

The proposed feature is to add a parameter (the priority ceiling) to `k_sched_lock(int prio_ceiling)` so that threads above that priority ceiling won't be locked.

andyross, 2022-06-10:
The sched_lock primitive is for synchronization, not priority control.  The idea is to prevent you from being interrupted while doing something that needs to be threadsafe vs. other threads (but not interrupts!).

I guess my intuition is that this sort of makes a hash of the two metaphors: using "priority" as a cue to the kernel to "allow this thread to break locks" seems borderline dangerous to me.

And it's mostly needless anyway: if you have a thread that you don't want to be preempted by threads of priority N or less, you can use `k_thread_priority_set()` to set its priority to N while you do your important task[1].  And conversely, if you have a thread that must preempt even sched-locked or cooperative threads, you can use a MetaIRQ priority (i.e. "it acts like an interrupt").

[1] The reason sched_lock exists in addition to priority_set at all is actually performance.  `k_sched_lock()` can be implemented by an atomic decrement of the thread priority value, whereas changing the actual priority requires removing/adding the thread to the run queue while holding the scheduler spinlock (and it's a rescheduling point too, which isn't really desirable for locking).

andyross, 2022-06-11:
Think I'm going to close this while I'm grooming all my other issues.  Please reopen if you want to debate the need for a feature like this or have details about the use case.  But IMHO I think the existing API covers what you'd want to try in this area well already.

closed 2022-06-11 by andyross