zephyrproject-rtos/zephyr #46347: MCUMGR_SMP_BT: system workqueue blocked during executing shell commands

mjm987, 2022-06-07:
**Describe the bug**
Executing shell commands which directly or indirectly submits system workqueue handlers works fine for the serial shell console (SHELL_BACKEND_SERIAL).
But not so when the shell command is executed via the MCUMGR_SMP_BT console: in this case, the submitted workqueue handler is delayed until the shell command has finished.
Because of this the result(s) and logs of the executed command can't be captured to the MCUMGR_SMP_BT console.

It seems that shell commands itself runs in a system workqueue handler in case of MCUMGR_SMP_BT.

**To Reproduce**
- Enable CONFIG_MCUMGR_SMP_BT, CONFIG_MCUMGR_CMD_SHELL_MGMT and CONFIG_SHELL_BACKEND_DUMMY etc.

- Following code fragment implements a shell command which itself executes a handler running on the system workqueue and then waits (with a timeout) until the handler is executed:

```
...
K_SEM_DEFINE(done_sema, 0, 1);
static struct k_work shell_work;

static void work_handler(struct k_work *work)
{
	LOG_INF("inside work handler");
	k_sem_give(&done_sema);
}

static int cmd_test(const struct shell *sh, size_t argc, char **argv)
{
	ARG_UNUSED(argc);
	ARG_UNUSED(argv);
	k_sem_reset(&done_sema);
	k_work_submit(&shell_work);
	int rc = k_sem_take(&done_sema, K_SECONDS(10));
	switch (rc) {
		case 0: shell_print(sh, "work_handler() executed successfully"); break;
		case -EBUSY: shell_print(sh, "ERR: work_handler() not executed"); break;
		case -EAGAIN: shell_print(sh, "ERR: work_handler() not executed (timeout)"); break;
		default: shell_print(sh, "ERR: unknown Error %d", rc);
	}
	return 0;
}

SHELL_CMD_REGISTER(wq_test, NULL, "Executing a shell cmd which runs a system workqueue handler", cmd_test);

int main(void)
{
	k_work_init(&shell_work, work_handler);
...	
}
```

**Expected behavior**
When executing the shell command via the serial UART console (SHELL_BACKEND_SERIAL) the message "_work_handler() executed successfully_" appears as desired.
But when executing the shell command via Bluetooth SMP console (MCUMGR_SMP_BT), the submitted handler is not executed unless the shell command has finished (in the code above after the timeout is expired), so the message "_ERR: work_handler() not executed (timeout)_" is shown in console.

**Impact**
It is not possible to execute eg. driver functions which itself needs the system workqueue during execution.
Eg. in my case executing LoRaWAN driver function which uses a gpio interrupt submitting a handler on the system workqueue.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: zephyr-sdk-0.14.0_linux-x86_64.tar.gz
 - Zephyr 3.1.99 --> Commit df5d1f220bb0f9ee8feb944ffa2f4a96b8df9c7a

mjm987, 2022-08-14:
I think this issue should be solved, because otherwise, the MCUMGR_SMP_BT console blocks other drivers workqueue handling resulting in unexpected behaviour.

jori-nordic, 2022-08-14:
> I think this issue should be solved, because otherwise, the MCUMGR_SMP_BT console blocks other drivers workqueue handling resulting in unexpected behaviour.

Yeah I'll get to it as soon as I've addressed #34600

carlescufi, 2022-08-31:
Might related to #49661

nordicjm, 2022-08-31:
I'll need to generate some test code and try, but my first thought would it be related to https://github.com/zephyrproject-rtos/zephyr/issues/49717
It is likely that SMP will need to be moved to it's own work queue

nordicjm, 2022-08-31:
https://github.com/nordicjm/zephyr/tree/mcuasyncerase resolves the issue (switches to separate SMP work queue).
On main (BT transport):
```
"\x02\x00\x00\x11\x00\t\x01\x00\xBF""dargv\x9Fgwq_test\xFF\xFF"
"\x03\x00\x00""8\x00\t\x01\x00\xBF""aox.\r\nERR: work_handler() not executed (timeout)\r\nbrc\x00\xFF"
```
On main (shell transport):
```
out:  "\x06\tABsCAAARAAkBAL9kYXJndp9nd3FfdGVzdP//+WI=\n"  ->  43
in:  "\x06\tAEIDAAA4AAkBAL9hb3guDQpFUlI6IHdvcmtfaGFuZGxlcigpIG5vdCBleGVjdXRlZCAodGltZW91dCkNCmJyYwD/qyA=\n"
"AEIDAAA4AAkBAL9hb3guDQpFUlI6IHdvcmtfaGFuZGxlcigpIG5vdCBleGVjdXRlZCAodGltZW91dCkNCmJyYwD/qyA="
Op:  3
Flags:  0
Length:  66
Group:  9
Sequence:  1
ID:  0
0 true
0 true
Text:  "o"
Text:  "\r\nERR: work_handler() not executed (timeout)\r\n"
Text:  "rc"
number:  0
```
With above PR (BT transport):
```
"\x02\x00\x00\x11\x00\t\x01\x00\xBF""dargv\x9Fgwq_test\xFF\xFF"
"\x03\x00\x00""2\x00\t\x01\x00\xBF""aox(\r\nwork_handler() executed successfully\r\nbrc\x00\xFF"
```
With above PR (shell transport):
```
out:  "\x06\tABsCAAARAAkBAL9kYXJndp9nd3FfdGVzdP//+WI=\n"  ->  43
in:  "\x06\tADwDAAAyAAkBAL9hb3goDQp3b3JrX2hhbmRsZXIoKSBleGVjdXRlZCBzdWNjZXNzZnVsbHkNCmJyYwD/cJY=\n"
"ADwDAAAyAAkBAL9hb3goDQp3b3JrX2hhbmRsZXIoKSBleGVjdXRlZCBzdWNjZXNzZnVsbHkNCmJyYwD/cJY="
Op:  3
Flags:  0
Length:  60
Group:  9
Sequence:  1
ID:  0
0 true
0 true
Text:  "o"
Text:  "\r\nwork_handler() executed successfully\r\n"
Text:  "rc"
number:  0
```

nordicjm, 2022-09-20:
Resolved in https://github.com/zephyrproject-rtos/zephyr/pull/49691

mjm987, 2022-09-23:
I tested it and now it works successfully!
Many thx and regards!