zephyrproject-rtos/zephyr #46350: net: tcp: When the first FIN message is lost, the connection does not properly close

ssharks, 2022-06-07:
**Describe the bug**
When the a Zephyr TCP implementation closes a TCP connection and either the FIN or the replied ack is lost, the connection does never properly close.
The Zephyr TCP stack keeps trying, but never accepts the incoming FIN/ACk messsages any more

The problems was discovered on an embedded platform communicating to a PC server. A testcase has been created and submitted as PR to reproduce the issue.

**To Reproduce**
Use the test test_close_partially_obstructed from PR https://github.com/zephyrproject-rtos/zephyr/pull/46349
Steps to reproduce the behavior:
1. ./scripts/twister -T tests/net/socket/tcp -p qemu_x86

**Expected behavior**
Even if the first FIN or FIN/ACK it lost, the TCP connection should still be able to properly close the connection at the retry. In perspective of the testcase, it should successfully complete

**Impact**
At this moment the impact is managable.

**Logs and console output**
The resulting error is:
```
===================================================================
START - test_close_partially_obstructed

    Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/tcp/src/main.c:1028: test_close_partially_obstructed: (time_diff < CONFIG_NET_TCP_INIT_RETRANSMISSION_TIMEOUT * 2 is false)
Socket must have been closed within 2 times the RTO
 FAIL - test_close_partially_obstructed in 0.648 seconds
===================================================================

```

**Environment:**
 - OS: Ubuntu 20.04
 - Commit of PR https://github.com/zephyrproject-rtos/zephyr/pull/46349

**Additional info:**
It looks like the accounting of the sequence number breaks at retransmission of the FIN message

rlubos, 2022-06-08:
Hmm, I've reproduced the "failure" with your test, although I don't observe the problem you describe. The connection is properly closed both sides, however the test fails due to incorrect timings.

Once you reopen the communication channel over loopback, you verify that the client net context is closed before `2 * RTO`. I don't think this is correct. The state machine for the client is as follows:
`TCP_ESTABLISHED -> TCP_FIN_WAIT_1 -> TCP_TIME_WAIT`
Once the client enters `TCP_TIME_WAIT` state, the only way to get out of it is to wait for the timeout to expire (`CONFIG_NET_TCP_TIME_WAIT_DELAY`) which defaults to 250ms. So in order to verify that the connection was closed correctly, you should rather check if it was closed before `RTO + CONFIG_NET_TCP_TIME_WAIT_DELAY`, not `2 * RTO`. After changing this in the test, it passes.

rlubos, 2022-06-08:
Although, the above doesn't prove that everything is fine with the Zephyr TCP stack, since we're testing Zephyr vs Zephyr here. If there are indeed some connection teardown problems vs some other stack, we might need some other way to reproduce it.

ssharks, 2022-06-08:
I can share you this section off a Wireshark trace publically. 18.196.213.123 is the Zephyr device.
[image]

The second thing I see is: When this occurs, the net context stays open until the fin timeout hits on the Zephyr side.

ssharks, 2022-06-08:
Looking at the trace it looks like the time way has elapsed when the retry of the server comes. So this section of the trace looks to be more caused by a mismatch between the server RTO and zephyr CONFIG_NET_TCP_TIME_WAIT_DELAY.

ssharks, 2022-06-08:
Another section of the trace:

rlubos, 2022-06-08:
Thank you for the details. Actually, that RST case should be pretty easy to identify, there aren't many places where Zephyr decides to send RST message. I think I'll try to reproduce this by adding some artificial packet drop to SLIP driver, hopefully I could see the same results.

rlubos, 2022-06-09:
@ssharks I was able to reproduce the scenario from https://github.com/zephyrproject-rtos/zephyr/issues/46350#issuecomment-1150228237 and I think I've identified the root cause - once Zephyr enters TIMED_WAIT state, it ignores any incoming packets. This can be a problem however if the final ACK from the Zephyr side is lost, as the peer in such case keeps retransmitting the final FIN packet. Therefore, Zephyr should be ready for that, i. e. acknowledge any incoming FIN in TIMED_WAIT state as well. I've posted a PR to address this, please give it a try:
https://github.com/zephyrproject-rtos/zephyr/pull/46407

There is one caveat however with the `CONFIG_NET_TCP_TIME_WAIT_DELAY`. Since it defaults to pretty low value (250 ms), in case the time between consecutive retransmissions from the peer is larger than than, Zephyr could already dismiss the connection. In results, any FIN retransmission received after this point results in ICMP destination unreachable response, which might be problematic as well. I'm wondering if we should extend the default value of `CONFIG_NET_TCP_TIME_WAIT_DELAY` a bit, to give the connaction teardown procedure time for a few retransmissions. What do you think?

rlubos, 2022-06-09:
As for https://github.com/zephyrproject-rtos/zephyr/issues/46350#issuecomment-1150236818, I was trying to trigger such behaviour with simulated packet drop, but failed. Specifically, this duplicated ACK from the Zephyr side (frame 365) is a mystery to me. Do you have any tips on how to trigger this scenario? Or maybe you could share a larger part of that pcap, to see the full packet flow since the connection is established?

ssharks, 2022-06-09:
> As for [#46350 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/46350#issuecomment-1150236818), I was trying to trigger such behaviour with simulated packet drop, but failed. Specifically, this duplicated ACK from the Zephyr side (frame 365) is a mystery to me. Do you have any tips on how to trigger this scenario? Or maybe you could share a larger part of that pcap, to see the full packet flow from the point the connection is established?

The recording was taken doing an HTTP/GET on, while introducing severe random packet loss in both RX as well as the TX path of the Zephyr device. At the same time the retry time out was the default 200 ms, but the link rather slow. So it could have been that it actually a stale re-transmit of the ACK after SYN-SYN-ACK.

As this post failed, the host decided to close the connection, instead of the Zephyr client. So this flow is different from the case of 18.196.213.123.

rlubos, 2022-06-09:
> As this post failed, the host decided to close the connection, instead of the Zephyr client. So this flow is different from the case of 18.196.213.123.

Yes, I noticed that in this case the connection teardown is initiated by the server. However, I have no success yet reporducing the issue by simply dropping packets on RX/TX side.

rlubos, 2022-06-09:
@ssharks I've increased the `NET_TCP_TIME_WAIT_DELAY` default to 1500ms  in https://github.com/zephyrproject-rtos/zephyr/pull/46407, I agree this should be a more suitable default for this parameter.

ssharks, 2022-06-09:
> Another section of the trace: [image]
> 
> In this case 167.248.133.62 is the Zephyr device. Apparently the Zephyr stack finds it needed to send a reset packet.

I do suspect it hits this section:
```
if ((conn->send_data_total < len_acked) ||
					(tcp_pkt_pull(conn->send_data,
						      len_acked) < 0)) {
				NET_ERR("conn: %p, Invalid len_acked=%u "
					"(total=%zu)", conn, len_acked,
					conn->send_data_total);
				net_stats_update_tcp_seg_drop(conn->iface);
				tcp_out(conn, RST);
				conn_state(conn, TCP_CLOSED);
				close_status = -ECONNRESET;
				break;
			}
```

Due to the FIN flag the len_acked can be larger then send_data_total. (PSH | FIN). In that case do_close is not set and the connection is not properly dereferenced.
No idea how to reproduce this in the field honoustly.

ssharks, 2022-06-09:
> And as for that part, I think that instead of calling `conn_state(conn, TCP_CLOSED);` directly, the stack should rather set the next state to TCP_CLOSED (i. e. `next = TCP_CLOSED`). Otherwise indeed `do_close` is not set, and the TCP connection is not dereferrenced.

I don't think setting the `next` will work, as the processing is already in the state machine processing next.

rlubos, 2022-06-09:
> I don't think setting the `next` will work, as the processing is already in the state machine processing next.

Not sure I follow this comment, setting `next` simply tells the state machine to process the next state in the same `tcp_in()` call (https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/ip/tcp.c#L2207), if we use `conn_state()` directly, the TCP_CLOSED state, which sets the `do_close` flag will not get processed.

ssharks, 2022-06-10:
Another observation is that we see the device trying to transmit something.

> Not sure I follow this comment, setting `next` simply tells the state machine to process the next state in the same `tcp_in()` call [...]

Thanks for your explanation, I missed this kind of 'do' - 'while' construction. You are right going to the closed state makes sense from this reset condition. It avoids duplicating code with the risk of forgetting something.

ssharks, 2022-06-10:
> Another section of the trace: [image]
> 
> In this case 167.248.133.62 is the Zephyr device. Apparently the Zephyr stack finds it needed to send a reset packet.

This is likely a spurious crawler instead of a Zephyr device. It it outside the region where our devices should be communicating as well as using timestamps, which are not supported in the Zephyr stack.

closed 2022-06-23 by carlescufi