zephyrproject-rtos/zephyr #46366: test_thread_stats_usage fail on arm64 fvp

povergoing, 2022-06-08:
```
INFO    - 1160/1160 fvp_base_revc_2xaemv8a    tests/kernel/usage/thread_runtime_stats/kernel.usage FAILED Failed (armfvp 3.568s)
...
START - test_thread_stats_usage
    Assertion failed at WEST_TOPDIR/zephyr-testing/tests/kernel/usage/thread_runtime_stats/src/test_thread_runtime_stats.c:517: test_thread_stats_usage: stats5.current_cycles < stats3.current_cycles is false
 FAIL - test_thread_stats_usage in 0.57 seconds
```

https://github.com/zephyrproject-rtos/zephyr-testing/runs/6745649272?check_suite_focus=true#step:11:1593

Note that this is a newly added test; see https://github.com/zephyrproject-rtos/zephyr/pull/45489

p.s. This also fails on `fvp_baser_aemv8r`; interestingly, it does not fail on `fvp_baser_aemv8r_aarch32`.

_Originally posted by @stephanosio in https://github.com/zephyrproject-rtos/zephyr/issues/45099#issuecomment-1146842381_

povergoing, 2022-06-08:
@SgrrZhf would you like to post some details here when you have some progress

SgrrZhf, 2022-06-09:
When I select CONFIG_NO_OPTIMIZATIONS, this assertion will fail.
https://github.com/zephyrproject-rtos/zephyr/blob/d84989a718272f29f3a19d4ea99afc7a6aa73211/tests/kernel/usage/thread_runtime_stats/src/test_thread_runtime_stats.c#L516
Adding print will change the results of this case.

I'm going to do other jobs and I will take look at this again some days later.

closed 2022-06-28 by carlescufi