zephyrproject-rtos/zephyr #46372: Intel-ADSP: sporadic core boot

lyakh, 2022-06-08:
**Describe the bug**
This started off as https://github.com/thesofproject/sof/issues/5733 but after weeks of debugging I now managed to reproduce the problem without SOF with just a slightly modified version of the https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/boards/intel_adsp/smoke test. In short, with the attached patch I'm getting an assertion failure
```
Running test suite intel_adsp
===================================================================
START - test_smp_boot_delay
Launch cpu1
Launch cpu2
Launch cpu3
 PASS - test_smp_boot_delay in 0.61 seconds
===================================================================
START - test_cpu_halt
start/halt core 1...
ASSERTION FAIL [z_spin_lock_valid(l)] @ WEST_TOPDIR/zephyr/include/zephyr/spinlock.h:142
```
which with some deep debugging shows that a CPU, that is supposed to be powered off unexpectedly boots. I used custom traces in an SRAM window for "deep debugging," I can make patches or decoded traces available.

**To Reproduce**
1. Apply [20220608-2-z.gitdiff.txt](https://github.com/zephyrproject-rtos/zephyr/files/8861399/20220608-2-z.gitdiff.txt), build and run the test on a TGL platform, e.g. on UP-Xtreme i11.

**Expected behavior**
The test should run through without failing, powered down CPUs should stay off until explicitly powered on.

**Impact**
Showstopper. Although currently only one platform hits this bug only in one test case, potentially all cAVS 2.5 and newer platforms can be affected.

**Logs and console output**
Decoded output from my memory window debugging
```
0x0a520500 0x00000001 0x0a521501 0x00030001 
soc_start_core(core 1)
		      soc_adsp_halt_cpu(core 1)
0x0a520600 0x00000003 0x0a521601 0x00070002 
soc_start_core(core 2)
		      soc_adsp_halt_cpu(core 2)
0x0a520700 0x00000007 0x0a521701 0x000f0003 
soc_start_core(core 3)
		      soc_adsp_halt_cpu(core 3)
0x0a520501 0x0a520601 0x0a520701 0x0a520500 
soc_adsp_halt_cpu(core 1)
	   soc_adsp_halt_cpu(core 2)
		      soc_adsp_halt_cpu(core 3)
				 soc_start_core(core 1)
0x00000001 0x0a521501 0x00030001 0x0a521501 
	   smp_init_top()
		  ^-- expect core 1
			   ^-- PWRCTL = BIT(0) | BIT(1)
			       ^-- running on core 1
*BUG*!				 smp_init_top()
					^-- expect core 1
0x00030003 0x00000000 0x00000000 0x00000000
     ^-- PWRCTL = BIT(0) | BIT(1)
	 ^-- running on core 3
```

lyakh, 2022-06-08:
@nashif @andyross @dcpleung @ceolin please have a look and prioritise accordingly. This issue doesn't occur with XTOS, so it seems like Zephyr is missing some crucial part of cAVS 2.5 multicore support. I can help with my debug setup if desired.

kv2019i, 2022-06-08:
I can reproduce this error with @lyakh your test steps and get the same result (not UP i11, but another cAVS2.5 system).

plbossart, 2022-06-09:
Thanks @lyakh, impressive detective work indeed.

@nashif this is not a 'medium' priority, this problem was identified on March 9 and blocks SOF CI. This is not a drill, failure is not an option, etc.

dcpleung, 2022-06-09:
Looks like it's failing validation for `sched_spinlock`. After disabling timer interrupts at CPU power down, the spinlock assertion is no more.

```
diff --git a/soc/xtensa/intel_adsp/common/mp_cavs.c b/soc/xtensa/intel_adsp/common/mp_cavs.c
index cfd60822cb..58fad442a9 100644
--- a/soc/xtensa/intel_adsp/common/mp_cavs.c
+++ b/soc/xtensa/intel_adsp/common/mp_cavs.c
@@ -43,6 +43,11 @@ __imr void soc_mp_startup(uint32_t cpu)
        if (!IS_ENABLED(CONFIG_SOC_SERIES_INTEL_CAVS_V25)) {
                CAVS_INTCTRL[cpu].l2.clear = CAVS_L2_IDC;
        }
+
+#ifdef CONFIG_CAVS_TIMER
+       /* Need to clear the timer interrupt mask for this cpu. */
+       CAVS_INTCTRL[cpu].l2.clear = CAVS_L2_DWCT0;
+#endif
 }
 
 static ALWAYS_INLINE uint32_t prid(void)
@@ -193,6 +198,16 @@ int soc_adsp_halt_cpu(int id)
                return -EINVAL;
        }
 
+#ifdef CONFIG_CAVS_TIMER
+       /*
+        * Mask timer interrupt for this CPU so it won't wake up
+        * by itself once WFI (wait for interrupt) instruction
+        * runs.
+        */
+       CAVS_INTCTRL[id].l2.set = CAVS_L2_DWCT0;
+       irq_enable(DSP_WCT_IRQ(0));
+#endif
+
        /* Stop sending IPIs to this core */
        soc_cpus_active[id] = false;
 
```

With the above changes, sometimes it would assert on the second start/halt of core 2. However, `test_cpu_behavior()` hangs at core 1. Maybe timer is not fired on core 1 so scheduler never gets out of idle loop?

marc-hb, 2022-06-09:
Impressive effort.

> I now managed to reproduce the problem without SOF with just a slightly modified version of the https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/boards/intel_adsp/smoke test. 

Could this be submitted and added as a new test? Just curious.

dcpleung, 2022-06-09:
Actually, `test_cpu_behavior()` is supposed to hang at CPU 1 due to calling `start_and_halt()` at the end of `test_cpu_halt()` so CPU 1 and up are not running.

lyakh, 2022-06-10:
I have a fix. It's very dirty. It's too dirty to be shown to anyone sane. The way it works is by accepting the bitter truth of being unable to prevent secondary cores from booting sporadically, so instead it just sends them back to sleep every time that happens. It passed both Zephyr native "smoke" test and SOF booting and streaming in the configuration that previously was failing every single boot attempt. Next I should test it on Gimble...

dcpleung, 2022-06-10:
@lyakh do you have another way of reproducing the issue? After disabling timer interrupt at cpu power down, I no longer see the issue on my board.

andyross, 2022-06-11:
The interrupt controller has per-CPU masking available, though right now it's exercised only in the cavs_timer driver.  We could add hooks such that cores that are "off" get their interrupts (all of them) masked until something external tries to boot them.

But note that if the core is bouncing because it received an interrupt, that implies strongly that the power and clock gating, which is supposed to be happening on entry to WAITI, isn't.  A core without power or clock shouldn't be able to boot, and an interrupt shouldn't (as I understand the hardware) be turning that stuff on.

lyakh, 2022-06-14:
> @lyakh do you have another way of reproducing the issue? After disabling timer interrupt at cpu power down, I no longer see the issue on my board.

@dcpleung how and where do you disable the timer interrupt? I was busy with another PR today, I'll clean up a bit and send you (or post here) my current diff, fixing this.

lyakh, 2022-06-14:
> The interrupt controller has per-CPU masking available, though right now it's exercised only in the cavs_timer driver. We could add hooks such that cores that are "off" get their interrupts (all of [...]

@andyross as far as I understood an explanation somewhere, setting a bit in `PWRCTL` doesn't really do anything with the core, it only *allows* it to enter a power-save mode when idle. Logically, since it's just an automatic power-save mode, the core should be able to also leave it automatically. Which is exactly what seems to be happening.

dcpleung, 2022-06-14:
> @dcpleung how and where do you disable the timer interrupt? I was busy with another PR today, I'll clean up a bit and send you (or post here) my current diff, fixing this.

I posted a snippet a few comments above (https://github.com/zephyrproject-rtos/zephyr/issues/46372#issuecomment-1151432969). You probably won't need the re-enabling part since `smp_timer_init()` takes care of that.

lyakh, 2022-06-15:
> I posted a snippet a few comments above ([#46372 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/46372#issuecomment-1151432969)). You probably won't need the re-enabling part since `smp_timer_init()` takes care of that.

@dcpleung but there you wrote that it wasn't a complete stable fix?

dcpleung, 2022-06-15:
> @dcpleung but there you wrote that it wasn't a complete stable fix?

I made another comment about that... the test hangs simply because your diff never re-enables those CPUs. Therefore any following tests requiring other CPUs to work would simply hang.

lyakh, 2022-06-15:
> I made another comment about that... the test hangs simply because your diff never re-enables those CPUs. Therefore any following tests requiring other CPUs to work would simply hang.

@dcpleung You mean after the `test_cpu_halt` test? Yes, that isn't important. As long as the first two tests pass - that would be enough for me. Do they always pass for you with just that diff? If so, I'll check and comment back next week, thanks!

dcpleung, 2022-06-15:
`test_cpu_halt` always passes on my UP X11 Extreme after disabling timer interrupt.

dcpleung, 2022-06-15:
I believe the issue is that due the stale `cpu` number set in `arch_start_cpu()` to `start_rec`. If a CPU wakes up not due to `arch_start_cpu()` but random interrupts, the `cpu` number in `start_rec` would have the previous value. For example, `.cpu == 3` after boot. Now you power down CPU #1, and it wakes back up due to interrupts. Since `start_rec` still contains `.cpu == 3`, this means that `z_mp_entry` will be using the wrong CPU number and thus CPU struct. In this example, the kernel thinks CPU #1 has id `3`. So when both CPU #1 and CPU #3 use the same spinlock at the same time, the validation function asserts because it thinks the same CPU has taken the same lock twice (but in reality, it is taken by two different CPUs).

andyross, 2022-06-16:
That makes a ton of sense to me.  Nice.

FWIW: I still think that this means the root cause is that the core is failing to enter a power gated state.  Is there a hardware mechanism to verify that?

lyakh, 2022-06-20:
> I believe the issue is that due the stale `cpu` number set in `arch_start_cpu()` to `start_rec`. If a CPU wakes up not due to `arch_start_cpu()` but random interrupts, the `cpu` number in [...]

@dcpleung exactly, that's why my "fixes" are full of `s/_current_cpu->id/prid()/g` to catch those cases and send the offending core back to sleep. I tested your diff and indeed it also fixes the problem with the (modified) Zephyr "smoke" test as well as the SOF problem on TGL (UP-Xtreme i11). Then I went back to the original Gimble problem and there your fix appears to be insufficient, whereas my dirty brute-force approach in #46587 works...

lyakh, 2022-06-21:
@dcpleung I've updated #46587 - it's a 2-liner now essentially. That fixes this bug for me.

dcpleung, 2022-06-21:
Hm... there is another issue since the cores are waking up "randomly". Since we are using the same stack for all cores, if multiple cores are waking up at the same time, they are going to overwrite the stack and possibly crashing the other core.

lyakh, 2022-06-23:
@dcpleung right, adding to my comment in https://github.com/zephyrproject-rtos/zephyr/pull/46587#issuecomment-1163323259 I can see several possible solutions to this now, based on our current understanding and available instruments (variants of the below can be proposed too of course):
1. add a Linux kernel quirk for affected platforms to only use cores 0 and 3. We might need a special topology for that
2. ban switching off cores 1 and 2 on all cAVS 2.5 platforms
3. add an SOF configuration quirk, that would be activated by the Linux driver on affected platforms, that would activate a per-core startup stack in Zephyr
4. implement a "return-to-idle" path in assembly without using the stack

plbossart, 2022-06-23:
@lyakh Linux quirks are a hard no from me. We don't have any issue with the non-Zephyr solution and cores are already in use in existing topologies. We can't prevent people from using cores.

dcpleung, 2022-06-23:
I wonder if we should also clear `INTENABLE` (for non-IDC interrupts) at power-down, and restore it during power-on. I am just guessing here... but is it possible that there are level 3+ interrupts being triggered?

Besides, if we have cores "randomly" waking up, this is bad for power consumption, especially on a laptop.

andyross, 2022-06-23:
Well, as I understand it (and I once thought I did but clearly don't anymore), INTENABLE shouldn't be involved.  It's an SR, which means it's part of the Xtensa core and on the same power well, and thus inactive when power is gated.  So it isn't capable of masking/enabling anything on a suspended core.  So at least in theory, any masking being done in the L2 interrupt controller should be sufficient.

But to repeat the same point, if my worries are correct and the cores aren't actually powering down but just entering the normal Xtensa architectural WAITI state, then... sure.  Anything is possible and INTENABLE might very well work.  But if it does, I think that becomes proof the the root cause is elsewhere.

lyakh, 2022-06-24:
> But to repeat the same point, if my worries are correct and the cores aren't actually powering down but just entering the normal Xtensa architectural WAITI state, then...

@andyross no, don't think that's the case. When waking up from a normal WAITI the core would continue to the next instruction. Whereas what we see is that the core resets, i.e. it really powers down. And yes, I tried masking all l2 and l5 interrupts, it didn't help.

dcpleung, 2022-06-28:
@lyakh I remember you mentioned that you tried disabling interrupts (but the cores still waking up). How did you do that? If we can figure out a way to keep the cores powered down, and power them up on demand one by one only, the stack issue would not be a problem as only one core will be using the stack.

lyakh, 2022-06-29:
> @lyakh I remember you mentioned that you tried disabling interrupts (but the cores still waking up). How did you do that? If we can figure out a way to keep the cores powered down, and power them up [...]

@dcpleung `z_xt_ints_off(~0)` in `idle()` but maybe that isn't effective when going to low-power WAITI? I also tried additionally masking all 2nd and 5th level interrupts to what you proposed. And I've never seen level 3 or 4, don't think we use any of them.

dcpleung, 2022-06-29:
> @dcpleung `z_xt_ints_off(~0)` in `idle()` but maybe that isn't effective when going to low-power WAITI? I also tried additionally masking all 2nd and 5th level interrupts to what you proposed. And I've never seen level 3 or 4, don't think we use any of them.

I don't think that's enough as it simply modifies `INTENABLE` which only works when the CPU is running (see @andyross comment above: https://github.com/zephyrproject-rtos/zephyr/issues/46372#issuecomment-1164643168). When the CPU is powered down, it is the L2 interrupt controller that wakes up the CPU. So you will to mask interrupts using `CAVS_INTCTRL`. Could you try https://github.com/dcpleung/zephyr/commits/tmp/sof_core_power_down and see if this helps? This masks/unmasks interrupts during suspend/resume.

lyakh, 2022-06-30:
> > @dcpleung `z_xt_ints_off(~0)` in `idle()` but maybe that isn't effective when going to low-power WAITI? I also tried additionally masking all 2nd and 5th level interrupts to what you proposed. And I've never seen level 3 or 4, don't think we use any of them.
> 
> I don't think that's enough as it simply modifies `INTENABLE` which only works when the CPU is running (see @andyross comment above: [#46372 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/46372#issuecomment-1164643168)). When the CPU is powered down, it is the L2 interrupt controller that wakes up the CPU. So you will to mask interrupts using `CAVS_INTCTRL`. Could you try https://github.com/dcpleung/zephyr/commits/tmp/sof_core_power_down and see if this helps? This masks/unmasks interrupts during suspend/resume.

@dcpleung unfortunately that didn't work with the original problem. Adding an assembly snippet to decouple stack for different cores worked, but I don't like that, I'd rather check the core ID and go back to waiti directly in asm without using stack at all. I'll try to implement that

lyakh, 2022-06-30:
I've updated #46587 - it returns to waiti in asm, so, no stack conflicts.

dcpleung, 2022-07-01:
#46587 is merged.

closed 2022-07-01 by dcpleung