zephyrproject-rtos/zephyr #46378: CONFIG_SYS_CLOCK_TICKS_PER_SEC affects app code speed with tickless kernel

kv2019i, 2022-06-08:
**Describe the bug**
With CONFIG_TICKLESS_KERNEL=y build of Zephyr, we observe the SOF application code is running slower with a higher CONFIG_SYS_CLOCK_TICKS_PER_SEC setting. This is not the expected behavior for a tickless kernel.

**case A) CONFIG_SYS_CLOCK_TICKS_PER_SEC=15000**

[00007179] <inf> ll_schedule: ll timer avg 2022, max 2552, overruns 0

DSP load 5.27% (2022/38400*100%)

**case B) CONFIG_SYS_CLOCK_TICKS_PER_SEC=50000**

[00009227] <inf> ll_schedule: ll timer avg 2040, max 2792, overruns 0

DSP load 5.31% (2040/38400*100%)

So increasing the systick setting introduces a +0.2MCPS overhead to audio pipeline execution
in this case. The difference is not huge, but the expectation is no difference. 

**To Reproduce**
- Build SOF application with Zephyr (https://thesofproject.github.io/latest/getting_started/build-guide/build-with-zephyr.html). 
- play one audio stream 
- observe "ll_schedule: ll timer avg" in SOF application log

**Expected behavior**
No overhead to application execution from higher sys-clock-ticks-per-sec setting.

**Impact**
Not meeting performance target

**Logs and console output**
n/a

**Environment (please complete the following information):**
 - zephyr: a78e7c5c90d8cb31ec550ce289503bd698675585
 - sof: ebc222dab56868ea3cb0d5abec5057990eae6bb9
 - HW: Intel Tiger Lake cAVS2.5 system

**Additional context**
n/a

kv2019i, 2022-06-08:
FYI @ceolin @nashif

kv2019i, 2022-06-21:
@nashif  @fabiobaltieri  @ceolin I propose to bump the priority of this bug. With bug, https://github.com/thesofproject/sof/issues/5921 , the sys-tick setting also impacts the k_timer behaviour and this led to subtle but severe audio quality issues.

ceolin, 2022-06-22:
> @nashif @fabiobaltieri @ceolin I propose to bump the priority of this bug. With bug, [thesofproject/sof#5921](https://github.com/thesofproject/sof/issues/5921) , the sys-tick setting also impacts the k_timer behaviour and this led to subtle but severe audio quality issues.

The priority makes sense to me.

peter-mitsis, 2022-06-23:
I've spent a few cycles looking into and thinking about this. I don't have a hard answer on it, but I do have a few thoughts.

At one level of analysis, it seems reasonable to expect to that raising CONFIG_SYS_CLOCK_TICKS_PER_SEC on a tickless kernel from 15000 to 50000 should not result in additional overhead. However, that does make certain assumptions.

Before getting to deep into them, I'd like to establish some baseline numbers. As I understand it, for tiger lake cavs, we have the following...

CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=38400000

When the tick rate is at 15000 Hz each tick is about 2560 cycles (66.7 usec). When at 50000 Hz, each tick is about 768 cycles (20 usec). There are two immediate implications of this. The first is that the time at which a tick is delivered/announced may change. This can occur if timeouts are specified in milliseconds or microseconds and then converted to ticks. It is not clear to me if the application is doing this.

The second is that the change in granularity can in theory lead to more ticks being delivered (this will be application specific though). This can occur if there should be timer expirations between 20 usec and 66.7 usec apart from each other. Under the 15000 Hz scenario, they may get rolled together into the delivery of a single tick, while in the 50000 Hz scenario, being of finer granularity it could allow for the delivery of separate ticks. Again, it is not yet clear to me if this is applicable to the SOF application.

I'm going to present one other set of numbers. As mentioned in the issue, at 15000 Hz the average and max numbers were 2022 and  2552 respectively--a difference of 530. It is my understanding that these are measure in cycles. And at 50000 Hz those numbers were 2040 and 2792--a difference of 752. It may be coincidence, but I can't help but notice that 752 cycles is awfully close to 768--the time for one tick. It may be a red herring, but my suspicion is that the change in granularity is giving rise to "extra" (as compared to the 15000 Hz case) ticks being delivered and processed.

Can someone more familiar with the SOF code provide some additional information?

ceolin, 2022-06-24:
> I've spent a few cycles looking into and thinking about this. I don't have a hard answer on it, but I do have a few thoughts.
>
> At one level of analysis, it seems reasonable to expect to that raising [...]

SoF has its own scheduler concept on top of Zephyr, but the general idea is that there is a timer, that ticks every 1000usec and all the work is done within this time. On Xtos (the implementation without Zephyr) all the works is done from the interruption context and the timer is programmed to tick every 1000 usec. In other hand on Zephyr,  the timer interruption "just" wakes up other threads that, effectively, do the work that is not done in the interruption context. AFAIR, the timer interruption is not masked. 

The trick part is that SoF uses `TICKLESS_KERNEL`, and all the periodic ticks shouldn't happen. Assuming (just to simplify) that the only timer event is that one, we should have an timer interruption every 1000us only.

@kv2019i ^

> 
> I'm going to present one other set of numbers. As mentioned in the issue, at 15000 Hz the average and max numbers were 2022 and 2552 respectively--a difference of 530. It is my understanding that these are measure in cycles. And at 50000 Hz those numbers were 2040 and 2792--a difference of 752. It is probably a coincidence, but I can't help but notice that 752 cycles is awfully close to 768--the time for one tick.
> 
> One other observation. The period in cycles at 15000 Hz is longer than the test's average time. And the period in cycles at 50000 is a bit more than a third of the test's average time.It may be a red herring, but my suspicion is that the change in granularity is giving rise to "extra" (as compared to the 15000 Hz case) ticks being delivered and processed.
> 
> Can someone more familiar with the SOF code provide some additional information?

kv2019i, 2022-06-30:
Thanks @peter-mitsis for the analysis. It indeed seems like a unlikely coincidence that the avg/max delta is so close to one tick. The 2040/2792 values are actually reads of the 38.4Mhz platform clock (behind k_cycle_get_32() on these platforms). The diff is measured over one 1000us tick, so difference of 752 cycles is 19.583usec, so indeed very close to the 20.000 usec of one 50kHz tick.

And ack @ceolin , SOF application indeed registers a k_timer() that fires every 1000us, and this is used to kick (with k_sem) threads to do the actual work. We can have occasional IPC or IDC irqs, but these do not happen in the stable state of this test where the measurement is made (a single stream is running and we just observe "stable state" performance). 

We also looked at the overall number of IRQs , and this seems to match with the intended design, e.g. nothing additional is fired between the 1000us k_timer ticks.

ceolin, 2022-06-30:
@kv2019i  I'm seeing a degradation in cavs timer that I'm not seeing in other implementations. I've a simple sample that just configure a timeout for every 1s and try to take a semaphore that will be given in the timeout callback. When the semaphore is taken the thread just busy/wait for 90% of the tick time.
Basically, with lower `CONFIG_SYS_CLOCK_TICKS_PER_SEC` the timer is constantly triggered, but with higher values the timer gets degraded over time.

There were no additional ticks  interruptions though.

ceolin, 2022-07-26:
So, I tried to come up with something similar to SoF scheduler but simpler. I run this sample in cavs25 with two different configurations, `CONFIG_SYS_CLOCK_TICKS_PER_SEC=80000` and `CONFIG_SYS_CLOCK_TICKS_PER_SEC=12000`. With 80k we missed some ticks and with 10k we have two cpus handling the "same" tick. Please see the logs below:

80k
[image]

12k
[image]

The code used to test it was:
```
#include <zephyr/zephyr.h>

#include <zephyr/logging/log.h>
LOG_MODULE_REGISTER(sample, LOG_LEVEL_INF);

#define STACK_SIZE 2048
#define TIMER_PERIOD_US 1000

K_THREAD_STACK_ARRAY_DEFINE(work_stacks, CONFIG_MP_NUM_CPUS, STACK_SIZE);

struct zephyr_domain {
	struct k_thread domain_thread[CONFIG_MP_NUM_CPUS];
	struct k_sem sem[CONFIG_MP_NUM_CPUS];
};

struct k_timer timer;
static struct zephyr_domain zd = {};

void work_fn(void *arg1, void *arg2, void *arg3)
{
	int cpu = (int)arg1;
	struct zephyr_domain *zephyr_domain = arg2;

	while (true) {
		k_sem_take(&zephyr_domain->sem[cpu], K_FOREVER);

		/* Lets spinning to simulate some work */
		k_busy_wait(TIMER_PERIOD_US * 0.7);
	}
}

static void timer_fn(struct k_timer *t)
{
	int core;

	struct zephyr_domain *zephyr_domain = k_timer_user_data_get(t);

	if (!zephyr_domain) {
		return;
	}

	for (core = 0; core < CONFIG_MP_NUM_CPUS; core++) {
		k_sem_give(&zephyr_domain->sem[core]);
	}

	LOG_INF("CPU[%d] uptime: %llu\n", arch_curr_cpu()->id, k_uptime_get());
}

void main(void)
{
	k_timeout_t start = {0};
	k_timeout_t timeout = K_USEC(TIMER_PERIOD_US);

	k_timer_init(&timer, timer_fn, NULL);
	k_timer_user_data_set(&timer, &zd);

	LOG_INF("number of ticks per second: %u\n", CONFIG_SYS_CLOCK_TICKS_PER_SEC);
	for (int i = 0; i < CONFIG_MP_NUM_CPUS; i++) {

		k_sem_init(&zd.sem[i], 0, 999);
		k_thread_create(&zd.domain_thread[i], work_stacks[i], STACK_SIZE,
				work_fn, (void *)i, &zd, NULL,
				-CONFIG_NUM_COOP_PRIORITIES, 0, K_FOREVER);
		k_thread_cpu_mask_clear(&zd.domain_thread[i]);
		k_thread_cpu_mask_enable(&zd.domain_thread[i], i);

		k_thread_start(&zd.domain_thread[i]);
	}
	k_timer_start(&timer, start, timeout);
}
```

teburd, 2022-07-27:
The race I believe I sorted out in my mind before in this code before goes with a flow like the following inside the adsp timer driver. Though was told there wasn't a race before, seeing this I'm less convinced.

Imagine the flow in intel_adsp_timer.c goes like the following.

core 0 takes driver lock, updates last count, releases driver lock and calls sys_clock_announce, takes k_timer lock
core 1 takes driver lock, updates last count, release driver lock and calls sys_clock_announce, waits on k_timer lock
core 0, takes the driver lock sets the next compare relative to the last count provided by core 1, releases driver lock.

Lock around the entire things instead *or* only process timer compares on one core and avoid the locks would be something I'd seriously consider trying.

teburd, 2022-07-27:
So I tried two things real quick with maybe more surprising results than I expected...

If I add an outter spin lock around the entire compare_isr call (including the announce) I lose *more* ticks.

If I bail early in compare_isr (if arch_curr_cpu()->id != 0) { return; } I get every tick exactly like I'd expect. Maybe some spin lock contention causing issues?

teburd, 2022-07-31:
To add to my analysis above, the issue comes when core 1 updates the last_count (when the last announce occurred) while core 0 is still announcing. If core 1 happens to be a tick or more ahead, time is lost from the perspective of core 0 as it will potentially appear as if the announce took 1 or more ticks less than reality. This is more likely to occur when the tick rate is faster as the tick time period is shorter so the race is more likely to be seen.

andyross, 2022-08-02:
Sneaking in:

> when core 1 updates the last_count (when the last announce occurred) while core 0 is still announcing. If core 1 happens to be a tick or more ahead

But... that's all correct.  There isn't a separate clock per core.  It's a single device.  If core 1 sees time as 1 tick ahead while core 0 is still announcing, that means that core 0 took more than one tick to finish the announce calls, by definition.  No time is being "lost", it's just slow.  No amount of timer architecture is going to make an ISR finish faster.

(FWIW: none of that explains the reported performance impact, though.  The ISR going longer than a tick is more common with high tick rates, but the performance impact is supposed to be the same regardless.)

Maybe the bug is on the other side?  Is the next timeout computation being done based on the "frozen" time when it really should be realtime?

peter-mitsis, 2022-08-03:
We have the following call sequence executed in the timer expiration handler (z_timer_expiration_handler). Note that this handler is a callback routine called from within sys_clock_announce() in the context of the timer ISR.

`   k_uptime_ticks() -> sys_clock_tick_get() -> curr_tick + sys_clock_elapsed()
`

sys_clock_elapsed() simply does ...

`       uint32_t ret = (count32() - (uint32_t)last_count) / CYC_PER_TICK;`

What I think is happening is that when a the system clock rate is high, processing the timer ISR can be long enough that at least a tick's worth of time has elapsed. On a tickless kernel, this is not guaranteed to trigger the timer. However,  `sys_clock_elapsed()` can detect that time has passed. Unfortunately this is being done from within the context of the timer ISR and that tick has not yet been *announced*. This premature detection of the tick is leading to k_timer drift.

This can be fixed by modifying `sys_clock_tick_get()` to use `curr_tick + elapsed()` instead as `elapsed()` is designed to be used from within sys_clock_announce().

PR #48664 (2022-08-04): Fixes sys_clock_tick_get()
Fixes an issue in sys_clock_tick_get() that could lead to drift in
a k_timer handler. The handler is invoked in the timer ISR as a
callback in sys_tick_announce().
  1. The handler invokes k_uptime_ticks().
  2. k_uptime_ticks() invokes sys_clock_tick_get().
  3. sys_clock_tick_get() must call elapsed() and not
     sys_clock_elapsed() as we do not want to count any
     unannounced ticks that may have elapsed while
     processing the timer ISR.

Fixes #46378
commit 71ef669e (2022-08-04): kernel: Fixes sys_clock_tick_get()

Fixes an issue in sys_clock_tick_get() that could lead to drift in
a k_timer handler. The handler is invoked in the timer ISR as a
callback in sys_tick_announce().
  1. The handler invokes k_uptime_ticks().
  2. k_uptime_ticks() invokes sys_clock_tick_get().
  3. sys_clock_tick_get() must call elapsed() and not
     sys_clock_elapsed() as we do not want to count any
     unannounced ticks that may have elapsed while
     processing the timer ISR.

Fixes #46378
commit a9aaf048 (2023-02-03): kernel: Fixes sys_clock_tick_get()

Fixes an issue in sys_clock_tick_get() that could lead to drift in
a k_timer handler. The handler is invoked in the timer ISR as a
callback in sys_tick_announce().
  1. The handler invokes k_uptime_ticks().
  2. k_uptime_ticks() invokes sys_clock_tick_get().
  3. sys_clock_tick_get() must call elapsed() and not
     sys_clock_elapsed() as we do not want to count any
     unannounced ticks that may have elapsed while
     processing the timer ISR.

Fixes #46378

(cherry picked from commit 71ef669ea4a73495b255f27024bcd5d542bf038c)
closed 2022-08-04 by nashif