zephyrproject-rtos/zephyr #46399: Suggestion on the framework of Flash driver

rogeryou, 2022-06-09:
**Is your enhancement proposal related to a problem? Please describe.**
A clear and concise description of what the problem is.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.
[image]

nvlsianpu, 2022-06-10:
I'm looking forward for this rework. Probably the key factor for success will be proper API interface between vendors low layer driver part and higher generic flash driver part.

rogeryou, 2022-07-07:
> @rogeryou since the flash subsystem maintainer, @nvlsianpu, seems to have no objections to the proposal as it stands, will you send a Pull Request to start work on this?

Hi，@carlescufi
 I have submitted the source code，please review.

manoj153, 2022-09-15:
Sorry for going off-topic. What application do you use to create the diagram? 
@rogeryou  they look great!

rogeryou, 2022-09-28:
> Sorry for going off-topic. What application do you use to create the diagram? @rogeryou they look great!

draw.io

MaureenHelm, 2023-04-20:
@rogeryou do you plan to return to this?

savvn001, 2023-06-05:
Any updates on this? Am interesting also in using NAND flash with Zephyr

rogeryou, 2023-08-03:
@MaureenHelm @carlescufi @nvlsianpu 
Hi,
Because some developers are also interested in using NAND flash with Zephyr, we will submit a PR for NFTL. The PR is based on the SPI NAND driver #50690. Can we reopen that PR  and discuss how to merge it?"

carlescufi, 2023-09-06:
> Because some developers are also interested in using NAND flash with Zephyr, we will submit a PR for NFTL. The PR is based on the SPI NAND driver #50690. Can we reopen that PR and discuss how to merge it?"

I see the PR is opened now, so let's continue the discussion in the PR.

rruuaanng, 2025-01-07:
Happy New Year everyone, it's 2025, do we still need to push it forward?

daniel-0723, 2025-01-13:
Hi @rruuaanng , we submitted a [PR](https://github.com/zephyrproject-rtos/zephyr/pull/65659) related to this issue if you want to get the Octal/Quad driver source code. And we are planning to push it forward, do you have some suggestions? @MaureenHelm @carlescufi @rruuaanng

rruuaanng, 2025-01-13:
> Hi @rruuaanng , we submitted a [PR](https://github.com/zephyrproject-rtos/zephyr/pull/65659) related to this issue if you want to get the Octal/Quad driver source code. And we are planning to push it forward, do you have some suggestions? @MaureenHelm @carlescufi @rruuaanng

I have reopened it. You can first address the suggestions from the current maintainers, and then we can move on to the next improvements.